betaflight/betaflight #10004: No "Bidirectional DSHOT"  and rpm filter when Custom Mixer used

joschtec, 2020-07-10:
tested with:
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

if custom mixer and rpm filters is used: no arming possible.

set pid_process_denom = 1
set motor_pwm_protocol=DSHOT300
set scheduler_optimize_rate=on
set dshot_burst=off
set dshot_bidir=on

mixer custom
mmix reset
mmix 1 1.000 -0.952 1.000 1.000
mmix 2 1.000 -1.000 -0.933 -1.000
mmix 3 1.000 0.952 1.000 -1.000
mmix 4 1.000 1.000 -0.933 1.000
save

Asizon, 2020-07-10:
Seems that you are ussing old rpm snipets, you need dhsot burst ON to work with the actual Bitbang implementation

joschtec, 2020-07-10:
> Seems that you are ussing old rpm snipets, in this board you need dhsot burst ON to work with the actual Bitbang implementation, and ofcourse, no more snipets needed since 4.1 version, only if you want timer based dshotbidir.
> Just make `set dshot_burst=on` save and try

Does not solve the issue. Neither does 'set dshot_burst=auto'.

etracer65, 2020-07-10:
There is nothing about using a custom motor mix that would prevent bidirectional DSHOT from working. You have some other configuration issue.

As requested when you opened the issue:

> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

joschtec, 2020-07-10:
This is my configuration which works (without Custom Mixer)
[rooster-diff.txt](https://github.com/betaflight/betaflight/files/4904168/rooster-diff.txt)

If you activate custom mixer in CLI: 

'mixer custom
mmix reset
save'

you get the status:
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=43degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3960, Max available config: 16384
Gyros detected: gyro 1
GYRO=MPU6000, ACC=MPU6000
OSD: MAX7456
System Uptime: 40 seconds, Current Time: 2020-07-10T15:53:21.941+00:00
CPU:3%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
**Arming disable flags**: RXLOSS CLI MSP **_RPMFILTER_**

When powering up the ESCs, the last two tones are missing.
Arming is not possible.

etracer65, 2020-07-10:
> If you activate custom mixer in CLI:
> 
> 'mixer custom
> mmix reset
> save'

Of course that doesn't work - you haven't defined any motors in your custom mix.

And the commands you listed in your original issue are incorrect. Motor mixing starts with motor 0, not 1. As a result you are not setting any mixing and to the flight controller there are no motors defined.

Try:
```
mixer custom
mmix reset
mmix 0 1.000 -0.952 1.000 1.000
mmix 1 1.000 -1.000 -0.933 -1.000
mmix 2 1.000 0.952 1.000 -1.000
mmix 3 1.000 1.000 -0.933 1.000
```

And finally note that there is rarely any need to define a custom motor mix for a QUAD-X motor layout. People seem to think you need to do this for "deadcat" or "H" style frames but it's really not necessary. Just use the default QUAD-X mixer.

joschtec, 2020-07-10:
Ah! 
Motor mixing starts with motor 0!
This fixed the issue.
Thanks a lot!

closed 2020-07-10 by joschtec