betaflight/betaflight #10007: Can't choose IBUS from serial based receivers

excusemua, 2020-07-10:
Hello, sorry if its wrong place to post this kind of issue, i have no idea where to seek for help in this matter. 
I just bought a quad (UAV115, similar to Kingkong 110GT) PNP, i installed the receiver myself (Fli14+)
The problem is that i cant choose IBUS from the receiver list. I mean i try to but everytime i disconnect and connect to the flight controller it resets to spectrum1024. Im not sure how but i once made the connection, could see my radio input in the configurator but cant make it to work again (it may be at the very first/second time i connected to the board)
I made a diff before making any changes, here it is
```
# diff all

# version
# BetaFlight/SPRACINGF3 3.0.1 Oct 18 2016 / 10:35:52 (48b7b4f)

# reset configuration to default settings
defaults

# name

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL

# beeper

# map

# serial
serial 1 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100
aux 1 1 1 900 1300
aux 2 2 1 1300 1700

# adjrange

# rxrange

# rxfail

# master
set motor_pwm_protocol = MULTISHOT
set serialrx_provider = SBUS
set gyro_sync_denom = 2
set pid_process_denom = 1

# profile
profile 0

set p_pitch = 22
set p_roll = 22
set p_yaw = 50
rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 1

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 2

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# restore original profile selection
profile 0

# save configuration
save
```
and here is the diff when it started not working
```
# 

# diff all

# version
# BetaFlight/SPRACINGF3 3.0.1 Oct 18 2016 / 10:35:52 (48b7b4f)

# reset configuration to default settings
defaults

# name

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL

# beeper

# map

# serial
serial 1 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1725 2100
aux 1 1 1 900 1250
aux 2 2 1 1325 1650

# adjrange

# rxrange

# rxfail

# master
set mid_rc = 0
set min_check = 0
set max_check = 0
set rc_interpolation = OFF
set rc_interpolation_interval = 0
set motor_pwm_protocol = MULTISHOT
set serialrx_provider = IBUS
set gyro_sync_denom = 2
set airmode_activate_throttle = 0
set rx_min_usec = 0
set rx_max_usec = 0
set pid_process_denom = 1

# profile
profile 0

set p_pitch = 22
set p_roll = 22
set p_yaw = 50
rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 1

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# profile
profile 2

rateprofile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# restore original rateprofile selection
rateprofile 0

# restore original profile selection
profile 0

# save configuration
save

# 
```

**Setup / Versions**
 - Flight controller LDARC Kingkong Micro F3 6DOF

Thanks

etracer65, 2020-07-11:
```
BetaFlight/SPRACINGF3 3.0.1 Oct 18 2016
```
Realistically not much chance for support for a firmware version that's nearly 4 years old. Further, this flight controller is no longer supported in current versions as the hardware is too old. The latest version you can try is 4.0.6.

excusemua, 2020-07-11:
Thanks for your reply. You mean try to connect to my drone with betaflight version 4.0.6 right?