betaflight/betaflight #10009: TPA for P and D seperated

SupaflyFPV, 2020-07-11:
This is a feature request to have separate TPA for P and D. 

I would be happy to have it in CLI or the configurator. For example rather than have the option in CLI for TPA = D or PD it could be in the configurator TPA for P term with breakpoint, and TPA for D with its separate breakpoint.

For racing I can understand why they want high P gains all the time, and a significant D only attenuation at high throttle for the sake of precision and noise control. 

However in freestyle flying the requirements are a little different IMO. I prefer a 'loosening' of P term over increased throttle which IMO makes motors run smoother on punch out where absolute accuracy of P term is not so important at high throttle. With freestyle a pilot will typically throw the quad up in the air hard to then do tricks. But also do very hard stop moves at 0 throttle which needs high P term low throttle. This way you can get a gradual loosening of the P gain over the throttle range which is generally smoother IMO.

The reason why D needs to be separated is because when pulling out of dives - reducing the D gain with throttle I believe is contributing to propwash. I check in logs and saw after a big dive the throttle level could be 80%. However the optimal TPA for P is 0.4 with a breakpoint at 1500. Currently that results in a significant reduction in d gains when pulling out of dives. IMO an ideal D gain TPA might be 0.30 at 1800 breakpoint.

Let me know what you think of this idea. Thanks!

ctzsnooze, 2020-07-13:
I am not a fan of attenuating P with throttle, especially for racing, since with a lot of TPA on P, quickly changing throttle changes P itself and hence turn radius.  

The request here appears to relate to two quite different flight problems:

- a desire to have more PID authority at zero / low throttle, for stability and authority doing tricks at zero throttle

- a desire to avoid cutting D via TPA when pulling out of dives, ie when throttling up at the end of a dive.

Thrust linear - especially the revised form as provided now in 4.3 nightlies - is a much better way of improving zero throttle authority than TPA.  While fundamentally intended to address the invariable reduction in thrust at lower rpm due to prop physics, it definitely makes a big improvement in low throttle stability and in low/zero throttle PID authority.  

However Thrust Linear applies most strongly when one pair is at zero and the other pair at say 10 or 20%.  It does very little when one pair is at zero and the other pair is at 50% or higher.  It does not 'boost P' at zero throttle in the same way that a higher P value would.

Hence TPA on P would allow more P at zero throttle, generating a higher P:D balance at low throttle than at higher throttle.

Adding more P at low throttle, but not having more D at low throttle, will lead to a P:D imbalance at low throttle which usually results in small wobbles on HD footage when cruising at low throttle.  

If the pilot wants more PID authority at low throttle, an absolute increase in both P and D, in proportion, usually works best.

We've learned so far is that having high P at high throttle is usually not a problem, even with significant D attenuation.  This is of itself a surprising finding.   We do know that attenuating D at high throttle is very important not just for motor heat but for wobble prevention.

Having an independent way to manipulate TPA on P would allow the pilot to try out different high throttle P:D ratios, and it may well be that tunes can be optimised a little better by doing that.  

I can imagine all kinds of 'recipes' coming out for TPA... and maybe some will be good.  

I support this, but only from a testing perspective.  We would need to get confirmation that it did result in significant improvement.  And I think I'd need help writing the tuning guide :-)

SupaflyFPV, 2020-07-13:
Thanks for the feedback and response...

TPA on P alone is not necessarily a 'problem fix' but a preference one - where the preference is a looser 'feel' over increased throttle. Partly based around the idea that the motors spin up better with lower P gains, a behaviour I picked up in the early days of tuning by ear pre-blackbox logging...I could and should log/test it to confirm however.

Thrust Linear 4.3 may improve low throttle authority resulting in lower overall P gain requirement, but it is unlikely to a 40% margin which is where I currently run my TPA PD. I currently have a very generic 4s freestyle build carrying a session that sits at 75 P gain to be locked in - but I do not want 75 P term at full throttle...

Here is a pilot flying my freestyle build and you can just about hear the motor noise above the music - the throttle is brief full pumps and lots of fast rolls/flips at low throttle...(by another pilot flying my build)

https://www.youtube.com/watch?v=ytwaQr6I8XU

The quad is being thrown around and weighs >700g so requires quite a lot of control at lower throttle and stability (but not necessarily exact tracking) at high throttle...I do suspect there is a significant divergence of the needs of racing and freestyle in this point...