betaflight/betaflight #10010: osd menu crashed when using Telemetry output on betaflight 4.3

m4ximuel, 2020-07-11:
**Describe the bug**
cannot open osd menu, just blanked

**To Reproduce**
Steps to reproduce the behavior:
1. all is well, can open osd menu
2. betaflight configurator -> connect -> configuration -> turn on 'TELEMETRY' under 'Other features'
3. can't open osd menu via controller.
It's stop working and cleared all osd object.

**Expected behavior**
maybe it cause via crsf telemetry.

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F411 (S411) 4.3.0 Jul 11 2020 / 11:55:43 (99f9a3446) MSP API: 1.44
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F411_PRO, version: bc19b7dc, date: 2020-04-01T04:37:12Z

# start the command batch
batch start

board_name IFLIGHT_F411_PRO
manufacturer_id IFRC

# name: vx130-cf

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1600 2025 0 0
aux 1 1 0 1425 1575 0 0
aux 2 1 0 1875 2025 0 0
aux 3 2 1 1250 1300 0 0
aux 4 2 1 1475 1525 0 0
aux 5 2 1 1675 1725 0 0
aux 6 2 1 1925 1975 0 0
aux 7 13 0 1250 1300 0 0
aux 8 13 0 1475 1525 0 0
aux 9 13 0 1675 1725 0 0
aux 10 13 0 1925 1975 0 0
aux 11 35 0 1200 1250 0 0
aux 12 35 0 1425 1475 0 0
aux 13 35 0 1625 1675 0 0
aux 14 35 0 1900 1950 0 0
aux 15 39 1 1425 1575 0 0
aux 16 39 1 1875 2025 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set gyro_lowpass2_hz = 400
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 140
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_min_hz = 120
set dyn_notch_max_hz = 650
set dyn_lpf_gyro_min_hz = 320
set dyn_lpf_gyro_max_hz = 800
set acc_calibration = 73,-41,-240,1
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = CRSF
set dshot_idle_value = 250
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_scale = 111
set ibata_scale = 150
set yaw_motors_reversed = ON
set small_angle = 180
set thrust_linear = 20
set osd_link_quality_pos = 2401
set osd_tim_2_pos = 2423
set osd_craft_name_pos = 2453
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 2433
set debug_mode = GYRO_SCALED
set cpu_overclock = 120MHZ
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5769
set vcd_video_system = NTSC
set name = vx130-cf

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 195
set vbat_sag_compensation = 100
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 20
set pidsum_limit = 1000
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 51
set i_pitch = 99
set d_pitch = 37
set f_pitch = 116
set p_roll = 46
set i_roll = 94
set f_roll = 109
set p_yaw = 50
set i_yaw = 99
set f_yaw = 109
set d_min_boost_gain = 30
set d_min_advance = 0
set ff_interpolate_sp = ON
set ff_smooth_factor = 0
set idle_min_rpm = 32

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set throttle_limit_type = CLIP
set throttle_limit_percent = 98

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: MOTOR 2
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: MOTOR 1
    CH3: MOTOR 2
TIM3: FREE
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: FREE
DMA1 Stream 5: MOTOR 1
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
FC: stm32f411 ( IFLIGHT_F411_PRO / iflight 16x16 2.1 stack )
VTX: tbs unify pro32 nano 1.1
RECEIVER: tbs crossfire nano rx, crsf protocol with rx tx connected

mikeller, 2020-07-12:
@m4ximuel:

> set cpu_overclock = 120MHZ

You are overclocking your CPU. This can lead to malfunctions. Have you tried without overclocking?

m4ximuel, 2020-07-12:
> @m4ximuel:
> 
> > set cpu_overclock = 120MHZ
> 
> You are overclocking your CPU. This can lead to malfunctions. Have you tried without overclocking?

I tried during finding what is problem. 
I'm in flight, and test again with under condition:
set cpu_overclock = off
telemetry on
Iosd menu still not working. It fine only telemetry off.

haslinghuis, 2020-07-13:
See also #2084

@m4ximuel You should get this message:
https://github.com/betaflight/betaflight-configurator/pull/2084/files#diff-4b047b995ef22f0fc7be5abe848b34c4R4154-R4156

It looks like the setting `osd_displayport_device` is undefined here.
Please try setting it to `AUTO` or `MAX7456` and save. Unplug USB, plug Lipo, wait few seconds and plugin the USB.
Guess you also need a nightly build for configurator for this to work.

m4ximuel, 2020-07-14:
@haslinghuis 

1. 
get osd_displayport_device
result:
osd_displayport_device = AUTO

2. 
set osd_displayport_device = MAX7456
feature TELEMETRY
result:
osd was turned off after enter osd menu. same problem.

3.
set osd_displayport_device = AUTO
feature TELEMETRY
result:
osd was turned off after enter osd menu. same problem.

haslinghuis, 2020-07-14:
Thanks, if `osd_displayport_device` is not `set` in target or config it seems to default to `AUTO`. So this seems unrelated.
However I cannot reproduce the issue flashing with full chip erase using latest Configurator 18.0 (local build) on Ubuntu 20.04 with MATEKF411 BF4.3 using latest development release # 2242
OSD tab displays even without using LiPo. Toggling TELEMETRY feature has no effect on OSD.

m4ximuel, 2020-07-15:
@haslinghuis 
I'll test it with same condition on my another fc. the another fc had F405, but I have no more time now. maybe it need 2-3 days.

m4ximuel, 2020-07-22:
@haslinghuis 
sorry for waiting, my fc delivery delayed.
same problem on my new fc, this is ELINF405

haslinghuis, 2020-07-22:
@m4ximuel 
Betaflight has recently changed the way of initialization to actual detection of the OSD chip. Can you please attach LiPo and wait a few seconds before you connect the FC to your PC and see if this changes behavior switching the telemetry feature on/off?

m4ximuel, 2020-07-22:
@haslinghuis 
test 1:
telemetry **on**
battery connect
usb connect
enter osd menu
result:
osd erased, camera still working.
flag 'CMS_MENU'

test 2:
telemetry **off**
battery connect
usb connect
enter osd menu
result:
osd working.
flag 'CMS_MENU'

haslinghuis, 2020-07-22:
@m4ximuel 
> Expected behavior
> maybe it cause via crsf telemetry.

Using crossfire nano (fw v3.28) here.

m4ximuel, 2020-07-22:
@haslinghuis
You mean need testing with receiver downgrade to v3.28? I think it's good point but also think it need must working with newest firmware crossfire because it working perfectly on betaflight 4.2 . 
I'll test it.

haslinghuis, 2020-07-22:
You can try, didn't notice there is crossfire update. I will update to 3.72 later. But I don't think the problem has to do with crossfire. Still waiting for Opentx 2.4. Just mentioning my test setup trying to isolate the problem. And to be clear you referring to disappearing OSD overlay in your goggles. Does it show in configurator (can you still choose PAL/NTSC ?)

m4ximuel, 2020-07-22:
working. And I hold to NTSC in almost case. ELIN sometimes can't choose but can now, and iflight f411 can choose everytime.

m4ximuel, 2020-07-22:
@haslinghuis
I'm building new quad, and SucceX-D Mini F7 ( IFLIGHT_F722_TWING ) have same problem.

wuzle, 2020-07-26:
I have had this happen as well on an Aikon F7 board.

I can't post any DIFFS/Settings at the moment as a friend has the quad with him. But the described Problem was the same (I could not nail it to the telemetry setting though, it may be that, I have not tested it yet) I'll report back if I have my Quad back and can test.

mikeller, 2020-07-27:
@m4ximuel:

> # Betaflight / STM32F411 (S411) 4.3.0 Jul 11 2020

You are using an unstable development version of the firmware - bugs and parts of it not working is normal. Can you try changing to a released version of the firmware and check if the problem perists?

m4ximuel, 2020-07-27:
@mikeller 
It's working well on both v4.2 and  v4.2.1
of course I know that the released version is well for all.

mikeller, 2020-07-27:
@m4ximuel: Ok. Have you tried with the latest development build?

m4ximuel, 2020-07-27:
> @ m4ximuel : 좋아. 최신 개발 빌드를 사용해 보셨습니까?
[image]
[image]
still not working..
Target: IFLIGHT_F722_TWING
Manufacturer ID: IFRC
Version: 4.3.0 #2250
Binary: betaflight_4.3.0_STM32F7X2-2250-1d78f9784.hex
Date: 27-07-2020 20:23

mikeller, 2020-08-01:
@m4ximuel: Can you please test with https://github.com/betaflight/betaflight/pull/10062#issuecomment-667477411?

m4ximuel, 2020-08-01:
@mikeller OSD still crashed with betaflight_4.3.0_STM32F7X2_f93b20ac34.hex . not tested other builds.

mikeller, 2020-08-01:
@m4ximuel: Thanks for testing - there must be another bug then, I tested with STM32F745, and it seems to be working fine there.

m4ximuel, 2020-08-01:
@mikeller something wrong in STM32F411, when I input data of diff all, It show error to me.
config: manufacturer_id: MTKS, board_name: MATEKF411, version: be0c9205, date: 2020-01-15T19:44:32Z
```
# defaults nosave

# resetting to defaults
###ERROR IN timer: INVALID ALTERNATE FUNCTION FOR C08: 'AF3'###
###ERROR IN timer: INVALID ALTERNATE FUNCTION FOR C06: 'AF3'###
###ERROR IN timer: INVALID ALTERNATE FUNCTION FOR C09: 'AF3'###
###ERROR IN timer: INVALID ALTERNATE FUNCTION FOR C07: 'AF3'###
###ERROR IN dma: INVALID DMA OPTION FOR PIN C08: '1'###
###ERROR IN dma: INVALID DMA OPTION FOR PIN C06: '0'###
###ERROR IN dma: INVALID DMA OPTION FOR PIN C09: '0'###
###ERROR IN dma: INVALID DMA OPTION FOR PIN C07: '1'###
```

mikeller, 2020-08-02:
@m4ximuel: I just tested with the STM32F411 target from the zip file, and for me this works just fine, including using CMS. Can you try flashing again, and make sure you follow the instructions in the video?

m4ximuel, 2020-08-02:
@mikeller: Unfortunately, I can't proceed with testing for STM32F411 anymore. because one  quad dead with my mistake and last one went to my friend. I checked once more but can confidently say that I followed the steps in the video correctly. I'm very sorry to leave a variable. I will test STM32F411 as soon as possible. just now, I can test STM32F7X2 target.
For STM32F7X2, I have the same problem about CMS on file of the zip. Do you need to me let's turn telemetry off and try again?

m4ximuel, 2020-08-02:
@mikeller: telemetry off and working. same situation.

mikeller, 2020-08-03:
@m4ximuel: Ah, so now I get what you meant in your bug report. Just to recap, please correct if I am wrong:

Original behaviour reported for 99f9a3446:
- CMS activated - lockup

Behaviour after #10062:
- CMS activated without CRSF & telemetry configured - works
- CMS activated with CRSF & telemetry configured - lockup

haslinghuis, 2020-08-03:
I finally can reproduce. CMS activated with CRSF & telemetry configured with latest dev build:

1. During CMS activation all OSD elements disappear permanent.
2. Another invocation (thr mid, yaw left, pitch up) shows CMS menu again but you can hardly move - only using thr mid, yaw left, pitch up. Can't escape the menu from here.

mikeller, 2020-08-03:
@haslinghuis: Instead of using build numbers that do not relate to anything in the source repository, can you please supply the changeset id (the value in brackets in CLI `version`) - this will make it possible to identify easily where the problem started.

haslinghuis, 2020-08-03:
Already had it written down: `a8085bef8` 
(seems changeset id is also part of the hex name in the (dev) build number in configurator)

mikeller, 2020-08-03:
@haslinghuis: Yes, the changeset id is part of the CI build filename as well (and if you don't put it into backticks, GitHub will automatically convert it to a link to the changeset).

Due to CI contracting concurrent merges into a single build, it looks like 944eb989d54a55f720accd4732c13a9841ab5319 is the culprit for this - this is the last changeset before the one of the problematic build to touch CRSF telemetry. @codecae do you have time to look into this?

m4ximuel, 2020-08-03:
[image]
sry, my desktop went to hell. so I can't work to Wednesday. All the our service centers closed for vacation.  I'll test it after Wednesday if necessary. and yes @mikeller and @haslinghuis . That issue is correct. I think I should study more because English is not my native language.

codecae, 2020-08-03:
Interesting... I'll see if I can see what's going on.

codecae, 2020-08-03:
Full disclosure, though -- getting married in a few days, so my time is pretty spotty.

ledvinap, 2020-08-03:
> Due to CI contracting concurrent merges into a single build, it looks like [944eb98](https://github.com/betaflight/betaflight/commit/944eb989d54a55f720accd4732c13a9841ab5319) is the culprit for this - this is the last changeset before the one of the problematic build to touch CRSF telemetry. 

I added some comments to [944eb98], 2 of them are potential culprit ...

haslinghuis, 2020-08-03:
@ledvinap I checked your 2 code snippets on a local branch, build ok with minor adjustment, but didn't do the trick.
Out of interest: The non optimal part with memcpy, do you have a suggested replacement snippet ;-)

codecae, 2020-08-03:
the issue is likely the assignment at:

https://github.com/betaflight/betaflight/commit/944eb989d54a55f720accd4732c13a9841ab5319#diff-3a380c5dce1e052338f9fae5ecdebf42R195

While I appreciate the optimization suggestions, I think we're missing the point.

codecae, 2020-08-03:
@haslinghuis If you already have the hood open, can you try changing:

` const bool cmsReady = (cmsInMenu && (pCurrentDisplay = &crsfDisplayPort))`

to:

`const bool cmsReady = (cmsInMenu && (pCurrentDisplay == &crsfDisplayPort))`

haslinghuis, 2020-08-03:
Sure. wait...

codecae, 2020-08-03:
@haslinghuis thought that might be it...

If you want to go ahead and submit that line change as a PR I'll review it.  We can do the optimizations in a subsequent PR.

haslinghuis, 2020-08-03:
I already have those 2 optimizations included. I do a PR. Only have to search how I add the non staged modified files to a branch. I am on synced master now

ledvinap, 2020-08-03:
> While I appreciate the optimization suggestions, I think we're missing the point. Let's fix the bug an optimize later.

Yes, I was reading the code and making suggestion on the fly. The are not important, just comments about coding style ...

ledvinap, 2020-08-03:
> `const bool cmsReady = (cmsInMenu && (pCurrentDisplay == &crsfDisplayPort))`

Hmm ... missed that one. Skipping unnecessary parentheses (which is IMO in BF coding style guide) will help - gcc would generate warning

codecae, 2020-08-03:
@mikeller #10075 is sign, sealed, just needs delivery.  Want me to merge or hold off?

PR #10075 (2020-08-04): Fixed crsf-cms
Fixes #10010 
Special thanks to @codecae and @ledvinap !
PR #10062 (2020-08-11): Fixed lockup when entering CMS.
Fixes #10010. 

This was introduced by #9746, from the looks of it because it is trying to modify read-only flash memory (constant strings) in some case. Surprisingly this was not discovered when testing with CRSF.

The current fix is to copy the uppercased string into a temporary buffer instead - there might be better ways to do this.
closed 2020-08-04 by mikeller