betaflight/betaflight #10012: Motor Output Limit Scale Factor % not working with Turtlemode

rhys100, 2020-07-13:
**Describe the bug**
If running high KV motors, 2550kv in this instance - turtle mode does not run at my 77% I have specifried. It spools up 100% (I think) and my escs blow.

**To Reproduce**
High KV, turtle mode, and say goodbye to your esc and cap.

**Expected behavior**
Should only spool motors to maximum %

Photo of my Cap popping and FET blowing - https://photos.app.goo.gl/mC6caHzF7MVx5nx57

It would make sense to fix this but with https://github.com/betaflight/betaflight/issues/9997

**Flight controller configuration**
```
# feature
feature TELEMETRY
feature DISPLAY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# serial
serial 2 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# aux
aux 0 0 1 900 1300 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 26 1 900 1300 0 0
aux 5 35 2 1300 1700 0 0
aux 6 36 3 1700 2100 0 0
aux 7 36 2 1300 1700 0 0
aux 8 40 1 900 2100 0 0
aux 9 49 4 1700 2100 0 0

# adjrange
adjrange 0 0 5 900 1300 12 5 0 0
adjrange 1 0 5 1325 1600 12 5 0 0
adjrange 2 0 5 1600 2100 12 5 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set gyro_lowpass_hz = 0
set yaw_spin_threshold = 700
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 450
set dyn_lpf_gyro_min_hz = 0
set acc_trim_pitch = 2
set acc_calibration = 99,17,29,1
set rc_interp = MANUAL
set rc_smoothing_derivative_type = PT1
set max_aux_channels = 12
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set min_throttle = 1070
set dshot_idle_value = 400
set dshot_bidir = ON
set dshot_bitbang = OFF
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set align_board_roll = 180
set bat_capacity = 800
set vbat_max_cell_voltage = 436
set vbat_warning_cell_voltage = 340
set ibata_scale = 170
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set runaway_takeoff_prevention = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 70
set osd_cap_alarm = 800
set osd_alt_alarm = 135
set osd_esc_temp_alarm = 95
set osd_tim1 = 2561
set osd_vbat_pos = 448
set osd_rssi_pos = 2426
set osd_link_quality_pos = 2395
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2519
set osd_flymode_pos = 333
set osd_g_force_pos = 2
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2465
set osd_crosshairs_pos = 200
set osd_ah_pos = 200
set osd_current_pos = 2487
set osd_mah_drawn_pos = 2456
set osd_motor_diag_pos = 45
set osd_craft_name_pos = 2507
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 352
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2433
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 2497
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2283
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_flip_arrow_pos = 2318
set osd_core_temp_pos = 13
set osd_log_status_pos = 353
set osd_stick_overlay_left_pos = 225
set osd_stick_overlay_right_pos = 246
set osd_pid_profile_name_pos = 352
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 4
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5880
set gyro_2_align_yaw = 900
set gyro_rpm_notch_q = 700
set dterm_rpm_notch_q = 1000
set rpm_notch_lpf = 100
set name = Q racer

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 340
set dyn_lpf_dterm_curve_expo = 10
set dterm_lowpass2_hz = 250
set anti_gravity_gain = 7000
set iterm_relax_cutoff = 20
set iterm_limit = 500
set pidsum_limit_yaw = 1000
set throttle_boost = 2
set p_pitch = 38
set i_pitch = 85
set d_pitch = 34
set f_pitch = 96
set p_roll = 37
set d_roll = 34
set f_roll = 95
set d_min_roll = 21
set d_min_pitch = 21
set d_min_boost_gain = 35
set motor_output_limit = 77
set auto_profile_cell_count = 6
set launch_trigger_throttle_percent = 45
set launch_angle_limit = 70
set ff_interpolate_sp = AVERAGED_3
set ff_smooth_factor = 30
set ff_boost = 17

rateprofile 1

# rateprofile 1
set thr_mid = 45
set thr_expo = 30
set rates_type = ACTUAL
set roll_rc_rate = 19
set pitch_rc_rate = 19
set yaw_rc_rate = 19
set roll_expo = 56
set pitch_expo = 56
set yaw_expo = 56
set roll_srate = 54
set pitch_srate = 54
set yaw_srate = 54
set tpa_rate = 75
set tpa_breakpoint = 1200
set throttle_limit_type = SCALE
set throttle_limit_percent = 98 
```

rhys100, 2020-07-15:
Hey @etracer65 do you mind checking this out to see if it can be fixed / issue stands as I have stated? I think this is your code so you would be best to check it over?

etracer65, 2020-07-15:
> I think this is your code so you would be best to check it over?

No, not my code.

rhys100, 2020-07-15:
@

> > I think this is your code so you would be best to check it over?
> 
> No, not my code.

Who wrote the turtle mode code then?

haslinghuis, 2022-05-10:
Fixed in https://github.com/betaflight/betaflight/issues/9997

closed 2022-05-10 by haslinghuis