betaflight/betaflight #10014: cannot open one .BFL file

Torchi, 2020-07-14:
I have retrofitted an old F450 frame with a new controller running 4.2 Betaflight.  I have made several quad so far but this was the first time I tried interfacing with an old frame, in particular with ESCs having  PWM interfaces.  Perfect on the bench, I run into crashed at liftoff.  After having crashed (oscillating like a top toy...  but one thing at the time) I tried to read the blackbox file with latest "Betaflight Blackbox Explorer" with no success as it seems to hang forever and does not open the file in the GUI.
I have tried "blackbox_decode" and instead that can decode the file.
I attach the blackbox file as .txt type.
[LOG00004.BFL.txt](https://github.com/betaflight/betaflight/files/4916807/LOG00004.BFL.txt)
I attach the diff file as .txt type.
[diff.txt](https://github.com/betaflight/betaflight/files/4916818/diff.txt)
bye

0crap, 2020-07-17:
[image]

PIDtoolbox does open the file and it shows there is almost nothing in it.
Just try again with a normal flight of some minutes and see how it goes.
Sometimes it happens you run into a bad file, nothing to worry about unless it's always corrupt.

Torchi, 2020-07-17:
**Ocrap** thank you for your comment.  I have installed PIDtoolbox and replicated your screenshot.  PIDtoolbox bases  on blackbox_decode, which works on the file as I said earlier,  plus reads the setup info.  So the BFL file looks well formed.  But is not read by Blackbox Explorer,  **which could indicate the tool has problems with small files**, as you said.   I recognise a limitation  in a tool is not important but it was just an attempt to understand why the quad got crazy.
Why the file is **so** small ?  I armed and then gave a bit of throttle to lift, once in the air it immediately looked wobbly, starting to behave  more and more like a "hula hoop" dancer. I then tried to decrease throttle to come down but it was useless:  it was oscillating and slowly climbing. Once it reached about 3 meter height, being out of control, I disarmed (falling and breaking one quad arm).   The screen of PIDtoolbox is perfectly consistent with the event.  I immediately thought 1) not to have configured well the rotation of the FC (I had to do that so that the FC  fits in the frame). I also thought 2) to have connected the ESCs to the wrong outputs.  I checked that I did not those  mistakes (I even  made  short clips of the model / configurator comparison).  So, this is the main problem I mentioned earlier.

0crap, 2020-07-17:
Nobody knows why your quad misbehaves. The logfile sure does not help because it's (almost) empty.
As said before, try again with a normal flight and see if the logfile is can be opened.
If so, please close this ticket.
If you need help troubleshooting why your quad misbehaves this is not the place to ask.
Please go to [Betaflight Slack](https://slack.betaflight.com/), Facebook, RCGroups etc. to seek assistance,

Torchi, 2020-07-18:
I have followed your suggestion to go to Betaflight slack:  somebody may have had a similar problem  (lets'hope...)

I have produced now even shorter log files by another, different and functional quad and they are opened with no problems by Blackbox explorer.    I have also produced longer log files by the fallen quad and none of them  can be opened (tests on the bench because of the broken arm).  It is certain  that, once I have repaired the quad still will not fly.  So I decide that, unless somebody spots a configuration problem from the diff, the best is for me to change the ESC-motor pairs with something newer e.g. supporting DSHOT so I can use a configuration similar to the ones that I have seen working in other quads.   

If there is nothing you can add, you can close the ticket.

mikeller, 2020-07-19:
@Torchi:

- try using a different SD card
- try re-formatting the SD card with the official tool: https://www.sdcard.org/downloads/formatter/

Torchi, 2020-07-19:
Thank you but it is not an SD problem.  I went to Slack as I was suggested and there a person named "etracer" has noticed that the problem was due to the fact I had not set the switch that makes PID loop frequency independent from PWM frequency.  Once I set it the problem with logging was fixed (now it logs normally) and hopefully also the problem with flying.  I say hopefully because until I receive by Amazon the replacement for the broken arm I have to wait for a new test. But at this point I believe it will (reasonably) fly. That was to update you that the problem is, I think, fixed.

closed 2020-07-19 by Torchi