betaflight/betaflight #10016: MATEKF411RX with Bidirectional DSHOT *and* SOFTSERIAL (SmartPort Telemetry)

fiechr, 2020-07-14:
First, I know this is basically a configuration issue, but I've searched everywhere and also found some information on this, but since I do not understand the hardware behind completely, I'm still unsure what the final answer actually is.

I want to use a R-XSR with Telemetry enabled with my Tinyhawk Freestyle II. For this I've re-mapped TX1 to SOFTSERIAL 1 (SERIAL_TX 11) to connect SmartPort to. This works fine.

As soon as this is enabled, Bidirectional DSHOT doesn't work anymore.
[image]

I found some information, that explains this:
https://github.com/betaflight/betaflight/pull/8978
https://github.com/betaflight/betaflight/pull/8892

So it seems the Timer needed for Bidirectional DSHOT is claimed by SOFTSERIAL first.

Still, I'm unsure if this means, that there is no chance at all to get both Features running at the same time on this board, or if I need to change the configuration or select another Pin (which obviously are rare on these small boards).

Happy to add the answer to the/a Wiki page for this target, so others don't need to ask this question again.

```
diff all

# version
# Betaflight / STM32F411 (S411) 4.2.0 Jun 14 2020 / 03:04:43 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF411RX, version: ee671311, date: 2019-10-16T11:49:37Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411RX
manufacturer_id MTKS
mcu_id 005d00443139510134363539
signature 

# name: TH Freestyle II

# resources
resource MOTOR 1 B06
resource MOTOR 2 B08
resource MOTOR 3 B10
resource MOTOR 4 B07
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# feature
feature -RX_SPI
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 1 1400 1600 0 0
aux 1 1 0 900 1225 0 0
aux 2 2 0 1225 1350 0 0
aux 3 13 3 1400 1600 0 0
aux 4 36 2 1900 2100 0 0
aux 5 47 0 1350 1500 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 UNKNOWN  U CUSTOM     0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 5 RACEBAND R CUSTOM     0    0 5732 5769 5806 5843    0    0
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# master
set acc_calibration = -66,-9,30,1
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set serialrx_provider = SBUS
set blackbox_p_ratio = 0
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set align_board_yaw = 90
set current_meter = NONE
set small_angle = 180
set vtx_band = 2
set vtx_channel = 5
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5809
set gyro_1_align_yaw = 1800
set name = TH Freestyle II

profile 0

# profile 0
set throttle_boost = 7
set acro_trainer_angle_limit = 45
set level_limit = 45

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_expo = 30
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 120
set roll_expo = 40
set pitch_expo = 30
set yaw_expo = 30

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# ```

dkustec, 2020-07-14:
Did you try the FPORT?

fiechr, 2020-07-14:
I was thinking about that, too. As far as I understand, even with FPORT firmware, I would need to modify the R-XSR to get the uninverted signal from SPORT on the receiver to TX1 on the FC. I would maybe try that, if there is no other way.

etracer65, 2020-07-14:
It's a timer conflict. Using the pins for UART1 is probably not going to work. You could probably use the LEDSTRIP pad for softserial instead. The F411 has very limited timer resources so it's challenging to use for this purpose.

etracer65, 2020-07-14:
And the R-XSR has uninverted pads on the bottom which you can use instead. The pad labeled "P" is uninverted and you could just use UART1 TX without doing any mapping or using softserial. This would be the best approach.

fiechr, 2020-07-15:
Ok, Thank you, I will try this!

fiechr, 2020-07-16:
Happy to report that it works with FPORT on TX!

I've flashed the FPORT firmware, connected the uninverted S.PORT pad on the receiver to TX1 on the FC.

Then, I've removed the SOFTSERIAL port, reset TX1 to default, set RX provider to FPORT and `serialrx_halfduplex` to ON.

Thank you again!

closed 2020-07-16 by etracer65