betaflight/betaflight #10018: Display Failsafe Procedure on Arming

Luq1308, 2020-07-15:
The title should be self-explanatory. A brief one or two seconds of currently enabled failsafe procedure on OSD when arming will prevent one from accidentally flying long range with "drop" procedure selected instead of GPS rescue.

etracer65, 2020-07-15:
Configuring your failsafe settings is a critical preflight activity and your responsibility. Very few people actually ever change the failsafe procedure after initial setup so displaying this all the time would be a bad idea.

Luq1308, 2020-07-15:
I know some people (including myself) who occasionally switch between "drop" for general or proximity flying, and "GPS rescue" for long range mission. Not everyone is blessed to fly robust 915M system and 2.4G could randomly failsafe in situations where GPS rescue is undesirable. I haven't (yet) make such mistake but I know a friend of mine who had an accident as I explained. The message is shown briefly and not all the time, and can you elaborate on why this would be a bad idea because I can't think a single reason why such prompt shouldn't be shown.

etracer65, 2020-07-15:
Making 99.999% of the people see a message that serves no purpose is not the correct way forward. Displaying a message all the time will just cause people to be annoyed and ignore it causing its value to be zero. A better solution would be to set it up as an OSD warning that's disabled by default. Then people who are interested in this type of warning could be the ones that enable and theoretically pay attention to it. But also the issue with this is that warnings are persistent and don't only display for a "couple of seconds" on arming.

0crap, 2020-07-17:
Please don't do this, it would be very annoying.