betaflight/betaflight #10027: Betaflight telemetry Fuel sensor

modsley, 2020-07-17:
**Describe the bug**
Betaflight 4.2 in telemetry sensor Fuel shows mAh used, not used battery percentage even if battery capacity is set.
Betaflight telemetry description page says "Fuel - if capacity set, remaining battery percentage mah drawn otherwise."
But it does not work that way. Shows mah drawn even if capacity is set.

**To Reproduce**
Reproduced on Crazybee F4 Pro v2 and matek f405-ctr controllers

**Expected behavior**
Show remaining capacity if battery capacity is set

**Flight controller configuration**
Any

**Setup / Versions**
Betaflight 4.2
Crazybee F4 Pro v2, Matek f405-ctr controllers

**Additional context**

haslinghuis, 2020-07-17:
Please provide a `diff` and `resource show all` enclosed in 3 backticks ( ` ). 

At least we need to know which telemetry protocol and other settings used to debug.
Where do you expect the value to show up? On the Taranis, OSD or BFC?
Without a current meter sensor the fuel gauge won't work.

battery.c has this function:
```
uint8_t calculateBatteryPercentageRemaining(void)
{
    uint8_t batteryPercentage = 0;
    if (batteryCellCount > 0) {
        uint16_t batteryCapacity = batteryConfig()->batteryCapacity;

        if (batteryCapacity > 0) {
            batteryPercentage = constrain(((float)batteryCapacity - currentMeter.mAhDrawn) * 100 / batteryCapacity, 0, 100);
        } else {
            batteryPercentage = constrain((((uint32_t)voltageMeter.displayFiltered - (batteryConfig()->vbatmincellvoltage * batteryCellCount)) * 100) / ((batteryConfig()->vbatmaxcellvoltage - batteryConfig()->vbatmincellvoltage) * batteryCellCount), 0, 100);
        }
    }

    return batteryPercentage;
}
```

etracer65, 2020-07-17:
Or the documentation is simply incorrect and it should always just be mAh drawn.

haslinghuis, 2020-07-17:
In sensors/battery.c: `calculateBatteryPercentageRemaining()` is only called by 'batteryUpdateConsumptionState()' used for battery warning levels and is (also) called by `batteryUpdateStates()` which is called in fc/tasks.c. 

But `calculateBatteryPercentageRemaining()` is used in other telemetry protocols: `crsf`, `frsky_hub`, `ibus_shared` and `mavlink`.

Changing telemetry/smartport.c could solve this:
```
case FSSP_DATAID_FUEL:
	smartPortSendPackage(id, getFuelLevel()); // given in mAh, should be in percent according to SmartPort spec
	*clearToSend = false;
	break;

static int16_t getFuelLevel(void)
{
    int16_t data;

    if (batteryConfig()->batteryCapacity > 0) {
        data = (uint16_t)calculateBatteryPercentageRemaining();
    } else {
        data = (uint16_t)constrain(getMAhDrawn(), 0, 0xFFFF);
    }
   return data;
}
```

etracer65, 2020-07-17:
Changing this runs the risk of "breaking" it for people that are expecting it to always be mAh drawn in the telemetry (as it's apparently been for a while - maybe forever). So it's not just as simple as "fixing" it. We could just as easily "fix" the documentation.

haslinghuis, 2020-07-17:
Yeah much easier ;-) Also people using smartport with softserial could be impacted by the added calculations (did not test).

etracer65, 2020-07-17:
The bigger issue would be people that might have logical switches set up in their radios to generate warnings in their radio and expect the telemetry sensor value to be a number representing total consumed mAh (I used to do this before the OSD days). The proposed change would break this unless they set their battery capacity to 0 in Betaflight (a change not needed before). Being forced to do this will then impact OSD elements that use it, battery state (warning, critical, etc.) and so on and cause negative side effects all for something that probably never worked before and just had incorrect documentation.

I would say in general that having a telemetry sensor that behaves unexpectedly differently based on non-required setting being populated is also not a good design.

VitaliyNimych, 2020-07-18:
This can be made switchable in the CLI, for example: battery_fuel_percent = ON / OFF
and I see two variable in enum, need del one 
FSSP_DATAID_CAP_USED   = 0x0600
FSSP_DATAID_FUEL       = 0x0600

etracer65, 2020-07-18:
Of course it can. But that means more code, less flash available, and more settings. All for something that never existed and could be argued to simply be a documentation error.

VitaliyNimych, 2020-07-18:
give it away battery_fuel_percent variable :)

haslinghuis, 2020-07-18:
@VitaliyNimych `FSSP_DATAID_CAP_USED = 0x0600` is defined but not used anywhere.

Also see:  https://github.com/iNavFlight/inav/pull/2050

VitaliyNimych, 2020-07-18:
This does not mean that it should be there, it is not used.
I see refactor and use smartport_fuel_percent 

if (batteryConfig()->batteryCapacity > 0) {
                        smartPortSendPackage(id, calculateBatteryCapacityRemainingPercentage()); 
                    } else {
                        smartPortSendPackage(id, mAhDrawn);
}
https://github.com/iNavFlight/inav/pull/2002/commits/fc4e8692169114125233ae909e8e72c4ef51824a

haslinghuis, 2020-07-18:
@VitaliyNimych 
`if (batteryConfig()->batteryCapacity > 0 && battery_fuel_percent) {`

haslinghuis, 2020-07-18:
@VitaliyNimych 
`smartPortSendPackage(id, batteryConfig()->batteryCapicity > 0 && battery_fuel_percent ? calculateBatteryCapacityRemainingPercentage() : mAhDrawn());`

VitaliyNimych, 2020-07-18:
I think good case

```
if (batteryConfig()->batteryCapacity > 0 || batteryConfig()-> battery_fuel_percent ) 
    smartPortSendPackage(id, calculateBatteryCapacityRemainingPercentage());
else
    smartPortSendPackage(id, mAhDrawn);
```

etracer65, 2020-07-18:
> This does not mean that it should be there, it is not used.
> I see refactor and use smartport_fuel_percent
> 
> if (batteryConfig()->batteryCapacity > 0) {
> smartPortSendPackage(id, calculateBatteryCapacityRemainingPercentage());
> } else {
> smartPortSendPackage(id, mAhDrawn);
> }
> [iNavFlight/inav@fc4e869](https://github.com/iNavFlight/inav/commit/fc4e8692169114125233ae909e8e72c4ef51824a)

Read the earlier comments. Changing something that has been one way forever will inevitably break it for some people that used it the way it is now. It's not that it's technically difficult to do this and we know how to change it. But there's a strong argument to **NOT** change it.

And it's just as likely that the documentation is wrong and the code is correct and as intended.

haslinghuis, 2020-07-18:
@etracer65 I will do a PR for changing documentation soon. We respect this will confuse current users. I think the issue should be closed. Frsky had it chance to change this years ago.

closed 2020-08-24 by stale[bot]