betaflight/betaflight #10028: Crash Flip Motor Percent making all motors rev up

AliB10, 2020-07-17:
**Describe the bug**
When crashflip_motor_percent is set to 70, all motors spin up when rolling left and right, instead of just the 2 motors on the side. 
All 4 motors spin up on pitch, All 4 motors spin up on yaw.

**To Reproduce**
Set crashflip_motor_percent=70
Reverting back to 100% doesnt fix it, but reflashing firmware fixes it.

**Expected behavior**
When moving roll, pitch or yaw in turtle mode, only 2 motors spin up.

**Flight controller configuration**
```# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 29 2020 / 08:08:49 (b480103d4) MSP API: 1.43
# config: manufacturer_id: FFPV, board_name: FF_RACEPIT, version: 6fac4231, date: 2019-10-18T22:07:46Z

# start the command batch
batch start

board_name FF_RACEPIT
manufacturer_id FFPV

# name: AliB-R

# resources
resource MOTOR 2 B11
resource MOTOR 3 B01

# feature
feature -RX_PARALLEL_PWM
feature TELEMETRY

# serial
serial 0 64 115200 57600 0 115200
serial 1 16384 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# aux
aux 0 0 0 1725 2100 0 0
aux 1 0 2 900 1225 0 0
aux 2 35 2 900 1600 0 0
aux 3 40 1 1725 2100 0 0

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set dyn_notch_width_percent = 4
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 550
set acc_calibration = -79,74,-4,1
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_derivative_type = PT1
set serialrx_provider = CRSF

set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set crashflip_motor_percent = 70
set small_angle = 180
set pid_process_denom = 2
set osd_warn_link_quality = ON
set osd_cap_alarm = 1300
set osd_vbat_pos = 375
set osd_rssi_pos = 37
set osd_link_quality_pos = 34
set osd_craft_name_pos = 2409
set osd_debug_pos = 35
set osd_avg_cell_voltage_pos = 2423
set osd_flip_arrow_pos = 2253

set vcd_video_system = NTSC
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 900
set name = AliB-R

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dyn_lpf_dterm_curve_expo = 0
set dterm_lowpass2_hz = 165
set vbat_sag_compensation = 100
set throttle_boost = 0
set d_pitch = 30
set f_pitch = 86
set d_roll = 28
set f_roll = 81
set f_yaw = 81
set d_min_roll = 18
set d_min_pitch = 20
set idle_min_rpm = 20

rateprofile 2

# rateprofile 2
set rates_type = ACTUAL
set roll_rc_rate = 25
set pitch_rc_rate = 22
set yaw_rc_rate = 21
set roll_expo = 44
set pitch_expo = 44
set yaw_expo = 44
set roll_srate = 39
set pitch_srate = 33
set yaw_srate = 46

# end the command batch
batch end```

```A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: I2C_SCL 3
A09: SERIAL_TX 1
A10: CAMERA_CONTROL
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 3
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: LED 2
B09: LED 1
B10: MOTOR 4
B11: MOTOR 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: ADC_CURR
C02: ADC_BATT
C03: BEEPER
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 2
C09: I2C_SDA 3
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: INVERTER 3
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH3: CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
``` 

**Setup / Versions**
 - Furious Racepit f4, Hobby RC
 - Racerstar 30a aniversary
 
I worked through every setting in my diff command untill i found the setting that fixed it.

AliB10, 2020-07-17:
I have also had this on both my quads. Hardware they are identical but problam was the same on both

etracer65, 2020-07-17:
Yes, that's what it's supposed to do and the purpose of the setting. There is nothing wrong. Put the value back to 0 if you don't want this behavior.

closed 2020-07-17 by etracer65

AliB10, 2020-07-17:
Oh. Sorry about that, I thought it was like a throttle cap for crash flip. IE only 70% of throttle is used in crash flip