betaflight/betaflight #10029: Add Dshot1200 back.

MB3Drift, 2020-07-18:
Hello everybody.
I’m here to request that Betaflight supports Dshot1200 again. I Can agree on the fact than not everybody feel the difference between 600 and 1200, but I 100% can and I miss it a lot. Dshot1200 feels more planted, more rigid and stable, and dshot600 just feels sloppy to me.
Before you ask, I cant go back to 3.5.1 since bidirectional Dshot fried for FCs and one ESC, I upgraded to 20x20 stack and I didn’t see it was an F7, and it doesn’t have a hex older than 4.1 for that target (PyroF7). I didn’t realize it was a F7 before buying, and since I didn’t update Betaflight since the 3.5.1 days, I didn’t realize either than Dshot1200 support was removed. 
I’m researching right now how to compile a 4.2 build with dshot1200, or how to compile a 3.5.1 build for my PyroF7, but since English is not my mothertongue, that I can’t find any infos in French and my coding skills are limited to typing CLI commands, I figure I came here directly.

By the way, why remove Dshot1200 and keep Proshot1000, that I never saw, heard of or read about? Who knows what it is and/or is actually using it?

I’m kinda begging here, but I just bought 3 full stacks, build one new quad and am rebuilding the 2 that blew the electronics when I tried RPM filter, I can’t just let them feel sloppy (at least to me). I fly for more than 5 years now, I know my gear and right now I don’t like how it feels and I can’t go back, so I’m here pushing to bring it back.

Or maybe If a good soul can show me how to add Dshot1200 back in a custom build?

Thanks for your time!

etracer65, 2020-07-18:
DSHOT1200 brings absolutely no benefit over DSHOT600 with our maximum 8K PID loop. There is no reason to have DSHOT1200 so it was removed. If you think that DSHOT1200 somehow improves flight performance then you're mistaken.

closed 2020-07-18 by etracer65

MB3Drift, 2020-07-18:
What I know is that I can blind test dshot600 and 1200 and found which is which.
Like I said, the 4.2 flies objectively better, handle way better propwash on my 2 years old top notch build, but the new 4.2 build feel sloppier than my 3.5.1 old build.
And even on 3.5.1, dshot 600 feels sloppier than 1200.
I fly dshot 1200 since the first blheli32 ESC came out, I now my gear.
It may not improve performance but feeling is better on dshot1200

MB3Drift, 2020-07-18:
That said, does it mean than oneshot125 and oneshot42 have a benefit over dshot1200?

mikeller, 2020-07-19:
@MB3Drift: 

> What I know is that I can blind test dshot600 and 1200 and found which is which.
Like I said, the 4.2 flies objectively better, handle way better propwash on my 2 years old top notch build, but the new 4.2 build feel sloppier than my 3.5.1 old build.
And even on 3.5.1, dshot 600 feels sloppier than 1200.

The most prominent difference between Dshot600 and Dshot1200 is that with BLHeli_32 ESCs Dshot1200 has significantly more frame errors - if you prefer the 'feel' of Dshot 1200 then this is probably due to the increased latency resulting from the missed updates - you should be able to re-create the feel of Dshot1200 by soldering some small capacitors across the Dshot signal lines to re-introduce these frame errors.