betaflight/betaflight #10030: iFlight SucceX-D F7 TwinG flight controller not connecting to computer

dobralaszloedgar, 2020-07-18:
My flight controller isn't connecting properly to my Windows 10 PC. I tried ImpulseRC driver fixer, but it is only searching for flight controller and not finding it. If I open my device manager it says that it is an unknown USB Device (Device Descriptor Request Failed). I also have the correct drivers from the Betaflight welcome page. This thing happened when I tried to update my firmware and didn't succeed. Any help?

haslinghuis, 2020-07-18:
Hold the boot button before connecting USB and you can will enter DFU mode. Now please flash the right target.

dobralaszloedgar, 2020-07-18:
I tried it, but whatever I do the red light is constantly on. And it is still not working.

dobralaszloedgar, 2020-07-18:
And it also doesn't show any com ports

haslinghuis, 2020-07-18:
Because you bricked the FC with wrong flash it can only connect using the boatloader. To be able to flash the FC you will have to disconnect all cables, press and hold the boot button on the FC and connect the USB from FC to PC. After releasing button you should be in DFU mode and should be able to flash the right target IFLIGHT_F722_TWING. After that the COM ports will appear (or now is the time to use the driver fix - but not before having fixed the flash) and you can configure your settings.

etracer65, 2020-07-18:
@haslinghuis The flight controller can't be "bricked".

@dobralaszloedgar 99.9% of the time it's just a bad cable. You need to make sure you're using a real USB data cable and not a cheap charging-only cable. Try different cables.

etracer65, 2020-07-18:
Also this is not the correct place to seek support for these kinds of problems. This is the place to report bugs with the firmware. Please try an alternate source like:

Facebook Group: https://www.facebook.com/groups/betaflightgroup/
Betaflight Slack: http://slack.betaflight.com
RCGroups Thread: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/

closed 2020-07-18 by etracer65