betaflight/betaflight #10032: Show DJI OSD elements in OSD Preview screen

CecilFF4, 2020-07-19:
In the configurator, the preview screen that contains the OSD elements would be easier to modify/align for those with DJI FPV Goggles if the DJI OSD elements could be displayed within that preview image.

etracer65, 2020-07-19:
Not possible. The DJI air unit doesn't communicate anything back to Betaflight.

CecilFF4, 2020-07-19:
Sorry for the misunderstanding, but I wasn't meaning anything like that. I mean in the configurator GUI, some option to put up on the preview screen generic DJI OSD elements that are in their proper locations.

etracer65, 2020-07-20:
No, I fully understood. But the answer is still the same. DJI doesn't communicate anything to Betaflight so we don't know anything about these elements. Which are enabled, where they're positioned, etc. And we're surely not going to be forced into flight control or Configurator changes anytime DJI updates their firmware and potentially alters these elements.

If DJI followed some open-source standard it might be a consideration. But implementing a one-off solution for one proprietary product is not viable.

CecilFF4, 2020-07-20:
Ah, I understand now. Thank you.

closed 2020-07-20 by CecilFF4