betaflight/betaflight #10035: [RFI] FFT element in Betaflight OSD

zehks, 2020-07-21:
Currently there's an option for the FFT be shown in the final stats of a flight. Although this is very useful, it could be more interesting if it was seen "in realtime" as an OSD item. 

What's more, if it was available as four separated items (one for each motor) it would be super useful to detect hardware failure in a faster and easiest way rather than inspecting and interpreting a .BBL log after flight. 

If you saw an odd FFT on one motor during flight then there isn't any need to get a blackbox log as you could detect a bad motor/propeller/anything that's causing a motor to perform worse by seeing that FFT during flight.

etracer65, 2020-07-21:
The FFT used in the dynamic filter is based on each axis of the gyro - not each motor. There is no way to correlate this with a motor.

Victek, 2020-07-21:
We were thinking how to check weird motor behaviour that could be feed by vibrations or bad motor performance, then if we'll be able to compare motor feedback to settings we're wondering something went wrong. It's the background issue. Any thought?, thanks.

closed 2020-08-29 by stale[bot]