betaflight/betaflight #10037: Failsafe cant be deactivated with switch in the aux mode setting

vaneulinh, 2020-07-22:
Hi when I flick switch failsafe to return to home, and flick back it has no cancellation input (it ignore my SWITCH command) to STOP the failsafe, it wont respond at all 

"I have no control over my sticks" 

This does NOT happen with aux 2 channel "gps rescue" in the mode settings of betaflight
 i switch and flick switch again back in order to CANCEL "RTH," it responds to my control, no problem so incase something goes wrong

But when I put on aux 2 failsafe mode of betaflight to perform a gps rescue it activate and when it flew to home the wrong direction I tried to cancel it  but it ignored my command

after taking it for a flight, and activate switch
i cant even CANCEL the failsafe which is set to gps home rescue in betaflight, THIS CAN CAUSE A FLYAWAWAY, THE FLICK SWITCH to cancel failsafe DOES NOT RESPOND

it scared the hell out of me!!! it wont cancel?))

etracer65, 2020-07-22:
Your issues will continue to be closed if you don't provide the information requested when you opened the issue. And also read the documentation related to GPS Rescue in the Betaflight wiki.

vaneulinh, 2020-07-22:
what information sir, I dont have it

etracer65, 2020-07-22:
If you'd read the information when you created the issue, you would have seen:

> **Describe the bug**
> A clear and concise description of what the bug is.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> **Expected behavior**
> A clear and concise description of what you expected to happen.
> 
> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
> 
> **Setup / Versions**
>  - Flight controller [what type is it, where was it bought from];
>  - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
>  - how are the different components wired up.
> 
> **Additional context**
> Add any other context about the problem here.

vaneulinh, 2020-07-22:
thankx

pevecyan, 2020-07-22:
Have you kept throttle raised when disabling failsafe switch.
Afaik, GPS rescue will not let you take over controls if sticks are more than 30% deflected.

etracer65, 2020-07-22:
Please read the documentation: https://github.com/betaflight/betaflight/wiki/GPS-Rescue-Mode#important-be-ready-to-deactivate-the-mode-and-take-back-control-if-your-quad-does-not-point-towards-you-and-starts-making-its-way-home

vaneulinh, 2020-07-22:
no sir, I just have no control at all, its scary

this never happen when I activate gps rescue

etracer65, 2020-07-22:
Read and understand the documentation before you use features: https://github.com/betaflight/betaflight/wiki/GPS-Rescue-Mode#important-be-ready-to-deactivate-the-mode-and-take-back-control-if-your-quad-does-not-point-towards-you-and-starts-making-its-way-home

> v4.1 and beyond - After the RC link is restored, Betaflight now requires that any combination of roll/pitch/yaw inputs be moved beyond the `failsafe_stick_threshold` percentage value before failsafe will be exited and control returned to the pilot. The default for this setting is 30 which means you need to move roll, pitch, or yaw at least 30% off center before it will cancel Failsafe and return control. You may choose to reduce that setting value if you need a less abrupt transition from failsafe to full control.

closed 2020-07-22 by etracer65

vaneulinh, 2020-07-22:
thank u sir
30% off center before it will cancel Failsafe and return control  t

i can circle around the center 30% off to regain control

i tried to disarm during failsafe and it ignores my switch, is this normal
sir?

vaneulinh, 2020-07-27:
+ Please integrate Google map on OSD

+ Please add "ready to fly"(RTF) NOTIFICATION when GPS lock is ready
and-home point is up date when 11 satellite  is locked available this is
beneficial to avoid flyaway like a DJI product,

+ please add this to GPS rescue feature

+ Please add drone indicator,  ok so we know which way the drone is heading
like DJI displays on map

+ Please add return to Home pathway line Marker guidance, this gives us an
idea how the drone will return to home in the event of Fly Away or return
to home idea picture so it does not Fly Away, like a dji product shown on
the map