betaflight/betaflight #10039: VTX stuck in pit mode

whasianben, 2020-07-24:
**Describe the bug**
On the VTX page in betaflight, the pit mode switch is permanently switched on, I am unable to turn it off and it shows in the OSD.

**To Reproduce**
Set up the VTX table and press save. The pit mode switch automatically turns on for me.

**Expected behavior**
I should be able to toggle the VTX pit mode on and off at will.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# name: Apex

# resources
resource MOTOR 1 C09
resource MOTOR 2 A09
resource MOTOR 3 C08
resource MOTOR 4 A08
resource SERIAL_RX 3 NONE
resource PINIO 1 B11

# feature
feature -RX_PARALLEL_PWM
feature INFLIGHT_ACC_CAL
feature ESC_SENSOR

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 16384 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 0 900 1300 0 0
aux 2 35 1 900 1300 0 0
aux 3 40 1 1700 2100 0 0

# rxrange
rxrange 0 1159 1841
rxrange 1 1159 1841
rxrange 2 1159 1841
rxrange 3 1159 1841

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 200 400 600
vtxtable powerlabels 25 200 400 800

# master
set gyro_lowpass2_hz = 500
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 400
set dyn_lpf_gyro_max_hz = 1000
set acc_calibration = -45,3,-56,1
set rssi_channel = 8
set serialrx_provider = SRXL
set blackbox_device = NONE
set dshot_bidir = ON
set ibata_scale = 102
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_rssi = ON
set osd_vbat_pos = 2433
set osd_rssi_pos = 2401
set osd_tim_2_pos = 2454
set osd_vtx_channel_pos = 2113
set osd_current_pos = 2446
set osd_mah_drawn_pos = 2440
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 385
set osd_flip_arrow_pos = 2254
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5800
set vcd_video_system = NTSC
set pinio_box = 40,9,9,9
set gyro_1_align_yaw = 1800
set name = Apex

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 140
set dyn_lpf_dterm_max_hz = 340
set dterm_lowpass2_hz = 300
set vbat_pid_gain = ON
set iterm_rotation = ON
set iterm_relax_cutoff = 10
set p_pitch = 94
set i_pitch = 108
set d_pitch = 101
set f_pitch = 172
set p_roll = 85
set i_roll = 102
set d_roll = 92
set f_roll = 162
set p_yaw = 92
set i_yaw = 108
set f_yaw = 162
set d_min_roll = 61
set d_min_pitch = 66
set auto_profile_cell_count = 4

rateprofile 1

# rateprofile 1
set rates_type = KISS
set roll_rc_rate = 198
set pitch_rc_rate = 198
set yaw_rc_rate = 165
set roll_expo = 21
set pitch_expo = 21
set yaw_expo = 21
set roll_srate = 64
set pitch_srate = 64
set yaw_srate = 64

# end the command batch
batch end

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: MOTOR 2
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: FREE
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: FREE
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: PINIO 1
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: MOTOR 3
C09: MOTOR 1
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

**Setup / Versions**
 - Mamba F722S
 - Matek VTX-mini
 - VTX control wire hooked up to TX3 / PINIO hooked up to RX3 (pin was changed in CLI)

**Additional context**
I can't change it manually on the VTX either.

etracer65, 2020-07-24:
> **Additional context**
> I can't change it manually on the VTX either.

Then it seems like you should contact the manufacturer or read the manual for the vtx.

> **Important: This is not the right place to report faulty / broken hardware. Your issue will be closed without further comment.**

closed 2020-07-24 by etracer65

badgyro, 2021-03-04:
For people looking for help on google:
Update your Matek VTX mini firmware. I had the same problem, new firmware fixes it.