betaflight/betaflight #10040: GPS indicates its fixed but Betaflight shows nothing

MagicalTurtle, 2020-07-24:
**Describe the bug**
Betaflight shows no GPS information even though the GPS is blinking and indicating it's working.

**To Reproduce**
1. Connect drone to computer and battery
2. Set GPS as sensor input in Ports tab
3. Enable GPS in Config tab

**Expected behavior**
Betaflight should be giving GPS data.

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: 310c299f, date: 2020-05-22T01:08:06Z

# start the command batch
batch start

board_name MATEKF405STD
manufacturer_id MTKS

# name: EugeneHaroldCrab

# resources
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C07

# feature
feature GPS
feature RSSI_ADC

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RETA1234

# serial
serial 0 2 115200 115200 0 115200

# aux
aux 0 0 0 1875 2100 0 0
aux 1 1 1 1350 1650 0 0
aux 2 13 2 1825 2100 0 0
aux 3 35 1 1850 2100 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# master
set acc_calibration = 6,116,103,1
set min_check = 1010
set max_check = 1990
set rc_smoothing_derivative_type = BIQUAD
set blackbox_p_ratio = 64
set min_throttle = 1073
set motor_pwm_protocol = DSHOT600
set current_meter = NONE
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set osd_warn_batt_not_full = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 40
set osd_cap_alarm = 5000
set osd_vbat_pos = 88
set osd_rssi_pos = 2200
set osd_link_quality_pos = 122
set osd_flymode_pos = 14785
set osd_crosshairs_pos = 269
set osd_ah_sbar_pos = 2318
set osd_ah_pos = 2190
set osd_gps_speed_pos = 2519
set osd_gps_sats_pos = 2177
set osd_home_dir_pos = 2191
set osd_compass_bar_pos = 2155
set osd_altitude_pos = 2145
set osd_warnings_pos = 14761
set osd_avg_cell_voltage_pos = 2166
set osd_battery_usage_pos = 49
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set name = EugeneHaroldCrab

profile 2

rateprofile 2

# end the command batch
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: ADC_RSSI
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: FREE
C05: ADC_BATT
C06: MOTOR 3
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 2
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 3
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 4
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 -  Eachine GPS F4 Flight Controller;
 - Beitian BN-220T GPS Module;
 - how are the different components wired up.

etracer65, 2020-07-24:
If the GPS icon at the top is not on, or the GPS tab is not showing any satellites or a 3D fix then you either have a configuration problem, the GPS wired incorrectly, or a failed GPS module. The most common mistake is that the Rx/TX to the GPS are connected incorrectly and the RX must go to the TX on the flight controller and so on.

jyunte, 2020-07-27:
I have regularly seen the same issue... Plug in the battery at the flying site and wait. The GPS gets a fix and you fly the pack. Switch out batteries and while the LEDs on the GPS are blinking correctly, the BF OSD display for Satellites is completely blank. That is, is not showing zero satellites, it's just not showing anything (no satellite icon and no number, just empty space). Unplugging and plugging the battery back in multiple times usually fixes the problem.

SJChannel, 2020-07-29:
I am seeing a similar issue on a quad with the DJI Air Unit that has the GPS info displayed in the OSD.  Sometimes (maybe 20% of the time), even though the GPS has a fix as indicated by its LEDs, nothing appears in the OSD.  And I do mean **nothing**.  When this problem manifests, the DJI OSD does not even show that quad's battery voltage, which is not a part of the "custom" OSD.  Furthermore, the vtx remains in the "locked" state at 25 mW.  Apparently, whatever is happening or failing to happen in Betaflight is confusing the Air Unit.  As @jyunte reported, the only workaround I've found is to unplug the battery and plug it back in.

This is on Betaflight 4.2.0 using the same GPS module and the same GPS configuration that works reliably for me on three other quads, all running Betaflight 4.1.x or 4.0.x.  All four of my GPS-equipped quads have variants of the Kakute F7 FC.

0crap, 2020-08-09:
@jyunte Well known issue.
https://github.com/betaflight/betaflight/issues/9652

SJChannel, 2020-08-09:
@0crap That's an interesting issue, but it doesn't apply in my case.  Autobaud for GPS is not enabled in my quads.

I will add that since reverting to BF 4.1.7 I have not seen the GPS vs. DJI OSD problems that I was seeing with BF 4.2.

SJChannel, 2020-08-14:
I have been looking into this issue, and I have some new information.  I modified Betaflight to add a new debug_mode "DJI_OSD".  It packs some MSP and GPS information into the four "debug" fields in the Blackbox logs.  My initial goal was to be able to log the interactions between the DJI air unit and the associated MSP task on the flight controller.  You can find the changes in the "dji-osd-debug" branch of my repository [here](https://github.com/SJChannel/betaflight/tree/dji-osd-debug).  (These changes are not intended to be put into the master BF code.  Also, note that the UART used for MSP is hard-coded.)

I used Blackbox Explorer to convert my log files into CSV files, and wrote a small Python program to decode the new debugging information.  When the system is operating normally, we see the following output, repeated over and over throughout the flight:

```
  53.190 HF  17   1.17 <> MSP_FC_VERSION
  53.210 HF  17   1.17 <> MSP_NAME
  53.230 HF  17   1.17 <> MSP_OSD_CONFIG
  53.260 HF  17   1.17 <> MSP_FILTER_CONFIG
  53.290 HF  17   1.17 <> MSP_PID_ADVANCED
  53.300 HF  17   1.17 <> MSP_STATUS
  53.320 HF  17   1.17 <> MSP_RC
  53.340 HF  17   1.17 <> MSP_RAW_GPS
  53.360 HF  17   1.17 <> MSP_COMP_GPS
  53.390 HF  17   1.17 <> MSP_ATTITUDE
  53.420 HF  17   1.17 <> MSP_ALTITUDE
  53.430 HF  17   1.17 <> MSP_ANALOG
  53.450 HF  17   1.17 <> MSP_RC_TUNING
  53.480 HF  17   1.17 <> MSP_PID
  53.490 HF  17   1.17 <> MSP_BATTERY_STATE
  53.511 HF  17   1.17 <> MSP_ESC_SENSOR_DATA
  53.521 HF  17   1.17 <> MSP_STATUS_EX
  53.540 HF  17   1.17 <> MSP_RTC
-------------------
  53.550 HF  17   1.17 <> MSP_FC_VERSION
  53.571 HF  17   1.17 <> MSP_NAME
  53.590 HF  17   1.17 <> MSP_OSD_CONFIG
  53.610 HF  17   1.17 <> MSP_FILTER_CONFIG
  53.631 HF  17   1.17 <> MSP_PID_ADVANCED
  53.651 HF  17   1.17 <> MSP_STATUS
  53.670 HF  17   1.17 <> MSP_RC
  53.691 HF  17   1.17 <> MSP_RAW_GPS
  53.720 HF  17   1.17 <> MSP_COMP_GPS
  53.730 HF  17   1.17 <> MSP_ATTITUDE
  53.761 HF  17   1.17 <> MSP_ALTITUDE
  53.780 HF  17   1.17 <> MSP_ANALOG
  53.801 HF  17   1.17 <> MSP_RC_TUNING
  53.851 HF  17   1.17 <> MSP_PID
  53.881 HF  17   1.17 <> MSP_BATTERY_STATE
  53.891 HF  17   1.17 <> MSP_ESC_SENSOR_DATA
  53.910 HF  17   1.17 <> MSP_STATUS_EX
  53.921 HF  17   1.17 <> MSP_RTC
-------------------

```
The output fields are as follows:

1. Elapsed time in seconds
2. GPS flags: "H" = home fix, "F" = fix
3. Number of satellites
4. HDOP
5. "<" indicating a command to the MSP
6. ">" indicating that a response was sent.  This would be "!" in case of an error respose, or "-" if no response.
7. The MSP command

In normal operation, the air unit sends these same 18 commands repeatedly to get the information needed to update its OSD.

I flew enough packs to experience a couple of different kinds of failure.  For one of the packs, the custom OSD fields were entirely missing from the screen, and the battery voltage was shown as "NA".  I armed the quad for a while in order to get a Blackbox log.  The log showed that there was **NO** communication between the air unit and the MSP _at all._  I have no way of knowing what happened before I armed the quad, since the BB log is only generated when the quad is armed.

On three other packs (out of eight total), all or part of the OSD froze and stopped updating at some point in the flight.  In some cases I would get widely-spaced sporadic updates of some portions of the OSD.  Here's what the log shows at the point where the OSD froze:

```
 283.831 HF  20   1.06 <> MSP_FC_VERSION
 283.851 HF  20   1.06 <> MSP_NAME
 283.861 HF  20   1.06 <> MSP_OSD_CONFIG
 283.892 HF  20   1.06 <> MSP_FILTER_CONFIG
 283.901 HF  20   1.06 <> MSP_PID_ADVANCED
 283.921 HF  20   1.06 <> MSP_STATUS
 283.952 HF  20   1.06 <> MSP_RC
 283.971 HF  20   1.06 <> MSP_RAW_GPS
[... 45 lines cut ...]
 284.884 HF  20   1.06 <> MSP_PID_ADVANCED
 284.893 HF  20   1.06 <> MSP_STATUS
 284.914 HF  20   1.06 <> MSP_RC
 284.944 HF  20   1.06 <> MSP_RAW_GPS
 284.964 HF  20   1.06 <> MSP_COMP_GPS
-------------------

```
The first set of commands shown was the usual group of 18.  But after that, the set of commands was truncated after a varying number of commands.  This continued to the end of the flight.  In at least one case, commands were missing from the _middle_ of a group:

```
 362.360 HF  17   1.13 <> MSP_FC_VERSION
 362.370 HF  17   1.13 <> MSP_NAME
 362.401 HF  17   1.13 <> MSP_OSD_CONFIG
 362.420 HF  17   1.13 <> MSP_FILTER_CONFIG
 362.440 HF  17   1.13 <> MSP_PID_ADVANCED
 362.471 HF  17   1.13 <> MSP_STATUS
 362.490 HF  17   1.13 <> MSP_RC
 362.521 HF  17   1.13 <> MSP_RAW_GPS
 362.540 HF  17   1.13 <> MSP_COMP_GPS
 362.560 HF  17   1.13 <> MSP_RTC
-------------------

```
My logging doesn't have much information, since the debug fields total 8 bytes.  But I do not see anything obvious that the FC is doing that would cause the sequence of commands from the air unit to change.  I'm not sure where to go with it from this point.

SJChannel, 2020-08-14:
Here's the quick & dirty Python program I used to dump the logs from the CSV files.

[mspdump.py.txt](https://github.com/betaflight/betaflight/files/5072052/mspdump.py.txt)
[msp_names.py.txt](https://github.com/betaflight/betaflight/files/5072053/msp_names.py.txt)

manc01, 2020-09-02:
Ok, I am having the same issue with a GepRc CineRun HD 3 with an "EXF722DUAL" flight controller. I definitely have the wiring correct. I have switched the RX and TX etc and tried different UARTs to no avail. Tried everything that I can think of, used different wires, resoldered etc but no luck.

I am using a Matek M8Q-5883 GPS which I know is fine as I have tested it on another quad. When connected to the quad with the problem I can get the GPS to gain a fix but in betaflight the GPS icon is not showing up. This is extremely frustrating and I do not know how to fix it. 

I wondered if there is such a thing as a compatibility issue with the board and the GPS and Matek said that they had not come across this issue. Is it a betaflight issue or could GepRC do something to help? I did contact them and had no luck. Any help would really be appreciated.

SJChannel, 2020-09-02:
Removing the SD card from the Air Unit completely eliminates the problem for me.  My SD card is a SanDisk Extreme V30, which is on DJI's list of approved SD cards.

Of course, flying without the SD card is not a reasonable solution.

manc01, 2020-09-02:
I am new to this forum, is it likely that betaflight will acknowledge this bug and be able to fix it? If so I wonder how long a fix could take, months? I am so glad that I found this thread, I was going crazy because I had troubleshot it enough to know that it had to be right.

SJChannel, 2020-09-02:
@manc01 I'm 99% sure it's a DJI bug, not a Betaflight bug.

manc01, 2020-09-03:
> I'm 99% sure it's a DJI bug, not a Betaflight bug.

I tried taking the SD card out but it was still the same. Full satellite lock but nothing in the betaflight configurator unfortunately. Surely there must be quite a few other people with this issue? At least I can stop messing around now knowing it is a bug. Hopefully betaflight can fix it as I can't see DJI being interested at all. What are the chances that they will look at it?

On a side note, I wish that the betaflight OSD direction home arrow pointed towards you rather than away. To me it seems counter-intuitive. I would love to see this changed or at least an option to point it in whatever direction you want.

jyunte, 2020-09-03:
Not sure what you mean about the RTH arrow, it points to wherever the home position is... At least it does on non-DJI builds. If BF is not showing the GPS symbol (at the top of the screen) as active, then maybe it's incorrectly wired to the UART (TX to RX and vice versa), or is not getting power (some flight controllers require a battery to power the GPS, some will power it from USB).

jyunte, 2020-09-03:
Are you sure you're not seeing a direction of travel indicator, rather than direction to home? I've not heard anyone else state that the arrow points 180 degrees from the home point.

etracer65, 2020-09-03:
The arrow pointing "up" means the home location is directly in front of you and you're heading towards it based on your current heading. The arrow pointing "down" means the home location is behind you and you're moving away from it.

manc01, 2020-09-04:
> The arrow pointing "up" means the home location is directly in front of you and you're heading towards it based on your current heading. The arrow pointing "down" means the home location is behind you and you're moving away from it.

Yes that is correct I personally just feel that it should have the arrows in the opposite direction.

0crap, 2020-09-04:
> Yes that is correct I personally just feel that it should have the arrows in the opposite direction.

Just think of it that the arrow is always pointing to where you're standing.
So always pointing to the home location.
This way it's totally clear and should remove your confusion.

0crap, 2020-09-04:
> I am using a Matek M8Q-5883 GPS which I know is fine as I have tested it on another quad. When connected to the quad with the problem I can get the GPS to gain a fix but in betaflight the GPS icon is not showing up. This is extremely frustrating and I do not know how to fix it.
> 

The native baud setting for that GPS is 9600 baud.
In the BF configurator set the UART to 9600. (The UART connected to your GPS.)
Also disable AUTO BAUD in the GPS tab of the BF configurator.

Does this fix your issue?

manc01, 2020-09-06:
> The native baud setting for that GPS is 9600 baud.
> In the BF configurator Ports tab set the UART to 9600. (The UART connected to your GPS.)
> Also disable AUTO BAUD in the Configuration -> GPS tab of the BF configurator.
> 
> Does this fix your issue?

I did try this and unfortunately it did not resolve my issue. I emailed Matek and they suggested 57600 with auto baud and auto config ticked. That did not work for me either.

manc01, 2020-09-06:
I emailed this link to Lee Schofield who has the Painless 360 channel on YouTube. He read over this and he thinks that it is a bug that hopefully can be resolved in an update. Fingers crossed!

manc01, 2020-10-11:
I was hoping that the last update would have resolved the GPS issue of not being detected in betaflight but it still persists. There is also a discussion within the Team Black Sheep lounge where other people are experiencing this issue. Is it going to be looked at?

mikeller, 2020-10-12:
@manc01: This is a community driven open source project - feel free to look at it.

closed 2021-02-08 by github-actions[bot]

Killerbeans, 2024-11-17:
Please check the wiring of the GPS to FC. Most wiring diagrams of the GPS to the FC online do not take into account that the color codes are not the same. SCL, SDA, Tx, and Rx should be checked thoroughly not just by cable color. For example, the color codes on this wiring diagram are similar to the color codes on the M100-5883 cable but don't match the pins on the GPS. Meaning if you wire the colors on the GPS  cable according to the pads of the FC according to this image online, you would get it wrong. The only correct cables in the wiring diagram below are the yellow (GPS TX which is the reason why you can see satellites but BF can't see the GPS), red and black
[image]

dannydeafo3-beep, 2026-05-18:
THIS IS STILL HAPPENING TO MY m100 PRO.  fOUND A VIDEO THAT THE BAUD RATE IS HIGHER THAN BETA FLIGHT WOULD ACCEPT IS THAT TRUE?  i HONESTLY WISH I WOULD HAVE BOUGHT A DJI DRONE AS BETA FLIGHT IS A PILE OF CRAP!!  THIS IS WHY YOU DONT CALL IT "PRODUCTION FLIGHT" BECAUSE ALL THE BUGS AND ISSUES WITH THIS SOFTWARE!!