betaflight/betaflight #10041: DPS310 on F405 FC doesn't work with latest 4.3.0

MATEKSYS, 2020-07-24:
DPS310 on F405-SE/F405-WING doesn't work with latest 4.3.0,  same issue on F405-STD after replacing onboard BMP280 with DPS310.
DPS310 on F405-SE/F405-WING is verified to work with INAV/Ardupilot.
DPS310 on F722/F765/H743 series FC is verified to work with latest 4.3.0

https://github.com/betaflight/betaflight/pull/9957

mikeller, 2020-07-24:
According to @jflyper the DPS310 was tested over I2C and SPI and was working (https://github.com/betaflight/betaflight/pull/9957#discussion_r446660719). Maybe if you provide test hardware we can investigate what isn't working in your case.

jflyper, 2020-07-24:
I'm working on this.

Mayk-CLP, 2020-07-26:
but it would be fair if they provide some testing Hardware for you guys.

mikeller, 2020-07-27:
@Mayk-CLP:

> but it would be fair if they provide some testing Hardware for you guys.

To be fair, @MATEKSYS normally does. :-)

Mayk-CLP, 2020-07-27:
That sounds great

closed 2020-09-05 by stale[bot]

haslinghuis, 2020-12-26:
Should add a `bug` label here

mikeller, 2020-12-26:
This seems to have slipped off the radar. @MATEKSYS, can you confirm that this is still an issue with latest 4.3 debug builds?
If so, can you please send me a engineering sample board with DPS310?
@jflyper: Any updates on this?

MATEKSYS, 2020-12-27:
@mikeller  It is still an issue.    We will send you a sample to check.

coryjfowler, 2021-02-12:
I am not sure if this issue with the DPS310 is related to the F405 specifically even though MATEKSYS says it was working with their  H743 boards last July.  I have a with a Nucleo H743 with a DPS310 connected and I have not been able to use the DPS310 after enabling it in the code, building, and flashing.

mikeller, 2021-02-18:
Ok, some insights - looks like the reason this is not working whereas the DPS310 is working on other boards is that in this case it is connected via I2C - looks like reading the compensation coefficients in this way is failing over I2C:

```
    uint8_t coef[18];
    if (!busReadBuf(busDev, DPS310_REG_COEF, coef, sizeof(coef))) {
        return false;
    }
```

The suspicion is that the bulk read is too fast for the device to handle. As a test, the following will succeed and the barometer will work as expected:

```
    uint8_t coef[18];
    if (!busReadBuf(busDev, DPS310_REG_COEF, coef, 13)) {
        return false;
    }
    if (!busReadBuf(busDev, DPS310_REG_COEF + 13, coef + 13, 5)) {
        return false;
    }
```

Whereas this will fail:

```
    uint8_t coef[18];
    if (!busReadBuf(busDev, DPS310_REG_COEF, coef, 14)) {
        return false;
    }
    if (!busReadBuf(busDev, DPS310_REG_COEF + 14, coef + 14, 4)) {
        return false;
    }
```

The I2C stack in Betaflight does not have a way to set the speed for individual drivers, so not sure what the best way is to deal with this:
- reduce the speed for all I2C devices (seems wasteful)
- split the reading of coefficients into 2 parts for this device (pragmatic and ugly, but works and does not affect other devices)
- rewrite the I2C stack (out of scope for this fix really)

PR #10578 (2021-03-28): Fixed initialisation of DPS310 when used over I2C.
Fixes #10041.