betaflight/betaflight #10042: Port and Merge Triflight Improved Tricopter Control

SJ-Innovation, 2020-07-24:
As originally mentioned in #1241 I wish to port and update Triflight to get better Tricopter yaw control in Betaflight. 

This will involve porting Triflight to BF4.X (currently in progress), testing and getting it all proven.

My initial work is quite barebones, just getting the core and key modifications that Triflight made to work in BF4.3 before moving on to some of the more esoteric improvements that could be added (servo harmonic filtering etc)

mikeller, 2020-07-24:
@SJ-Innovation: Sounds like a good idea - preliminary work was done to facilitate this move in #4999, so you can probably take it from there.

SJ-Innovation, 2020-07-29:
Working on this is #10051. Though currently experiancing issues relating to #10060

GTorreil, 2020-07-31:
@SJ-Innovation This sounds great, I miss triflight's yaw control... I can test this if you need, my setup is a 5" tri with feedback-less servo. Just let me know.

SJ-Innovation, 2020-07-31:
@GTorreil That would be great. 
It'll be a little while yet due to having to work through a couple of blocking technical issues on this end, as well as the fact I'm moving house in a week... 

Out of curiosity, what board do you have? And are you capable of building yourself, or would you like me to provide you with a hex?

GTorreil, 2020-08-02:
Yes no problem :) good luck to you
i'm running an OmniF4. Last time I built betaflight was before the new "unified targets", but i'll certainly manage to do it again.

GTorreil, 2020-08-04:
@SJ-Innovation Just cloned your branch for checking, I can compile it for my FC :)

SJ-Innovation, 2020-08-04:
@GTorreil Excellent, please be aware that i have run _zero_ flight tests on this at the moment (due to my situation) and will not be able to for a little while. Therefore I cannot warrant _any_ of the Triflight code at this point. 

You do your tests _entirely_ at your own risk. 

I should be able to be considerably more active once my commitments have eased a little. Best of luck, if (when) you have issues be sure to let me know :)

GTorreil, 2020-08-05:
Aha yes of course, totally agreed. take care :)
I haven't tested yet either, but i'll tell you.

closed 2020-09-12 by stale[bot]