betaflight/betaflight #10049: gps support

vaneulinh, 2020-07-27:
+ Please integrate Google map on OSD

+ Please add "ready to fly"(RTF) NOTIFICATION when GPS lock is ready and-home point is up date when 11 satellite  is locked available this is beneficial to avoid flyaway like a DJI product, 

+ please add this to GPS rescue feature

+ Please add drone indicator,  ok so we know which way the drone is heading like DJI displays on map

+ Please add return to Home pathway line Marker guidance, this gives us an idea how the drone will return to home in the event of Fly Away or return to home idea picture so it does not Fly Away, like a dji product shown on the map

0crap, 2020-07-27:
Have a look at [iNav](https://github.com/iNavFlight/inav/wiki)
GPS functions like a DJI is not something you can expect on BF.

ATKCA, 2020-07-27:
Part of his suggestion has merit though. When using GPS failsafe and a GPS lock is not present ready to allow arming for takeoff then it would be useful to have a message on screen. Right now we just see a satellite count and you have to try to arm to find you cannot arm yet. It's confusing because the satellite count can be above the set value but arming still is not allowed because although satellites are detected they have a low signal and so are not counted as suitable for a lock.
It would only take a small adjustment to the warnings OSD element to add messages for "Poor GPS signal" , "GPS not ready" and "GPS ready".

vaneulinh, 2020-07-29:
I can still arm with min 5 satellites even though I set it 8 gps in cli,
bug?

vaneulinh, 2020-07-29:
Request

panic aux

please add panic aux mode in options,
incase, u need to pause flight, feel sick or received emergency phone call,
or have a stroke, the drone can autopilot by hover, by allowing the drone
to swing left right back and forward by half a second and raising the
altitude and lower altitude no more than 1 metre in half a sec repeatedly

Request

failsafe I have no control of yaw or pitch or roll
during  failsafe RTH, it needs full control Even when RTH when switch on,
this includes altitude control, pitch, roll and yaw

failsafe no control of yaw or pitch or roll ir altitude control
during RTH, it needs full control Even when RTH aux on  channel or auto
assign

failsafe can activate, but cannot abort SWITCH, even when moving joystick

(gps rescue and failsafe code input)

return to home if travel away more than 20 metre must abort codes

return to home if travel away more than 30 metre must abort coded

return to home if travel away more than 40 metre must abort coded

turn around craft if detect distance greater than home direction by 10
metre range except, altitude

turn around craft if detect distance greater than home direction by 20
metre range except, altitude

turn around craft if detect distance greater than home direction by 30
metre range, except altitude

turn around craft if detect distance greater than home direction by 100
metre range except, altitude

GPS RTH 8 satellite and under can cause a fly away advised to disable, as I
can fly with a min 6 satellite which is dangerous

add set home point manually on betaflight app as second backup for failsafe
and gps rescue

add remote controller home point

add remote controller home point, if RTH drone lost home point, it *will*
RTH to remote controller location

add remote controller home point, if RTH drone lost signal, it RTH to
remote controller location

a message  notifications RTH or if no GPS lock.......it returns to remote
controller distance

add set a 2nd home point manually on betaflight app as second backup for
failsafe and gps rescue incase it flies away (2x return to home backup)

homepoint and remote home must not be more than 30 metres Away to add

I designed the dji return to home feature

mikeller, 2020-07-29:
@vaneulinh: As @0crap says in https://github.com/betaflight/betaflight/issues/10049#issuecomment-664253617, have a look at iNav (https://github.com/iNavFlight/inav/wiki) - autonomous flight is not a focus area for Betaflight.

closed 2020-09-05 by stale[bot]