betaflight/betaflight #10050: No connection to the Tinyhawk 2

Frank417, 2020-07-27:
I entered and saved the commands "set usb_hid_cdc = on" via "Betaflight Configurator".
Later I wanted to put "OFF" again, but now the drone does not connect to "Betaflight". What can I do now?

Thanks from Duisburg

etracer65, 2020-07-27:
Try using another computer that has an operating system that supports connections from HID devices (like Windows 10 or MacOS) and turn the feature back off. Alternately refresh the firmware by holding the BOOT button (see the product manual) while connecting the USB. Then flash with a "full chip erase" and set up from scratch.

Frank417, 2020-07-27:
Thanks, it worked with Windows 10, conbected. -- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

closed 2020-07-27 by etracer65