betaflight/betaflight #10052: Betaflight - Flashed firmware to Tinyhawk 2, not detecting gyro, accelerometer, motors, etc.

D-Barr, 2020-07-27:
**Describe the bug**
Connect Tinyhawk 2 via USB from FC to laptop. COM3 appears and you can connect device, but once in betaflight no recognition of ESC and motor protocol.

Arming Disable Flags: | NO_GYRO
                                      BOOT_GRACE_TIME MSP
Battery voltage: | 0 V
Capacity drawn: | 0 mAh
Current draw: | 0.00 A
RSSI:

Plug in a battery and still no recognition of voltage, arming sounds or LED lights, nothing except a really hot circuit board!?!?!

**To Reproduce**
Flash firmware for tinyhawk 2

**Expected behavior**
I expected to flash the new 4.2.1 firmware as described in your website and have an operating micro drone.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
``` # diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F411 (S411) 4.2.1 Jul 19 2020 / 06:18:29 (caa0d683c) MSP API: 1.43
###ERROR: diff: NO CONFIG FOUND###
# start the command batch
batch start

# name: TinyHawk II

# feature
feature -RX_PARALLEL_PWM
feature OSD
feature RX_SPI

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# aux
aux 0 0 1 900 1075 0 0
aux 1 1 0 1900 2100 0 0
aux 2 2 0 1350 1650 0 0
aux 3 13 1 1400 1600 0 0

# master
set gyro_calib_duration = 0
###ERROR: diff: CORRUPTED CONFIG: gyro_calib_duration = 0###
set gyro_calib_noise_limit = 0
set gyro_overflow_detect = OFF
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set rc_smoothing_derivative_type = BIQUAD
set rc_smoothing_auto_smoothness = 0
set rx_spi_protocol = FRSKY_D
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT600
set motor_poles = 0
###ERROR: diff: CORRUPTED CONFIG: motor_poles = 0###
set vbat_max_cell_voltage = 435
set small_angle = 180
set gps_sbas_mode = AUTO
set pid_process_denom = 4
set gyro_1_sensor_align = DEFAULT
set gyro_2_sensor_align = DEFAULT
set name = TinyHawk II

profile 0

# profile 0
set p_pitch = 105
set i_pitch = 125
set d_pitch = 95
set p_roll = 105
set i_roll = 125
set d_roll = 95
set p_yaw = 95
set i_yaw = 95
set d_yaw = 45
set angle_level_strength = 60
set horizon_transition = 80

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_rc_rate = 80
set pitch_rc_rate = 0
###ERROR: diff: CORRUPTED CONFIG: pitch_rc_rate = 0###
set yaw_rc_rate = 80
set roll_expo = 60
set yaw_expo = 20
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 25
set tpa_breakpoint = 1650
set throttle_limit_percent = 0
###ERROR: diff: CORRUPTED CONFIG: throttle_limit_percent = 0###
set roll_rate_limit = 0
###ERROR: diff: CORRUPTED CONFIG: roll_rate_limit = 0###
set pitch_rate_limit = 0
###ERROR: diff: CORRUPTED CONFIG: pitch_rate_limit = 0###
set yaw_rate_limit = 0
###ERROR: diff: CORRUPTED CONFIG: yaw_rate_limit = 0###

# end the command batch
batch end ```

 ```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: FREE
A06: FREE
A07: FREE
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE ```

**Setup / Versions**
# version
# Betaflight / STM32F411 (S411) 4.2.1 Jul 19 2020 / 06:18:29 (caa0d683c) MSP API: 1.43
###ERROR: version: NO CONFIG FOUND###

 - how are the different components wired up.
Stock from manufacturer.

etracer65, 2020-07-27:
You flashed the incorrect target. See the manufacturer's documentation for which target to use for this quad.

D-Barr, 2020-07-28:
Thanks Bruce, Quad is back to normal,  but looks like manufacturer have used MATEKF411 and didn't add RX on the end. Lucky the drone community is out to help, and pointed me in the right direction.

closed 2020-07-28 by etracer65