betaflight/betaflight #10055: Betaflight 4.2.1 Caddx Vista Arming-BUG

luedi82, 2020-07-27:
Hello

I make a new drone-setup with holybro f7, a teko32 4in1 esc, a xm+ and a caddx vista.

I solder the power and ground from the vista directly to the battery. and the rx and tx from the vista to the rx2 and tx2 from the holybro f7 flight controller. 

After the hole wirering I make the update in betaflight to the latest firmeware.
Then I make the update from the vista in the dji assitent 2.

After that, I'v try the hole stack. The Problem was, I had conection from the vista to the google. I saw the image clearly and also the bateryvoltag, wich cames from the konfig-tab MSP on on UART2.

But after I made the drone arming, the conection was very bad and I cant fly.

After hours of searching the problem, I found out, that the my little one, also with the vista setup and where it works, i had betaflight 4.2.0 - so I downgrade the holybro f7 and now it works.

Please fix this BUG than I think more people than I will have the same problem.

Cheers and Greetings from
Switzerland

luedi82, 2020-07-28:
What information do you missing?

mikeller, 2020-07-28:
All the one you were asked to provide when opening this issue.

closed 2020-07-29 by no-response[bot]