betaflight/betaflight #10056: ESC-Telemetry Offset values for Amperage Meter (and Votage Meter as well)

wuzle, 2020-07-27:
When I use the ESC-Telemetry to read out Current and Voltage Data, I can't set an offset to Current and Voltage. Most ESCs don't provide a way to scale the Voltage and the current the ESCs read out are limited to the ESCs themselves, thus missing the "idle current" from VTX, RX, LEDs and so on.

**Describe the solution you'd like**
I would love to see an option to add an offset to the incoming ESC-Telemetry data regarding current and voltage, similar to how it is currently done with the "normal" current meter:
[image]

**Describe alternatives you've considered**
If it's too much work to also put it in the configurator it could just be a CLI setting for the current meter.

A lot of pilots find the current meters useless, but for me it's part of my analytics after flight. So, to get a better overview I'd love to be able to add those .26A as an offset. :)

etracer65, 2020-07-27:
The calibration is configured in the BLHeli32 firmware. Use the BLHeli32 configuration tool.

closed 2020-07-27 by etracer65

wuzle, 2020-07-27:
But That's the point. You can't calibrate an offset for the current, just a scale:
[image]
Also it would be more precise to have the peripheral current from vtx, LEDs and so on on top and not bundled into one to four ESC's Current readings.

Also I am not necessarily only talking about BLHeli32, but also Fettec ESCs.

Well, you can't offset any of those anyways.

wuzle, 2020-08-02:
I looked through the files and there even is an offset in there:
[image]
now I only need to be able to set it ... there's something missing for it to work.

I'll figure this out ^^

haslinghuis, 2020-08-02:
Did some digging. There seems to be an `esc_sensor_current_offset` setting in cli to configure this offset.

wuzle, 2020-08-02:
Now that's a bit embarrassing for me now. Well. At least I learned how to setup the building environment...

I'll fiddle a bit around with it and maybe also try to build the configurator. Maybe some day I can be of help and implement small things somewhere. Like that little input box for the offset in the power tab.

I'll test the `esc_sensor_current_offset` tomorrow.

mikeller, 2020-08-03:
I thought I'd added this offset while ago... :thinking:

wuzle, 2020-08-04:
(just to have this concluded: `esc_sensor_current_offset` works as intended, thanks again and sorry to have bothered you all)

haslinghuis, 2020-08-04:
Don't worry.