betaflight/betaflight #10058: Apply Manufacture ID

shackelton1978, 2020-07-28:
We a new hardware producing factory in shenzhen,china.
We need apply the manufacture id for our own hardware board.
The name of the company selling the board and supporting customers of it is :CODDAR LIMITED
The website URL for this company is :www.coddar.net
Board_name: CODDAR_F722DUAL
manufacturer_id: CODDAR

mikeller, 2020-07-28:
@shackelton1978: Your website seems to be having a problem:

[image]

Can you please let me know when this is fixed?

shackelton1978, 2020-08-01:
@Mikeller:
 The website is OKAy.Thanks for your help.

closed 2020-08-02 by mikeller

shackelton1978, 2020-08-02:
Mikeller :
 The Brand is CODDAR,is not CODAR.

mikeller, 2020-08-03:
@shackelton1978: Sorry, got that wrong. Fixed the pull request title. Can you please confirm that you are the maintainer for this manufacturer's targets?

shackelton1978, 2020-08-03:
Mikeller:
 Yes， I am the R&D  of CODDAR Brand.

shackelton1978, 2020-08-03:
Mikeller:
 Would please tell me what's the Manufacture ID for CODDAR ? and I will push the configuration file .

haslinghuis, 2020-08-03:
https://github.com/betaflight/unified-targets/blob/master/Manufacturers.md
```
HEBI | Shenzhen High Energy Bili Co., Ltd. | http://www.coddar.net/
```

shackelton1978, 2020-08-03:
haslinghuis:
Thanks for your help.

shackelton1978, 2020-08-03:
> Mikeller:
> Yes， I am the R&D of CODDAR Brand.
The Manufacture ID we want using CODDAR,Would you  please help us to modify it ?
 The CODDAR is the Brand name we will promition .
Thanks !

shackelton1978, 2020-08-04:
Dear mikeller:
 We are using the following information ,i am sorry for our mistake to trouble you again.
Brand Name is :CODDAR
THe webiste is :www.coddar.net
The Company name is :ShenZhen GNBL CO.LTD
So please help to mdoify the manufacture id.
Thanks for your help.
James Lee

shackelton1978, 2020-08-05:
Dear Mikeller:
 Would you please modfiy the Manufacture ID for CODDAR as following:
CODDAR
And the company is :ShenZhen GNBL CO.,LTD, The website is :www.coddar.net
Thanks very much.
James Lee

haslinghuis, 2020-08-06:
Don't know if @mikeller get's notified without using @

shackelton1978, 2020-08-06:
> Can you please let me know when this is fixed?

Dear Mikeller:
Would you please modfiy the Manufacture ID for CODDAR as following:
CODDAR
And the company is :ShenZhen GNBL CO.,LTD, The website is :www.coddar.net
Thanks very much.
James Lee

mikeller, 2020-08-07:
@shackelton1978: If you do not stop flooding this repository with requests just because they are not actioned immediately you will be banned and your manufacturer id removed.

shackelton1978, 2020-08-07:
mikeller:
 Thanks for your reminding.
 I apologize for my behavior to you .
Thank you. I won't keep posting like this
James Lee