betaflight/betaflight #10060: [F7] Linker Problem: Excess code from .text section ending up in .tcm_code

SJ-Innovation, 2020-07-29:
Currently Investigating.

F7 targets with linker scripts that include the use of the ITCM RAM are including excess code that is not qualified with FAST_CODE (or any other section attribute) in ITCM RAM. This means it is highly unpredictable whether or not code will end up in the ITCM RAM. This is leading to issues with exceeding the available ITCM RAM space with code that should not be in there. 

Also, as quite a lot of code is qualified as FAST_CODE, it is copied (as expected) into ITCM. The other issue is that ITCM is only 16k... We are effectively limited to 16k of program space by the linker trying to squeeze most of the code into the ITCM.

This is not happening on F4 targets as they do not have a smaller ITCM ram section.

SJ-Innovation, 2020-07-29:
We probably need to have a sort out and consoliate the "performant code" qualifiers. 

I notice there are several including:
`FAST_CODE` & `CCM_CODE` which are used often interchangably by different F4/7  and F3 targets respectively to mean the same thing: "Put this code in the fastest exec place we can"

I propose the following consoliation:
FAST_CODE - Fastest code exec location on the target, also consolidated with those that use CCM_CODE.
SLOW_CODE - Code that can deliberately be offloaded (if the target supports it) to larger, but slower memory. (Im thinking good for initialisation and human-speed code)

As at the moment there is a *lot* of initialisation code taking up extremely precious ITCM memory. This could also unlock

mikeller, 2020-07-29:
Question: What would `SLOW_CODE` actually do, how would it do this, and how would this be different from not-attributed code?

I think the system is actually working the way it is intended, and it is working to give us what we need in a way that is probably close to how well it can be - it is just the case that 'as well as it can be' is not anywhere close to 'easy to use'.
The reality is that with respect to assigning code to different memory regions, we are fighting one important approach that the compiler takes to optimise code for speed - aggressive inlining. Especially for code that we want to be optimised (as defined in the Makefile at file level), we do want everything to be inlined as much as can be, to avoid expensive JSR calls.
But at the same time this means that pretty much all function calls inside a `FAST_CODE` function will be inlined, with the called functions' code ending up in FAST RAM as well - this is by design. The 'antidote' for this behaviour is to use `FAST_CODE_NOINLINE` on functions that are called by optimised code that should _not_ be optimised - this will suppress inlining, resulting in the function's code ending up in normal memory. (The attribute name is a bit of a misnomer, but we've never managed to come up with a better one.)
With respect to initialisation code, there is another aspect - as stated, compiler optimisation is per file, and a lot of legacy Betaflight code combines initialisation code and runtime code in the same files. This has the effect that initialisation code that is only ever run once is eating up a lot of **normal, non-FAST** memory due to aggressive inlining and loop unrolling, which is leading to memory problems on 512 kB flash MCUs. But this has got nothing to do with the use of FAST RAM, as this is done on a per function basis with attribution, not on a per file basis.

SJ-Innovation, 2020-07-29:
The idea for `SLOW_CODE` - Currently a placeholder but with the understanding that initialisation and prepopulation code could be offloaded to external flash chips (which will obviously _kill_ loop performance but be irrelevent to init). 

I understand and agree that the system is working as currently intended. Though definitely yes, it is far from easy to use. My main issue with it though is the un-statisticity of the inlining, there is no way to predict if a code change will completely change the behaviour of the optimisations and result in a large amount of extra memory being used (this is the issue that prompted this investigation).

Ahh! Thank you for the clarification of `FAST_CODE_INLINE` that _might_ be a viable workaround for my issue. (I'll keep you posted)

Another interesting point about the initialisation code, I figured there would be some kind of issue with file-level optimisation. Could the `pragma optimize` be used to get around this? That supports function-by-function optimisation according to the specs. (Not sure if its implemented by the ARM compilers though. Ill do some digging) 

Thanks for the detailed response :)

mikeller, 2020-07-29:
@SJ-Innovation:

> Ahh! Thank you for the clarification of FAST_CODE_INLINE that might be a viable workaround for my issue. (I'll keep you posted)

Speaking from own experience it is - it will allow you to precisely exclude any code from being moved to ITCM RAM by extracting it into a function, and then attributing it with `FAST_CODE_INLINE`. This is clumsy, but not sure if it can be done any better, given the contstraints.

Ideally, for any cases of 'the size of the required code to support feature X makes it not viable to be optimised' or 'we can affort to not optimise this code because it is only run every Y loop iterations' cases, this should be done in a way that checks the condition in FAST RAM code:

```
FAST_RAM runSomethingInLoop()
{
  [...]
  if (conditionForExecutionIsMet) {
    runNonFastCode();
  }
  [...]
}

FAST_CODE_NOINLINE runNonFastCode()
{
  [...]
}
```

> Could the pragma optimize be used to get around this?

Struggling to find the reference now, but this was discussed in the past, and as it turns out gcc currently does optimisation only at file level, even if `#pragma optimize` is used.
Anyhow, we have started refactoring the code and separating initialisation and runtime into separate files, and what has been done so far looks promising, as it enforces cleaner / more conscious separation of initialisation and runtime, and the use of clear data structures for runtime code.

SJ-Innovation, 2020-07-29:
@mikeller 

Cheers for the heads-up. That should help a lot. 

I've just done a couple o` quick tests and it seems that the optimize attribute works on a function-by-function basis, 

`#define O_FAST __attribute__((optimize("Ofast")))`

`#define O_SIZE __attribute__((optimize("Os")))`

(in my test in `common_pre.h`)
Then prefixs the function in the same way as FAST_CODE. 

Glad to hear that the legacy is being refactored, maybe the above snippets could help alliviate a fair bit of work there?

mikeller, 2020-08-01:
@SJ-Innovation:

> I've just done a couple o` quick tests and it seems that the optimize attribute works on a function-by-function basis,

The attribute does, the pragma does not. Moving to function-by-function optimisation settings would be an approach, but it would not alleviate the problem of running out of FAST RAM that seems to be the core of this issue.

SJ-Innovation, 2020-08-04:
I'm working on implementing a "priority" system to extend FAST_CODE, following the idea that the priority of functions to be put in fast execution memory is defined by the macro. The final locations of the  priorities is then decided based on the actual amount of memory present on the target device. 
This would allow code that _absolutely_ needs to be as high perfomance as possible (PID core, gyro core etc) to be prioritised over _slightly_ less important (but still would be helpful to be boosted if possible) code. This would allievaite the running out of ITCM ram and would make porting to other targets with different memory structures easier, as perhaps different priorities could be isolated to different sections. 
Initial results seem promising, though will be put on hold for a little while as im moving house soon: :)

SJ-Innovation, 2020-08-19:
Okeidokie. I've moved house. I'm back. 

I've made a method of prioritising which code is "more important" than others so should get a higher chance of ending up in bonkers-fast memory. Please see my PR - #10116

closed 2020-10-04 by stale[bot]