betaflight/betaflight #10061: MSP_RAW_GPS request returns Hdop = 0

dev-fred, 2020-07-29:
**Describe the bug**
Using the MSP library from https://github.com/fdivitto/MSP, I can get all the GPS data except Hdop.
Exemple : 
Lat=120223312500 Long=1543101351251 Alt=11 Sat=10 Hdop=0.00
I can see Hdop in OSD using CLI osd_gps_sats_show_hdop = ON
Looking at code, 
betaflight/src/main/msp/msp.c lines 1366-1374

```
    case MSP_RAW_GPS:
        sbufWriteU8(dst, STATE(GPS_FIX));
        sbufWriteU8(dst, gpsSol.numSat);
        sbufWriteU32(dst, gpsSol.llh.lat);
        sbufWriteU32(dst, gpsSol.llh.lon);
        sbufWriteU16(dst, (uint16_t)constrain(gpsSol.llh.altCm / 100, 0, UINT16_MAX)); // alt changed from 1m to 0.01m per lsb since MSP API 1.39 by RTH. To maintain backwards compatibility compensate to 1m per lsb in MSP again.
        sbufWriteU16(dst, gpsSol.groundSpeed);
        sbufWriteU16(dst, gpsSol.groundCourse);
        break;
```
I can see that `gpsSol.hdop` is missing

**To Reproduce**
Betaflight 4.2.1, MSP configured on Uart3, connected to an esp8266 using `msp.request(MSP_RAW_GPS, &gps, sizeof(gps));` and `Serial.println((float) gps.hdop/100);` from Arduino

**Expected behavior**
gps.hdop/100 should contains the value I can see in OSD using a VTX.
Note: The same program with Inav returns the correct Hdop value.

**Flight controller configuration**
`
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.1 Jul 19 2020 / 06:19:20 (caa0d683c) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722SE, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# feature
feature GPS

# serial
serial 1 2 115200 57600 0 115200
serial 2 1 57600 57600 0 115200

# master
set acc_calibration = 111,13,269,1
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set osd_gps_sats_show_hdop = ON
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end

# `
**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Thanks a lot for your help

etracer65, 2020-07-29:
It's not really a bug if the `MSP_RAW_GPS` message doesn't have HDOP when it wasn't designed to. You might be asking for a feature request to add HDOP to this message, but it's not a bug.

etracer65, 2020-07-29:
And if you look at the README of the GitHub project you referenced you'll see it's designed to work with INAV which likely has a different implementation of this MSP message.

dev-fred, 2020-07-29:
OK, should I create a feature request and you close this issue ?

dev-fred, 2020-07-29:
I have added `sbufWriteU16(dst, gpsSol.hdop);` in `betaflight/src/main/msp/msp.c`  line 1374 and that works fine, I got hdop from the MSP port :

[image]

We need this fonctionnality for setting the home position when `numsat > 6 and hdop < 2.0` for a DIY beacon wifi system (esp8266)  linked to the ` UAV Law ` in France. With MSP we can share the GPS between the FC and the Beacon system.

What's the next step to make a feature request ?

Thanks for your help.

closed 2020-07-29 by etracer65