betaflight/betaflight #10063: Adds hdop to MSP_RAW_GPS request

dev-fred, 2020-07-29:
**Is your feature request related to a problem? Please describe.**
We need this fonctionnality for setting the `home position` when numsat > 6 and ` hdop < 2.0` for a DIY beacon wifi system (ESP8266) linked to the UAV Law in France. 
With MSP we can share the GPS between the FC and the Beacon system.
Note: linked to https://github.com/betaflight/betaflight/issues/10061
**Describe the solution you'd like**
I would like to get hdop using the MSP library from https://github.com/fdivitto/MSP

**Describe alternatives you've considered**
Add an arbitrary delay but it's not reliable.

**Additional context**
To get hdop via MSP simply add `sbufWriteU16(dst, gpsSol.hdop); `in `betaflight/src/main/msp/msp.c line `1374 and that works fine, 
```
   case MSP_RAW_GPS:
        sbufWriteU8(dst, STATE(GPS_FIX));
        sbufWriteU8(dst, gpsSol.numSat);
        sbufWriteU32(dst, gpsSol.llh.lat);
        sbufWriteU32(dst, gpsSol.llh.lon);
        sbufWriteU16(dst, (uint16_t)constrain(gpsSol.llh.altCm / 100, 0, UINT16_MAX)); // alt changed from 1m to 0.01m per lsb since MSP API 1.39 by RTH. To maintain backwards compatibility compensate to 1m per lsb in MSP again.
        sbufWriteU16(dst, gpsSol.groundSpeed);
        sbufWriteU16(dst, gpsSol.groundCourse);
	sbufWriteU16(dst, gpsSol.hdop);
        break;
```
I have tested and I got hdop from the MSP port :
[image]

McGiverGim, 2020-07-29:
Seeing that you are able to do the code changes, maybe is easier open directly a pull request if you want.

dev-fred, 2020-07-29:
Why not but I am not experienced

dev-fred, 2020-07-29:
What's the next step ?

haslinghuis, 2020-07-29:
Look at: https://github.com/betaflight/betaflight/blob/master/docs/development/Development.md

closed 2020-08-01 by mikeller