betaflight/betaflight #10066: BN-880 magnetometer not working

sweebee, 2020-07-31:
I've got a Mamba f405 fc running Betaflight 4.2.1. I've connected a BN-880 unit and I cant get the magnetometer to work. I thought i wired it wrong, but i flashed Inav and there it works fine.

dump:

```
# 

# dump

# version
# Betaflight / STM32F405 (S405) 4.2.1 Jul 19 2020 / 06:18:01 (caa0d683c) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# name: DC7

# resources
resource BEEPER 1 A08
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 B01
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 C09
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
[... 1029 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

```

McGiverGim, 2020-07-31:
Maybe the magnetometer is a QMC that mimics the HMC?

We have a very long thread with some incompatibility with some magnetometers, that work in inav but don't work in betaflight, you can find it here: https://github.com/betaflight/betaflight/issues/4531

augustozanellato, 2020-08-05:
For me the bn880 is working fine on kakute f7 4.2.1, I just had to run in cli `set mag_hardware = AUTO`

mikeller, 2020-08-06:
@sweebee: The manufacturer does not want magnetometers to be supported on this hardware;

https://github.com/betaflight/unified-targets/blob/71b47b6ddbcbed5b091afc8f8c7dd9ec6c9cecea/configs/default/DIAT-FURYF4OSD.config#L86

If you want to use a magnetometer please buy a different flight controller that supports magnetometers.

closed 2020-08-06 by mikeller