betaflight/betaflight #10067: Updated BTF Configurater from 10.6.0 to 10.7.0 now configurator does not recognise the flight controller when the USB is connected and I cannot select any actions tabs including the CLI tab

d605, 2020-07-31:
I have updated the Betaflight configurator from 10.6.0 which was all working fine to the latest 10.7.0 in order to update the flight controller to the latest BTF 4.2 firmware. When plugging in the USB cable and connect in the top left corner I get the configurator version as 10.7.0, the firmware version as 4.1.2 (as not up dated yet) and my target. I dont get the 3D model of the quad on the screen all there is is a purple font box that says BACK in capital letters. I cannot enter into any of the set up tabs including the CLI tab so i cannot enter into boot loader or flash new firmware, it is like the configurator is not recognising the flight controller. I already have the STM32 drivers installed from the previous version and the version appear to be the same so I have not downloaded and installed those at this point. I dont know what the issue is or if there is there a fix for this

Asizon, 2020-07-31:
Duplicated, @d605 please try solution posted here:
https://github.com/betaflight/betaflight-configurator/issues/2063

haslinghuis, 2020-07-31:
Right click on Betaflight Configurator and choose properties.
There you can add ` --disable-gpu` to the target field like in the picture

d605, 2020-07-31:
This is a screen shot showing what my properties page is, I cannot find a target field?

haslinghuis, 2020-07-31:
Please also do not ask the same question on three different places. People feel less inclined when you are wasting their valuable time.

d605, 2020-07-31:
Apologies, I am new to all this and don’t really know what I am doing, just trying to find some help. The last thing I want to do is waste any ones time.

haslinghuis, 2020-07-31:
You can find your icon? Right click on it and choose properties.
See in picture in which field to add the argument on the end of the line. This problem is being addressed in a feature or bug release.

haslinghuis, 2020-07-31:
If you are new to the hobby I suggest to watch some recent Joshua Bardwell video's on youtube.
https://www.youtube.com/watch?v=nALPi8cTXGY&list=PLwoDb7WF6c8k_Bpx8QsRHX1mcFjN5lGuw

etracer65, 2020-07-31:
@d605 Please stop replying via email. This doesn't allow you to attach images or screenshots so we can't answer your questions.

d605, 2020-07-31:
> If you are new to the hobby I suggest to watch some recent Joshua Bardwell video's on youtube.
> https://www.youtube.com/watch?v=nALPi8cTXGY&list=PLwoDb7WF6c8k_Bpx8QsRHX1mcFjN5lGuw

I am not new to the hobby been flying for some time I think I have watched just abut every video JB has released, just new to 
 GitHub as didnt know how els i could find the solution to the issue

d605, 2020-07-31:
Thank you Haslinghuis, 
My properties page does not look like yours. I dont have an option for shortcut

[Betaflight configuration properties page.docx](https://github.com/betaflight/betaflight/files/5008200/Betaflight.configuration.properties.page.docx)

Asizon, 2020-07-31:
@d605 just go to H:/Betaflight/Betaflight 10.7.0
And find here the correct executable icon to change properties

etracer65, 2020-07-31:
Moving this to the proper repo.