betaflight/betaflight #10068: I lost gyro updaye frequency cant change it from 8khz

moodz801, 2020-07-31:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Asizon, 2020-07-31:
Please look here:
https://github.com/betaflight/betaflight/wiki/4.2-Tuning-Notes#more-accurate-loop-times

moodz801, 2020-07-31:
Someone help I have the mobula 7 with the crazybee F4FR v2.0 and can't change the gyro frequency anymore it is just locked at 8khz. And I also need the stock cli dump for this configuration so I can start from scratch it updated to 4.2 in betaflight and I can't seem to figure out what has happened

haslinghuis, 2020-07-31:
This is not the correct place to seek support for these kinds of problems. This is the place to report bugs with the firmware. Please try an alternate source like:

Facebook Group: https://www.facebook.com/groups/betaflightgroup/
Betaflight Slack: http://slack.betaflight.com
RCGroups Thread: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/

closed 2020-07-31 by etracer65