betaflight/betaflight #10074: The blackbox logging rate is not fraction of current PID loop frequency

DusKing1, 2020-08-03:
I'm using LSM6DSO(ODR @6.66khz) on betaflight 4.3, the problem I met is although I set the sample_rate to 1/2 of PID loop frequency(@6.66khz), the actual sampling rate calculating from blackbox is not the same.

Here's my log file: 
[lsm6d_001.zip](https://github.com/betaflight/betaflight/files/5017022/lsm6d_001.zip)

To example the actual time interval between two frames, you can use those code:

```
df=pd.read_csv(dataFile)
loopIteration= df[ 'loopIteration'].values.T
time= df[ ' time (us)'].values.T
time=time-time[0]
diffOfTime = np.diff(time)
averageUs = np.median(diffOfTime)
print(averageUs)
samplingRate = 1000000/averageUs
print(samplingRate)
```

It will give the result of 382us and 2618hz. 

I was wondering whether the mapping between the p_ratio to sample_rate is correct. Can you take a look? @fgiudice98

etracer65, 2020-08-03:
At a high level I think the problem is due to the assumptions in the logging around 8K sample rate gyros (Invensense). The I-frame is based on a 32ms frequency with the P-frame rate based on a ratio to attain the desired logging rate. The problem is that non-8K based gyros don't work out to integer loop counts for 32ms. Like the BMI160, BMI270 and LSM6DSO (also technically the ICM20649 which is a 9K part). For example the LSM6DSO runs at 6.67K and has a loop time of 150us. Unfortunately this doesn't divide evenly into 32ms (0.032 / 0.000150 = 213.333) so the logging is always missing a fraction of a sample. Since the time constants used in the log analysis are based on the 32ms assumptions they miscalculate the timing of the samples and this can lead to artifacts in the FFT and graphs.

I don't see any easy way to fix this. It would likely take making the 32ms I-frame interval variable based on the sensor type so that the number of loops can be integer factors at different logging rates. For example 38.4ms would be a good choice for the LSM6DSO part (0.0384 / 0.000150 = 256). The changes on the firmware side are trivial for this, but I imagine it will be a significant amount of work on the BBE side. Additionally this will break any other log consumers (PID Toolbox, Plasmatree, etc.) until they can be updated deal with a variable logging timebase (which would have to be added to the log header).

DusKing1, 2020-08-04:
The "loopIteration" in BBL is in us, is that still 32ms timebase?

I don't think currently 32ms timebase won't break their (PIDtoolbox, Plasmatree, etc.) result. But to be on the safe side, in order to avoid unnecessary troubles for everyone, I would like to come to a conclusion after getting the exact answer to the above questions.

fgiudice98, 2020-08-04:
@DusKing1  I tested my code with a bunch of different looptimes and all was correct, so yes, i'm pretty sure that the formula is correct and it will always be a fraction of the pid looptime. The data rate seems to be 2609Hz but opening your log i see a bad configuration: the I interval should be 213 and i see 106, the P interval is 1 (should be 2), the P ratio is 106 (as it should be, but is calculated incorrectly); furthermore the logging frequency that you have can be made with a P ratio of 83 (and this isn't even half of the 106 of the P ratio in the log header). Can i get a diff just to see your configuration? What build are you using?

DusKing1, 2020-08-04:
Custom build. G4 prototype FC, LSM6DSO. About the firmware, maybe @jflyper can give us some information?

mikeller, 2020-08-04:
@etracer65: Agreed with your analysis. But I think there is no good way forward with what we have now, and with the currently low proliferation of non-8k gyros with users I think it might be best to rip that band aid off now, and implement the changes to log at non-32 ms I-frame rates, and put the actual timing into the header.
blackbox explorer will be able to catch up by the time the next release rolls around, and other tools will just have to catch up too.

DusKing1, 2020-08-04:
I'm not sure there's a need to put that in header, in my own tool, I calculated the sampling rate by using the 'time(us)' column. I think PTB also did in the same way, right? @bw1129 

Not sure about PID-analyzer

McGiverGim, 2020-08-04:
All of this is based in the assumption that the blackbox logging can "loose" frames. This is why a fixed time was defined. In this way is easy to see if you loose or not some.
For this reason, the best way is to write this info in the header and not calculate it from the data directly.

DusKing1, 2020-08-04:
@McGiverGim I think this is based on "blackbox logging can 'loose' some frames but overall it's working as design".

Let's say there's an FC in which its blackbox logging is working perfectly, diff(time-time[1]) will give like 2485x500, 2x501, 89x 496. The median number is still 500us which says its logging rate is 2khz. The calculation result corresponds to the theoretical result.

But for edge cases like my log, although we get 2602hz from I and P values, I can still see there are at least 2614 frames in 1 second. If 'time(us)' column's value does not any serious shift compare to real-world, from my calculation we should take the logging rate as 2618hz, and we should use 2618 for FFT's NFFT parameter, not 2602. Without considering artifacts this won't break the FFT percision.

But I don’t know if that 16hz really changes anything...

fgiudice98, 2020-08-04:
@mikeller but the calculations of the BBE are based on I interval and pid looptime, i am right?
But here the problem isn't either my code or the 32ms I frame interval, i think here the problem is a bad configuration or a problem in the firmware (@DusKing1 you didn't answered to my questions. What firmware versione are you using? What blackbox parameters?)

etracer65, 2020-08-04:
@fgiudice98 The problem is for non-8K gyros it's impossible to get an integer number of loops in 32ms. The code:
```
    blackboxIInterval = (uint16_t)(32 * 1000 / targetPidLooptime);
```
Doesn't work for example with bmi160 (312us), bmi270 (312 or 156us) or lsm6dso (150us) as `targetPidLooptime` will never divide an integer number of times into 32000. So it's impossible to get the "number of loops" that will occur in 32ms.

So the concept of logging 1/1, 1/2, 1/4, etc. of the PID rate breaks down if the gyro is not based on a 8K multiple sample rate.

fgiudice98, 2020-08-04:
@etracer65 i'm sorry, but it's not how it works. The blackbox sample rate depends only on the pid looptime and not on the i interval (as in the old implementation)

DusKing1, 2020-08-04:
@fgiudice98 I can't make commit about firmware information since the code was held be @jflyper , I don't want to make any mistake that might waste others' time to correct me. PID frequency was 3.33khz, bi-dir dshot300, blackbox_sample_rate=1. that's all. I didn't make any other change.

fgiudice98, 2020-08-04:
Before you wrote different settings, this is why i asked for the diff and firmware information. The new informations now match what is inside the blackbox header. Can you do a log with 1/4 sample rate just to see how it goes?

etracer65, 2020-08-04:
The `blackboxIInterval ` is actually 106.667 because of the non-integer relationship between 32ms and the pid loop time (300us). This gets truncated to **106**. This messes up the sample timing calculations when decoding the log.

In the example of a 1/4 logging rate then p_ratio gets calculated (because of two integer math truncations) to be 26 instead of 26.667. So every I-frame is missing a fraction of a P-frame. So if calculate back from the blackbox log parameters with 26 loops per 32ms you get a sample rate of 812.5hz when it's really 833.33hz.

fgiudice98, 2020-08-04:
@etracer65 now the p_ratio doesn't exist anymore, the pinterval is calculated directly from the sample ratio, so the frequency will always be a fraction of the pid loop

DusKing1, 2020-08-04:
@fgiudice98 then with your commits how can you explain my PID @3.33khz but blackbox sampling rate at 2618hz (or say 2602hz in your calculation)?

fgiudice98, 2020-08-04:
@DusKing1 that's what i'm trying to understand, because it really doesn't make sense, just look at the code and do some math. As i said, try with 1/4 sample rate and send the log.

etracer65, 2020-08-04:
@fgiudice98 You're missing my point. At some stage during the log parsing or data analysis it needs to turn the individual P-frame samples back into samples with accurate timing (when the sample was taken). This is important to the FFT analysis. The `blackboxIInterval` is supposed to represent the number of samples that occur in 32ms based on the current PID rate. Because of the integer math this value truncates and doesn't accurately represent the sample timing. As repeatedly said, this is because the 32ms time constant doesn't work for non-8K sampling gyros. So one less P frame is written per I frame and the timing of the samples as viewed in the decoded log is incorrect.

DusKing1, 2020-08-04:
@fgiudice98 
[lsm-0.25logging.zip](https://github.com/betaflight/betaflight/files/5023183/lsm-0.25logging.zip)

fgiudice98, 2020-08-04:
@etracer65 yes, i understand what you are saying, but the frequency is calculated from the timestamp generated by the FC and recorded in the log, so i don't understand the problem with the i interval floating point error (that in this case is like 0.6%, when the frequency error is like 45%). The only things that i can think about this problem are: something involved in the logging can't keep up with this rate, either the fc or the logging device; the pid loop frequency isn't costant and lower that expected. Because really, the code is just to make the blackbox logging frequency exactly a fraction of the pid loop frequency.

fgiudice98, 2020-08-04:
@DusKing1 now the error is less, in the 10% range. Look what i wrote just above, can you confirm that the cpu usage and pid loop frequency are good?

DusKing1, 2020-08-04:
Always below 11%

fgiudice98, 2020-08-04:
And the pidloop is good?

DusKing1, 2020-08-04:
Always stay at 300, what else do you want to know?

fgiudice98, 2020-08-04:
I'm trying to help you, if it's a problem that i'm asking things no problem for me. If you want to help yourself figuring out the problem take a log with some debug (like cycletime for example), but as i said, if it's a problem for you no problem, but to answer the first tagged question, the samplerate to p_ratio conversion (that by the way exists only for a backward compatibility in the msp) is correct.

DusKing1, 2020-08-05:
Alright alright.

```
# diff

# version
# Betaflight / AIRFLEETG4V11 (AFG4) 4.3.0 Jul  7 2020 / 19:41:57 (144b1c3947) MSP API: 1.44

# start the command batch
batch start

board_name AIRFLEETG4V11

# dma
dma ADC 5 0
# ADC 5: DMA1 Channel 1 Request 39

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# serial
serial 0 1 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# mode_color
mode_color 6 0 2
mode_color 6 1 6

# aux
aux 0 0 1 1950 2050 0 0
aux 1 13 3 1950 2050 0 0
aux 2 26 0 1950 2050 0 0

# master
set gyro_lowpass_hz = 0
set dyn_notch_width_percent = 0
set dyn_notch_q = 100
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = 33,-5,56,1
set mag_hardware = NONE
set rssi_channel = 16
set serialrx_provider = IBUS
set rx_min_usec = 998
set rx_max_usec = 2001
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 60
set pid_process_denom = 2
set osd_vbat_pos = 14380
set osd_rssi_pos = 14701
set osd_current_pos = 14724
set osd_mah_drawn_pos = 14757
set osd_avg_cell_voltage_pos = 14764
set debug_mode = GYRO_SCALED

profile 0

# profile 0
set dterm_lowpass_type = BIQUAD
set dterm_lowpass2_hz = 0
set feedforward_transition = 80
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set throttle_boost = 3
set d_pitch = 46
set f_pitch = 48
set d_roll = 42
set f_roll = 45
set f_yaw = 45
set d_min_roll = 28
set d_min_pitch = 30
set d_min_boost_gain = 30
set d_min_advance = 0

rateprofile 1

# rateprofile 1
set roll_rc_rate = 101
set pitch_rc_rate = 101
set yaw_rc_rate = 101
set roll_expo = 1
set pitch_expo = 1
set yaw_expo = 11
set roll_srate = 59
set pitch_srate = 59
set yaw_srate = 59

# end the command batch
batch end
```

closed 2020-09-12 by stale[bot]