betaflight/betaflight #10076: VTX table for SA2.1 missing 26dBm power lever

theely, 2020-08-03:
According to [TBS website](https://www.team-blacksheep.com/products/prod:unifypro32_nano) Smart Audio 2.1 do also support a 26dBm power mode that is currently missing in the Smart Audio 2.1 US VTX table:
https://raw.githubusercontent.com/wiki/betaflight/betaflight/resources/vtx_tables/vtx_table_smart_audio_2_1_us.json

etracer65, 2020-08-03:
Looks like it's there to me...
```
        "powerlevels_list": [
            {
                "value": 14,
                "label": "25 "
            },
            {
                "value": 20,
                "label": "100"
            },
            {
                "value": 26,
                "label": "400"
            }
        ]
```

theely, 2020-08-03:
My bad I meant 28dBm, the power levels are 4 in total.

etracer65, 2020-08-03:
The problem is that Smartaudio v2.1 doesn't define the output power, the individual VTX defines what its capable of.

For example the TBS UNIFY PRO32 HV supports different outputs:

> Output Power: 14dBm (25mW), 20dBm (100mW), 26dBm (400mW), 30dBm+ (1000mW+)

So there's no way a generalized "SA 2.1" table can be accurate for all. Individual tables for the specific VTX's could be sourced by TBS.

haslinghuis, 2020-08-03:
Agreed my TBS Evo has
> Output Power: 14dBm (25mW), 20dBm (100mW), 26dBm (400mW), 29dBm (800mW)

etracer65, 2020-08-03:
And according to the manual, the TBS UNIFY PRO32 5G8 nano looks like it originally (version 1.0) only had 3 power levels. The newer version 1.1 seems to have the extra 4th power level. So again, it's not possible to have a single SA 2.1 vtxtable that will support all these. The current table with 3 entries is the only one that will partially support all of the vtx's.

[image]

theely, 2020-08-04:
Makes sense, to generalise.
That being said, how can we help the average user to get most out of their VTXs?
Should we add specific VTX tables for the most common VTXs?
Or be very vocal about the generalisation and provide guidance how to extend the table to match the specific VTX power levels?

haslinghuis, 2020-08-04:
Betaflight already support users by means of the wiki and documentation:
```
https://github.com/betaflight/betaflight/wiki/VTX-tables
https://github.com/betaflight/betaflight/blob/master/docs/VTX.md#vtx-table
```

etracer65, 2020-08-04:
Ideally the manufacturer is the one responsible for supplying the proper vtxtables for their products.

m4ximuel, 2020-08-04:
many guys confused to vtx tables, tbs make me study how to use smartaudio 2.1 . 
Only people who are reading wiki download vtx tables.
Only people who are crazy about drones read how to use Smart Audio 2.1.
Therefore, manufacturers are obliged to provide vtx table.
Otherwise, DJI will hunting you.

haslinghuis, 2020-08-04:
Manufacturers already are responsible for delivering and maintaining unified target configurations for the Flight Controller.
VTX table configurations supplied and maintained by the manufacturer to pick from configurator would be a nice addition.

theely, 2020-08-05:
Right!
Have BF dev reached out to manufacturers to inform them about the need of creating and publishing per device VTX tables?

I tried and got ignored :-( I assume BF devs have a preferential channel they could leverage.

m4ximuel, 2020-08-05:
@theely: Only rush cherry provide vtx tables on their official facebook page and rcgroups page.

closed 2020-09-12 by stale[bot]