betaflight/betaflight #10077: Last 30 Seconds logging

montify, 2020-08-04:
Since my flight computer (Mamba F722s) has only a limited memory, I wanted to ask if it would be possible to add a feature (similar to airplanes) that saves the last 30 seconds before a failsafe.

This is helpful to find out what caused the fail-safe.

haslinghuis, 2020-08-04:
Our hardware doesn't have a CMU (Conflict Monitor Unit) to detect faults or conflicting signals and switch an intersection to an all flashing error signal (from wikipedia) ;-) So by what means are we going to detect this 30 seconds before it is going to occur?

montify, 2020-08-04:
I did not mean a 1:1 implementation ;)

We could use a ring buffer, we write continuously all data we want to log, and when a failsafe is triggered we stop writing to the buffer.
Shouldn't use too much cycletime, it's enough if you just write the errorcode or something like that.

montify, 2020-08-04:
Or more simply, we can use the existing logging function.

When the memory (of the FC) is full, just start writing again from the beginning.

Is that possible?

haslinghuis, 2020-08-04:
Round robin

montify, 2020-08-04:
Like here
https://github.com/betaflight/betaflight/blob/b488d388ba6a4747f3e90df3458362fe7ecc0117/src/main/blackbox/blackbox.c#L1816

insteal of 
`blackboxSetState(BLACKBOX_STATE_STOPPED);`

we need a state like 
`blackboxSetState(BLACKBOX_STATE_RESET);`
and just write from the start of the SD-Card.

so we have atleast one or a few frames of data.

haslinghuis, 2020-08-04:
Erase could take up 30 seconds? I have no blackbox yet so no expierence with timing irl
So keeping a ringbuffer in memory and write to flash when fail-safe occurs is a better solution if erase takes that long.

montify, 2020-08-04:
> So keeping a ringbuffer in memory and write to flash when fail-safe occurs is a better solution if erase takes that long.

You could overwritte instead of erase.

basically, if the buffer is full, please start writing at the beginning again.

haslinghuis, 2020-08-04:
It is not as easy as we think. Related / duplicate of https://github.com/betaflight/betaflight/issues/3343

montify, 2020-08-04:
this is not good :(
But at least after the failsafe the reason for the failsafe can be stored?

Before the ARM we delete the memory so it is free and ready to write.

When a failsafe is triggered, the log is simply written.

Of course the logging function is optional

haslinghuis, 2020-08-04:
But the only reason for a fail-safe is loss of Remote Control Link or triggered with switch by user? You already have your RSSI and maybe LQ if using Crossfire to monitor the health of that link.

montify, 2020-08-04:
> But the only reason for a fail-safe is loss of Remote Control Link or triggered with switch by user? You already have your RSSI and maybe LQ if using Crossfire to monitor the health of that link.

I think thats not true, GPS-Rescue can cause a failsafe too.

https://github.com/betaflight/betaflight/blob/master/src/main/flight/gps_rescue.c#L543-L560

haslinghuis, 2020-08-04:
GPS-Rescue is second stage phase action defined as procedure (drop, land, gps-rescue)?

montify, 2020-08-04:
What happens if I manually trigger the GPS Rescue and something goes wrong here?

haslinghuis, 2020-08-04:
It probably drops when satellites drops below threshold.
TL:DR Have a look at https://github.com/betaflight/betaflight/issues/7005 (resolved)

montify, 2020-08-04:
Anyway, that shouldn't be an issue here :)

Hopefully a dev reads this and can tell us if this is possible to write to a empty SD Card the failsafe reason.

etracer65, 2020-08-04:
No, not possible. The type of onboard flash used can't be overwritten. An entire block must be erased (slow) before being rewritten. And the erase is too slow to implement a rolling log. That's why erase the entire flash so that it's only necessary to write (erase already done).

And the MCU doesn't have enough memory to keep the data in memory.

etracer65, 2020-08-04:
> I think thats not true, GPS-Rescue can cause a failsafe too

GPS Rescue can't trigger a failsafe. GPS Rescue can disarm for a variety of reasons, but a disarm is not a failsafe. Failsafe is loss of signal and this can lead to a disarm.

closed 2020-09-12 by stale[bot]