betaflight/betaflight #10078: AKK - RDQ VTX Patch

alexdibi, 2020-08-05:
betaflight 4.2.1 is missing in the AKK - RDQ VTX Patch branch

mikeller, 2020-08-05:
@alexdibi: Sorry, forgot about that. Fixed now.

RipperDrone, 2020-08-05:
Looking back, adding this bespoken one single line of code and adding the option in BFC would have been the route of less maintenance efforts for you devs ;-)

mikeller, 2020-08-06:
@RipperDrone: If we'd added this hack to 'fix' the broken AKK hardware back then then we'd have about half a dozen other hacks for VTX / RX / cameras... with broken protocol implementations that Betaflight was approached about as well, along with all the user confusion and questions about them. I am glad we didn't go down that path.
Interesting observation: All the people that bought defective hardware from these other manufacturers just did the obvious thing and got it fixed or replaced by the manufacturer or supplier. Only AKK owners act with a sense of entitlement and demand that Betaflight 'fix' the broken hardware that they were sold...

closed 2020-09-12 by stale[bot]