betaflight/betaflight #10084: Range problem D16 frsky-x spi

kevindehecker, 2020-08-07:
With BF 4.0.6 onwards Frsky xjt D16 has very limited range on CrazybeeF4FR v3.0 and newer boards.
I've pinpointed the problem to this commit: https://github.com/betaflight/betaflight/commit/09ece0837658b6cdccea21377b82ac658d12794c#diff-fd08e2d940e44778faea0f50818cbd49
If I take BF4.0.6 and revert that commit, the problem is gone. The problem is still there with 4.2.1 and even with current master. (however reverting that commit there gives me merge conflicts which I couldn't figure out)

Interestingly:
-The original board of my Trashcan, the CrazybeeF4FR v2.0, does not have this problem. I think in this commit there was done a fix to prevent spi lockups specifically for the v2.0 hardware. It seems to me that this fix somehow breaks with the newer hardware. 
- Also other hardware, e.g. the original Mobula 6 board, is affected (which has a recent 1S version of the CrazybeeF4FR target), although not as bad (for the one I tested at least).
- BInding in LBT mode or not does not matter for this problem. (But I need LBT to be legal)
- The D8 protocol is unaffected by this problem. It works fine as far as I can tell on old and new hardware, in old and new BF. However, I don't like using that since it's sort of illegal to use in my country.

**To Reproduce**
1. Flash a CrazyBeeF4FR board v3 or higher with BF4.0.6 or higher.
2. Bind an x-lite with D16 8ch protocol.
3. Configure to have the led strip on a rc switch.
4. Frequently switch the leds while walking away from the quad. The led starts missing out after maybe a few meters. (note that it does depend on the hardware a bit, some boards seems more affected then others, but generally the range is bad. I've tested with around boards and got ranges between 1.5 meter and ~20 meter.)

**Flight controller configuration**
```
# 

# diff

# version
# Betaflight / STM32F411 (S411) 4.2.1 Jul 19 2020 / 06:18:29 (caa0d683c) MSP API: 1.43
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: 56f796fb, date: 2019-10-26T09:47:21Z

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id HAMO

# feature
feature -RX_PARALLEL_PWM
feature LED_STRIP

# led
led 0 4,7::CW:0
led 1 5,7::CT:6
led 2 6,7::CT:2
led 3 7,7::CT:6
led 4 8,7::CW:0

# color
color 1 0,0,1
color 3 0,0,255
color 5 359,0,1
color 6 359,0,1
color 7 359,255,255

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3
mode_color 6 0 0
mode_color 6 1 2
mode_color 7 0 7

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 1 1975 2100 0 0
aux 3 15 3 900 1025 0 0
aux 4 28 1 1300 2100 0 0
aux 5 35 2 1650 2100 0 0

# master
set frsky_spi_tx_id = 183,101
set frsky_spi_offset = -71
set frsky_spi_bind_hop_data = 6,24,44,64,84,104,124,144,164,184,204,224,9,29,49,69,89,109,129,149,169,189,209,229,14,34,54,74,94,114,136,154,174,194,214,234,19,39,59,79,99,119,139,159,179,199,219,0,0,0
set frsky_x_rx_num = 1
set gyro_1_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: LED_STRIP
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5:
    CH1: LED_STRIP
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
Thrascan with replaced flightcontroller to CrazybeeF4FR v3.

**Additional context**
I saw this issue https://github.com/betaflight/betaflight/issues/8603 which started out as having binding problems with the new x-lite S. I think this issue is different though, it binds just fine and use the x-lite original with xjt instead of acts. However, throughout the thread I suspect the problem of this issue comes up a few times (e.g. https://github.com/betaflight/betaflight/issues/8603#issuecomment-519617416) . Thought it'd be better to create a new issue to untangle this.

haslinghuis, 2020-08-07:
Related to #8603 
Could you please try a to flash a development build to test if it has improved like mentioned in the last comment in the linked issue. I would prefer to keep everything in one issue for a better overview.
> You may try one of the recent development builds. This has some improvements mainly for LBT support. This also includes ACCST 2.1.x support.

kevindehecker, 2020-08-08:
Ah, OK. The #8603 issue title focussing in on ACCTS confused me into thinking this was a different issue then. I did hope that my findings of reverting that commit and the hardware dependency on newer crazybee boards give new insights there as well. 

Anyway, I'll comment over there then.

MJ666, 2020-08-08:
@kevindehecker the commit from your first post is not affecting any of the CC2500 registers directly,  I don’t understand how it can affect rage. One idea to investigate is may be the LNA (if this is present on your FC) may stay disabled or get disabled for some reason with the new changes? What is the rage you get problems? With my BETAFPV F4 FC I have not seen any range issues but I fly it LOS and this means it is in the range of about 30-40m only.

kevindehecker, 2020-08-10:
@MJ666 `USE_RX_CC2500_SPI_PA_LNA` and `RX_CC2500_SPI_LNA_EN_PIN` are not defined for the CRAZYBEEF4FR target. They weren't in 4.0.5 when things were still fine, so not sure if this is the direction I should be looking into. It is defined for the MATEKF411RX though. So, I enabled it for testing for in CRAZYBEEF4FR target, but no noticeable change. Possibly because `RX_CC2500_SPI_LNA_EN_PIN` is wrong.

The range I get since https://github.com/betaflight/betaflight/commit/09ece0837658b6cdccea21377b82ac658d12794c#diff-fd08e2d940e44778faea0f50818cbd49 is only a few meters. Typically between 1.5 and 6 meters. So very close by I can fly LOS fine, but after only a few meters control quite suddenly becomes choppy, not much later I get "Telemetry Lost" and the quad failsafes. While before with the same hardware it definitely was 30+ meter. If I go back and forth between BF versions I have good/bad range and have done so with multiple boards quite a few times now. Just to say it is very reproducible.

kevindehecker, 2020-08-10:
I found and fixed the problem! The problem is gone if I remove the following line: https://github.com/betaflight/betaflight/blob/b1af9befe547204fc46e902e064371ce04cd1399/src/main/rx/cc2500_frsky_shared.c#L210

I found it by simply going line by line through the changes of https://github.com/betaflight/betaflight/commit/09ece0837658b6cdccea21377b82ac658d12794c#diff-fd08e2d940e44778faea0f50818cbd49 

I lack understand why this does the trick though. Apparently it is `0x07` by default, I don't understand why it has to be `0x14` during `!binding`.

Shall I create a PR for this? I'm testing on more hardware for now.

kevindehecker, 2020-08-10:
Range test results after two packs each: 
Original Trashcan Crazybeef4fr v2.x: good (i.e. unchanged)
Upgraded Trashcan Crazybeef4fr v3.0: good 
Mobula 6 (seems to be yet another Crazybeef4fr version): reasonable (slightly worse then D8 )

So this fix seems to work on all my hardware, although the performance on the Mobula6 and the Crazybee v2 is unaffected.
I tested with both D16 on the x-lite, and D16 LBT 8ch with a recently flashed multimodule.

dmrlawson, 2020-08-10:
I don't want to confused this issue at all but I think I have a similar problem with a Meteor65.

My friend has one with a v2.1 flight controller and has no issues. They've also never updated the version of betaflight it came with.

I've had issues with two different ones with v2.2 flight controllers, to the point I sold my taranis and bought a brand new one but got the same results.

There was a discussion I read on the ACCST 2.x thread about there being some difference which I thought might be related. I'll try to give your change a try to see if it works for me.

The symptoms I see are very similar. Sometimes it can be only a few meters away and gets an rxloss failsafe.

MJ666, 2020-08-10:
@kevindehecker this is a good find. The changed register is affecting the RX/TX fifo threshold’s. From reading the datasheet the value of 0x14 should be even invalid. The high nibble of the value should be always 0? May be it was meant to set the lower value to 14 0x0e. May be @mikeller can give some insight here since he made this change. If it works without this change not sure what the intention was. I also can not understand how the fifo handling is affecting range? The BETAFPV F4 FC is using an LNA/PA chip and this may hide the problem.

kevindehecker, 2020-08-11:
Yeah interesting. Well, the cc2500 datasheet does say:
 > Write 0 for compatibility with possible future extensions

So who knows what undocumented features may have been put in this particular chip batch... Certainly explains why some older hardware doesn't have this issue.

Otherwise, if I assume that even though 0x14 is invalid, it translates to only the first four bits being used ( ->0x04 ), meaning the rx/tx buf is 20 / 45 bytes instead of the usual 32/33. Given the D16 package length of [32 bytes](https://github.com/betaflight/betaflight/blob/1466acd5f381b0ef018c0c71ace63ec2c51af977/src/main/rx/cc2500_frsky_shared.h#L36), the Exti pin asserts 12 bytes too soon in which time the code is polling `cc2500ReadReg` hard. 
Maybe that causes some problems? I mean, what I notice besides rc loss itself is very choppy control just before that. As if the last received rc command is stuck for a few hundred of miliseconds, but I guess it could also be the cpu getting hold up on polling `cc2500ReadReg`...
Will try that out tomorrow by properly setting 0x04 as the fifo thresh and see what happens.

In any case, instead of simply removing: https://github.com/betaflight/betaflight/blob/b1af9befe547204fc46e902e064371ce04cd1399/src/main/rx/cc2500_frsky_shared.c#L210  ... I think it should be set such that the rc fifo thresh >= the packet size employed by the [protocol](https://github.com/betaflight/betaflight/blob/1466acd5f381b0ef018c0c71ace63ec2c51af977/src/main/rx/cc2500_frsky_shared.h#L36).

kevindehecker, 2020-08-11:
Setting cc2500WriteReg(CC2500_03_FIFOTHR,  0x04); has similar range instability problems as the invalid 0x14.

mikeller, 2020-08-11:
Thanks @kevindehecker for figuring this out. `0x14` for `FIFOTHR` is indeed invalid, it should always have been 14 / `0x0E`.

@MJ666: The reason for this change is to make the RX buffer larger to avoid overruns / multiple reads.

PR #10091 (2020-08-13): Fixed bug in FrSky X SPI code affecting RX range.
Fixes #10084. 

Thanks @kevindehecker for figuring this out!
closed 2020-08-13 by mikeller