betaflight/betaflight #10087: Custom names for pid, rate and osd profiles.

Silverwolf2332, 2020-08-09:
**Is your feature request related to a problem? Please describe.**
While using multiple mini quads ex. tinyhawk s and babyhawk R. that have different pid and rate profiles depending on what I'm doing with them, trying to remember what profiles I had set up for what actions I want to do I end up wasting a decent amount of time switching the pid and rate profiles not remembering what was set for what. 

**Describe the solution you'd like**
I feel that adding the ability to set custom names for PID,rate and osd profiles will be a quality of life improvement for people who own multiple quads and have issues related to mine.

etracer65, 2020-08-09:
These already all exist. To set the current profile or rate profile names:
```
set profile_name =
set rateprofile_name =
```

To name the OSD profiles:
```
set osd_profile_1_name =
set osd_profile_2_name =
set osd_profile_3_name =
```

Silverwolf2332, 2020-08-09:
I didn't know this existed, thank you.

closed 2020-08-09 by Silverwolf2332