betaflight/betaflight #10088: Cant get gps to work inav to betaflight

Ziov1, 2020-08-09:
I switched from in to betaflight, in inav I'm getting 16 Satalites in betaflight I can't get a GPS lock. Have left the drone out for 30 mins plus but nothing. Nothing shows in gps tab. 

In cli status I get i2c error = 1.

F722 se, m8q-5883.

What to do?

arkguy, 2020-08-11:
I am also having gps issues since upgrading to 4.2

this is from an iflight f7-d, and RDQ m8 gps

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.1 Jul 19 2020 / 06:19:20 (caa0d683c) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING, version: bc19b7dc, date: 2020-04-01T04:37:12Z

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# name: Batman

# feature
feature GPS

# serial
serial 0 68 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 2 1 900 1200 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 29
vtxtable powerlabels 25 100 400 800

# master
set acc_calibration = 33,-119,-11,1
set rssi_channel = 16
set dshot_bidir = ON
set failsafe_switch_mode = STAGE2
set vbat_scale = 84
set vbat_divider = 1
set gps_provider = UBLOX
set gps_rescue_descend_rate = 100
set osd_vbat_pos = 14756
set osd_rssi_pos = 14773
set osd_current_pos = 14724
set osd_craft_name_pos = 14796
set osd_gps_speed_pos = 14435
set osd_gps_sats_pos = 14424
set osd_home_dir_pos = 14451
set osd_home_dist_pos = 14419
set osd_altitude_pos = 14403
set osd_warnings_pos = 14761
set vtx_band = 2
set vtx_channel = 5
set vtx_power = 4
set vtx_freq = 5809
set gyro_2_align_yaw = 900
set name = Batman

profile 0

rateprofile 0

# end the command batch
batch end

# 

and the GPS info......
# get gps
gps_provider = UBLOX
Allowed values: NMEA, UBLOX, MSP
Default value: NMEA

gps_sbas_mode = NONE
Allowed values: AUTO, EGNOS, WAAS, MSAS, GAGAN, NONE

gps_sbas_integrity = OFF
Allowed values: OFF, ON

gps_auto_config = ON
Allowed values: OFF, ON

gps_auto_baud = OFF
Allowed values: OFF, ON

gps_ublox_use_galileo = OFF
Allowed values: OFF, ON

gps_ublox_mode = AIRBORNE
Allowed values: AIRBORNE, PEDESTRIAN, DYNAMIC

gps_set_home_point_once = OFF
Allowed values: OFF, ON

gps_use_3d_speed = OFF
Allowed values: OFF, ON

gps_rescue_angle = 32
Allowed range: 0 - 200

gps_rescue_initial_alt = 50
Allowed range: 20 - 100

gps_rescue_descent_dist = 200
Allowed range: 30 - 500

gps_rescue_landing_alt = 5
Allowed range: 3 - 10

gps_rescue_landing_dist = 10
Allowed range: 5 - 15

gps_rescue_ground_speed = 2000
Allowed range: 30 - 3000

gps_rescue_throttle_p = 150
Allowed range: 0 - 500

gps_rescue_throttle_i = 20
Allowed range: 0 - 500

gps_rescue_throttle_d = 50
Allowed range: 0 - 500

gps_rescue_velocity_p = 80
Allowed range: 0 - 500

gps_rescue_velocity_i = 20
Allowed range: 0 - 500

gps_rescue_velocity_d = 15
Allowed range: 0 - 500

gps_rescue_yaw_p = 40
Allowed range: 0 - 500

gps_rescue_throttle_min = 1100
Allowed range: 1000 - 2000

gps_rescue_throttle_max = 1600
Allowed range: 1000 - 2000

gps_rescue_ascend_rate = 500
Allowed range: 100 - 2500

gps_rescue_descend_rate = 100
Allowed range: 100 - 500
Default value: 150

gps_rescue_throttle_hover = 1280
Allowed range: 1000 - 2000

gps_rescue_sanity_checks = RESCUE_SANITY_ON
Allowed values: RESCUE_SANITY_OFF, RESCUE_SANITY_ON, RESCUE_SANITY_FS_ONLY

gps_rescue_min_sats = 8
Allowed range: 5 - 50

gps_rescue_min_dth = 100
Allowed range: 50 - 1000

gps_rescue_allow_arming_without_fix = OFF
Allowed values: OFF, ON

gps_rescue_alt_mode = MAX_ALT
Allowed values: MAX_ALT, FIXED_ALT, CURRENT_ALT

gps_rescue_use_mag = ON
Allowed values: OFF, ON

frsky_gps_format = 0
Allowed range: 0 - 1

osd_warn_no_gps_rescue = ON
Allowed values: OFF, ON

osd_warn_gps_rescue_disabled = ON
Allowed values: OFF, ON

osd_gps_speed_pos = 14435
Allowed range: 0 - 15359
Default value: 234

osd_gps_lon_pos = 234
Allowed range: 0 - 15359

osd_gps_lat_pos = 234
Allowed range: 0 - 15359

osd_gps_sats_pos = 14424
Allowed range: 0 - 15359
Default value: 234

osd_gps_sats_show_hdop = OFF
Allowed values: OFF, ON

# 

I was using GPS rescue before upgrading and was able to arm without issues.... now I cannot get the thing to recognize the GPS data.
I am aware of the issue with the sat info not being displayed on the OSD, but I dont think BF is seeing the modules data at all.

Any suggestions would be appreciated.

haslinghuis, 2020-08-12:
Related / Duplicate of https://github.com/betaflight/betaflight/issues/10040
Also take a look at https://github.com/betaflight/betaflight/issues/9652 for some configuration pointers.

arkguy, 2020-08-12:
Well it really is a shame but I gotta go back to 4.1    

BF 4.2 performance on my quad is WAY better than 4.1 but near as I can tell whatever changed in the GPS routines simply don't work.  In all seriousness though I can ALMOST live without the GPS for the improved flight performance... its really good.

haslinghuis, 2020-08-20:
Did you even try the suggestions in the supplied link (#9652)?

arkguy, 2020-08-25:
:) thanks for the follow up.  I did try all the suggestions I could  
find in a couple different forums.  I tried flashing back to  4.17 and  
the problem persists so perhaps blaming 4.2 is inaccurate.  Been busy  
with work but when I have time to investigate more I'll update the  
post.  again, thanks for asking.

closed 2020-10-12 by stale[bot]