betaflight/betaflight #10089: RXLOSS on Frsky SPI RX (CrazybeeF4FR) and Devo 7e (Deviation firmware + CC2500 radio)

unknownPT, 2020-08-09:
Hello,

I have successfully bind in D8 mode (Frsky_D), but range very short, 5-7m maximum, after got RXLOSS on OSD.
RSSI also very low, about 60-70% maximum.
How to sort this issue out?

**Flight controller configuration**
Flight controller info, identifier: BTFL, version: 4.0.4

``` 
# version
# Betaflight / CRAZYBEEF4FR (C4FR) 4.0.4 Jun 30 2019 / 14:06:48 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id 

# name
name Cinecan

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -AIRMODE
feature MOTOR_STOP

# beeper
beeper -ARMED

# beacon
beacon RX_LOST

# map

# serial
serial 0 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 2 1 1300 2100 0 0
aux 2 28 1 1500 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_trim_pitch = -8
set min_check = 1110
set rx_spi_protocol = FRSKY_D
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = STAGE2
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 330
set ibata_scale = 470
set small_angle = 180
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 2091
set osd_rssi_pos = 2081
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2102
set osd_remaining_time_estimate_pos = 2135
set osd_flymode_pos = 2451
set osd_g_force_pos = 65
set osd_throttle_pos = 2423
set osd_crosshairs_pos = 2253
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2437
set osd_craft_name_pos = 2443
set osd_power_pos = 2272
set osd_warnings_pos = 2377
set osd_disarmed_pos = 331
set osd_esc_rpm_pos = 2305
set vtx_band = 1
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5825
set frsky_spi_tx_id = 186,162
set frsky_spi_offset = -12
set frsky_spi_bind_hop_data = 0,30,60,91,120,150,180,210,5,35,65,95,125,155,185,215,10,40,70,100,130,160,190,221,15,45,75,105,135,165,195,225,20,50,80,110,140,170,200,230,25,55,85,115,145,175,205,0,0,0

# profile
profile 0

set p_pitch = 50
set i_pitch = 50
set d_pitch = 30
set f_pitch = 60
set p_roll = 50
set i_roll = 45
set d_roll = 27
set f_roll = 60
set p_yaw = 72
set i_yaw = 55
set f_yaw = 100

# rateprofile
rateprofile 1

# end the command batch
batch end
```
----------
``` 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
[Eachine Cinecan ](https://www.eachine.com/Eachine-Cinecan-85mm-4K-Cinewhoop-3-4S-FPV-Racing-Drone-BNF-or-PNP-Crazybee-F4-PRO-V3_0-Caddx-Tarsier-Cam-DVR-p-1511.html) with Frsky NON-EU receiver Version
SPI BUS receiver
Frsky D8/D16 switchable
Compatible Non EU transmitter both D8 and D16 mode
Channels: 8ch or 16ch
Failsafe support
No ground interference ( Transmitter and receiver 1 m from the ground): 300m

Gozzz, 2020-08-16:
Maybe give the latest dev build a try? #10091

unknownPT, 2020-08-31:
> @unknownPT use Betaflight Configurator 10.7.0 to flash it

Flashed BF4.2.2 but nothing changed, same issue with range. Any ideas how to sort out?

closed 2020-10-12 by stale[bot]

Gerd-tech, 2021-08-03:
Hello, 

I notice exactly the same problem. I guess that there is a bug introduced somewhere between Betaflight 3.5.7 and 4.2.2. Maybe a mistake in a setting/register/configuration of the RF chip?. Even Happymodel states this at various places on their website. Example:
_Notes:
SPI Frsky version comes with Betaflight 3.5.7 firmware in order to get better RSSI range . Bind code is “Frsky_bind” in the CLI Command
SPI Flysky version comes with Betaflight 4.2.2 firmware . Bind code is “Bind_rx” in the CLI command_
[http://www.happymodel.cn/index.php/2020/09/29/happymodel-crux3-1-2s-3inch-toothpick-fpv-racer-drone/](url)
In the Notes in red below the movies

As most of the people use the small quads/whoops out of the box, they will not notice the problem (as flashed with BF 3.5.7 by HappyModel), however if you want to benefit from RPM filters, you need to upgrade BF and will see the problem. As it working with BF 3.5.7, it is very likely not a HW problem but a SW problem.

Did you find out something in the mean time? Should we re-open this problem?

Could this be related to problem #10084 or is problem #10084 only related to D16 mode?

Thank you