betaflight/betaflight #10090: T Motor F4(STM32F4) (non HD) External BMP280

chumunga, 2020-08-10:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
Installed an external barometer, using the SDA and SCL pads
I used the CLI to set baro_hardware = auto as well as bmp280
I also change the default bus from SPI to I2C and checked that in resource show, that the pad I am using is configured for I2C which appears to be and the board has the pads for them.
Also checked with multimeter that the flight controller is providing 3.3v

But despite my attempts I am not sure how else to proceed to troubleshoot why Barometer isn't working.

2. Include ways to reproduce the problem;
Not sure if its my board or betaflight

3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
T Motor F4 (Non-HD)
T Motor Pacer P50amp ESC
GY-BMP280 

- the version of all the software used;
4.1.1
- how things are connected / wired up.
Baro_VCC to FC_3v3 
gnd to gnd
SDA to SDA
SCL to SCL

4. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/);

```
# resource show
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: LED 1
C06: FREE
C07: FREE
C08: FLASH_CS
C09: CAMERA_CONTROL
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: INVERTER 1
C15: INVERTER 5
D02: SERIAL_RX 5
```

haslinghuis, 2020-08-10:
Please also provide the `diff` as requested. Hint: `see step 4`

etracer65, 2020-08-10:
Try:
```
set baro_hardware = AUTO
set baro_bustype = I2C
set baro_i2c_device = 2
save
```
This also assumes you've connected the SCL/SDA lines correctly.

chumunga, 2020-08-10:
@etracer65 ty!!! it was 

set baro_i2c_device = 2 that did it, what does this command express exactly?

Is the value 2 because of 2 in below pin out:
B10: I2C_SCL 2
B11: I2C_SDA 2

etracer65, 2020-08-10:
@chumunga Pins B10 and B11 are connected the the 2nd I2C bus in the microcontroller. So the setting tells the baro driver to use the 2nd bus and those corresponding pins.

chumunga, 2020-08-11:
I had a feeling that may of be the case. Thank you so much.

closed 2020-08-11 by chumunga