betaflight/betaflight #10093: Telemetry not working with F. Port 2.0 in Betaflight 4.2.0

guilleortas, 2020-08-13:
**Bug description**: Telemetry doesn't work on Betaflight when using F. Port 2.0 with the latest ACCESS receivers such as the Archer M+. RSSI on OSD and BF configurator reads 0, and LUA script controls from the radio do not work. Control link works as expected.

**To reproduce the issue**, just connect an ACCESS receiver using the latest firmware (i.e. v2.1.0) to a FC using F. Port.
The control link works, but telemetry does not.

**Flight controller configuration (diff all)**
``` 
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.0 Jun 14 2020 / 03:04:43 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: HAMO, board_name: CRAZYBEEF4FR, version: 56f796fb, date: 2019-10-26T09:47:21Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FR
manufacturer_id HAMO
mcu_id 003400593339510739363632
signature 

# name: Mobula6

# feature
feature -RX_PARALLEL_PWM
feature -RX_SPI
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1625 2050 0 0
aux 1 1 0 1775 1900 0 0
aux 2 2 0 1625 1775 0 0
aux 3 13 1 1575 1675 0 0
aux 4 31 1 1850 1925 0 0
aux 5 35 1 1675 1800 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# rxfail
rxfail 3 h

# master
set acc_calibration = 50,50,-127,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set fpv_mix_degrees = 12
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_device = NONE
set dshot_idle_value = 1000
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 450
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 310
set ibata_scale = 1175
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set osd_warn_rssi = ON
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_tim2 = 1537
set osd_vbat_pos = 2433
set osd_rssi_pos = 2425
set osd_tim_2_pos = 2454
set osd_current_pos = 2444
set osd_mah_drawn_pos = 2401
set osd_warnings_pos = 14602
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bb_no = OFF
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5695
set frsky_spi_tx_id = 128,4
set frsky_spi_offset = -1
set frsky_spi_bind_hop_data = 2,17,32,49,62,77,92,107,122,137,152,167,182,197,212,227,7,22,37,52,67,82,97,112,127,142,157,172,187,202,217,232,12,27,42,57,72,87,102,117,132,147,162,179,192,207,222,255,255,255
set frsky_x_rx_num = 2
set gyro_1_align_yaw = 900
set name = Mobula6

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
``` 

**Setup / Versions**
 - Flight controller: CrazybeeF4FR (Mobula 6)
 - RX: FrSky Archer M+ with ACCESS firmware v2.1.0.
 - TX: FrSky Q X7 ACCESS radio with ACCESS firmware v2.1.0 on the ISRM internal module.
 - F.Port configuration: uninverted SmartPort from the receiver connected to TX1 pad on the FC. (UART 2 is taken by the vtx for SmartAudio).

**Additional context**
I updated my Q X7 ACCESS radio internal ISRM module to v2.1.0, I flashed my Archer M+ receivers to ACCESS 2.1.0. The receiver is registered and bound, F.Port selected in the TX menu, all telemetry sensors deleted, and then re-discovered from the radio.
Control link works fine on all channels, I get RSSI on TX but RSSI is showing as 0 in Betaflight (both on OSD display and on setup page in the BF configurator). Tried switching Serial_Halfduplex from On to Off but did not fix the issue and broke control link until switched back. That's not the fix, I know how to configure F.Port.
The telemetry control through the radio using LUA scripts also doesn't work, for instance, to control the VTX through the radio (mind you SmartAudio is indeed working fine when configured from the OSD directly).

The conclusion is that Betaflight is not able to properly read telemetry from the receiver using the newer F.Port 2.0 protocol or the newer ACCESS firmware versions. The control link is completely fine, but telemetry is not being correctly interpreted by the FC. Telemetry is received correctly at the transmitter. The issue is with Betaflight.

Gozzz, 2020-08-16:
@guilleortas try this- 

`set tlm_halfduplex = OFF
save
`

guilleortas, 2020-08-16:
> @guilleortas try this-
> 
> `set tlm_halfduplex = OFF `

I already did, please read my whole post. That is not the issue.
The problem is that Betaflight can't properly interpret F. port 2.0

Gozzz, 2020-08-16:
> > @guilleortas try this-
> > `set tlm_halfduplex = OFF `
> 
> I already did, please read my whole post. That is not the issue.
> The problem is that Betaflight can't properly interpret F. port 2.0

I do not see that mentioned at all in your diff or things you have tried. Regardless, telem will not work if tlm_halfduplex is not set to off when using fport.

guilleortas, 2020-08-16:
> I do not see that mentioned at all in your diff or things you have tried. Regardless, telem will not work if tlm_halfduplex is not set to off when using fport.

Refer to the section **Additional context**, where I mention the option `tlm_halfduplex`
That's not the issue. Telemetry was working fine on F. Port 1.0, but now doesn't on version 2.0

Gozzz, 2020-08-16:
> Refer to the section **Additional context**, where I mention the option `tlm_halfduplex`
> That's not the issue. Telemetry was working fine on F. Port 1.0, but now doesn't on version 2.0

Maybe you should re-read the **Additional context** of you post then and then re-read what I wrote. Because I still don't see you have tried **tlm_halfduplex=off**

guilleortas, 2020-08-16:
> Maybe you should re-read the **Additional context** of you post then and then re-read what I wrote. Because I still don't see you have tried **tlm_halfduplex=off**

My bad, you are correct.
However, that does not solve the issue, it persists no matter what.

Gozzz, 2020-08-16:
Odd, so you saved and power cycled? Still no new sensors discovered? 

I can test this myself later on and verify. Just won’t have time until tomorrow.

guilleortas, 2020-08-16:
> Odd, so you saved and power cycled? Still no new sensors discovered?
> 
> I can test this myself later on and verify. Just won’t have time until tomorrow.

Nope, same thing.
Sensors show up on the radio, the thing is that RSSI is not automatically picked up by betaflight as normal when using F. Port.
Furthermore, you cannot control the VTX through BF's LUA scripts. When you try to change the band or channel it gets stuck in "Retrying..." and does nothing.
Please try to test this out yourself tomorrow if you can

etracer65, 2020-08-16:
> Sensors show up on the radio

This doesn't mean that telemetry from the flight controller is working. All it means is the built-in handful of sensors from Frsky are working between the receiver and transmitter. In fact the behaviors you've described are exactly consistent with telemetry from the flight controller to the receiver not working.

For telemetry to work on a F4 like you have, first you need to ensure the S.port wire from the receiver is connected to the TX pad of the UART. Not the SBUS pad, not the RX pad. Next since this is an F4 you need to provide an uninverted S.port signal. This generally means you need to modify the receiver, or if you're lucky it might have an extra uninverted S.port connection (like the little solder pads on a Frsky R-XSR). Since you seem to be using some clone receiver I can't tell you if this is available or not. IF you can't obtain an uninverted S.port signal then you're out of luck. Assuming you've met the first 2 requirements, then see here to get the appropriate settings to use for your flight controller.

https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol

klutvott123, 2020-08-16:
There's a bug in the Archer M+ firmware https://github.com/FrSkyRC/Firmware-Test/issues/72. It shouldn't prevent telemetry from working but it causes BF LUA issues.

guilleortas, 2020-08-16:
> This doesn't mean that telemetry from the flight controller is working. All it means is the built-in handful of sensors from Frsky are working between the receiver and transmitter. In fact the [...]

Thanks for your message. Unfortunately, I already did everything you mentioned in your post, as I'm familiar with setting up a receiver with F.Port. Let me point out that I'm not using some clone receiver as you say, The Archer M+ receiver is from FrSky's new Archer series and replaces the XM+. More info: https://www.frsky-rc.com/product/archer-m/

> There's a bug in the Archer M+ firmware [FrSkyRC/Firmware-Test#72](https://github.com/FrSkyRC/Firmware-Test/issues/72). It shouldn't prevent telemetry from working but it causes BF LUA issues.

Wow it seems that might actually be the source of the issue, thanks for the heads-up

organizedkaos, 2020-08-20:
I have two (2) Archer M+ receivers installed with ACCESS and F.Port2.0 on two betafpv f4 boards.  They are flashed with the first release (and only firmware as of this writing) access firmware for the archer m+.

F.Port and telemetry with bf lua scripts works.  The only issue I have is 0 rssi which is on frsky to fix.

I doubt there is anything wrong on betaflight or frsky concerning the archer m+ f.port2 not working.

Perhaps people are trying to wire to an improper uart.  Again, besides the rssi, f.port2 on the archer m+ and bf4.2 works provided it is setup properly.

guilleortas, 2020-08-20:
> I have two (2) Archer M+ receivers installed with ACCESS and F.Port2.0 on two betafpv f4 boards. They are flashed with the first release (and only firmware as of this writing) access firmware for [...]

You're correct, telemetry partly works, but not fully. Also, there's no issue with the UART I'm using.
We both have the same situation, some functionality does work with the LUA script, but some doesn't.
For example, try to change the channel of your VTX or the output power through your radio using BF's LUA script. For me, it gets stuck in "Retrying...", and can't complete the command.
RSSI as you say still doesn't work at all.

organizedkaos, 2020-08-20:
I have no issues using any page in the lua script including the vtx page.  I have changed my vtx channels many times now to accommodate my flying areas.  Maybe I'm just lucky.  I am using the x9lite-s but I don't want to say the tx is making a difference because I don't know.

klutvott123, 2020-08-20:
@organizedkaos Try setting any value on any page to 125 or 126 and save. Does it work?

organizedkaos, 2020-08-20:
@klutvott123 Interesting.  I haven't had the need to get to those two exact numbers.  I tested it and it does indeed hang on 125 or 126.  Any other value or setting in any other page are no issues.

klutvott123, 2020-08-20:
@organizedkaos The crc for the msp/telemetry frame can also be 125 or 126 and make it fail even if you didn't set any value to those numbers. The user doesn't see the crc so this behaviour can appear to be random. Several of frskys receivers have been affected by different variants of this bug. Frsky has been informed and acknowledged that it's a bug multiple times but they keep implementing it in their firmwares.

whosmatt, 2020-09-05:
I have the exact same issue with BF 4.2.2 and an Archer M+ on a Crazybee F4 v2.0 and with an x-Lite Pro. Everything is on the newest firmware.  

Telemetry mostly works: 
Betaflight OSD gives me a valid LQ reading. 
Betaflight lua scripts work perfectly on my tx. 
I get what I believe is full telemetry readings on my tx. 

Its only RSSI which always shows a blinking zero. 

Can I somehow help by logging the F Port bus?

klutvott123, 2020-09-12:
@whosmatt It's a bug. Frsky has to fix it.

ozonisator, 2020-09-12:
@whosmatt I've got the same problem with my Archer M+ and xlite pro via F.Port. As a quick&dirty workaround you can configure your radio to send the RSSI value via a RC channel back to the flight controller. Apply an offset so that 0-100 RSSI is scaled to -100 to +100 so that you can use the full range of the RC channel.

guilleortas, 2020-10-01:
UPDATE:
FrSky has fixed this issue as of Sept 30 with the firmware version 2.1.2 for the Archer M+ receiver.
I checked and automatic RSSI input with F.Port works for me now.

Disclaimer: The FW version still shows as 2.1.0 on the RX after updating for some reason.

closed 2020-12-25 by stale[bot]