betaflight/betaflight #10102: Missing space in default config BKMN-NERO.config (sdcard_spi_bus)

bbs1, 2020-08-16:
**Describe the bug**
There is a missing space in BKMN-NERO.config
setsdcard_spi_bus = 3
should have been
set sdcard_spi_bus = 3

**To Reproduce**
Check config file
https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/BKMN-NERO.config

**Setup / Versions**
 - NERO 7 Flight controller

DusKing1, 2020-08-16:
Wrong repo. Should be betaflight/unified-target.

bbs1, 2020-08-16:
https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/BKMN-NERO.config is the path that is used in BetaFlight Configurator 10.7.0.
This caused the controller to crash as soon as SD card log was enabled

bbs1, 2020-08-16:
Sorry I misunderstood. Please close the issue and I will submit it in betaflight/unified-target

etracer65, 2020-08-16:
See https://github.com/betaflight/unified-targets/issues/334

closed 2020-08-16 by etracer65