betaflight/betaflight #10107: Onboard ADC cannot detect battery and showing 0.0v in 4.2.2

dumbfingers, 2020-08-17:
**Describe the bug**
Under targer CLRacing F7, version 4.2.2, betaflight configurator will not detect battery voltage and the voltage will also shown as 0.0V in OSD.

**To Reproduce**
Steps to reproduce the behavior:
* enter betaflight configurator's firmware flasher
* choose target CLRACING F7, version 4.2.2 (released 16/08/2020)
* flash the firmware as usual
* keep the default settings, plug in a fully charged healthy battery
* betaflight configurator will show no battery and the battery voltage is 0

**Expected behavior**
With default settings, betaflight should detect battery voltage.
quick testing by flashing version 4.2.1 with all default settings proved that 4.2.1 works as intended.

**Flight controller configuration**
diff is not applicable as this will happen with default settings.

**Setup / Versions**
N/A. 
default settings in 4.2.1 works

etracer65, 2020-08-17:
Your settings are corrupted (`dma ADC 1 NONE`). Reset to `defaults` and reconfigure.

dumbfingers, 2020-08-17:
reset to `defaults` will resolve this issue.
I also discovered this line
```
# dma
dma ADC 1 NONE
```
in my diff backup.

By commenting out this line and I'm able to recover the settings and configurations whilst the battery voltage is showing up as normal.

etracer65, 2020-08-17:
There's no way the voltage meter would have worked before if you had the `dma ADC 1 NONE` setting, regardless of the version.

closed 2020-08-17 by dumbfingers