betaflight/betaflight #10109: No pitch Control in Air mode F4 PRO V3 Betaflight

makalele666, 2020-08-18:
Hi guys,I have a problem, I have absolutely no pitch control when I arm and in air mode, once i change it into angle mode it suddenly becomes fine. In the reciever tab it all seems to be fine and i can operate pitch, roll, yaw etc, even after switching to air mode, rc works fine. However, while in the air after angle mode is turned off, pitch control stops working. 
Betaflight 4.2.0 on omnibus F4 Pro V3 (omnibusF4SD firmware), previously on 4.1.1 everything worked perfectly. 
Interestingly, after switching from 4.1.1 to 4.2.0, it suddenly turned out that I need to set the Board Alignment at 270 degrees on the yaw axis, because the orientation has changed, maybe there is a problem?
Identical problem was described here, but not resolved:
https://intofpv.com/t-no-pitch-control-in-air-mode

etracer65, 2020-08-18:
Please provide the requested information or the issue will be closed.

> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
>

makalele666, 2020-08-18:
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# feature
feature -AIRMODE
feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature MOTOR_STOP
feature GPS

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2 115200 9600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10

# aux
aux 0 0 2 1300 2075 0 0
aux 1 1 0 900 1400 0 0
aux 2 2 0 1300 1700 0 0
aux 3 46 4 1625 2100 0 0
aux 4 13 1 1550 2100 0 0
aux 5 19 3 900 1350 0 0
aux 6 28 0 1650 2100 0 0

# rxfail
rxfail 4 s 1000
rxfail 5 s 2000
rxfail 11 s 1050
rxfail 15 s 750

# master
set gyro_calib_duration = 0
###ERROR: diff: CORRUPTED CONFIG: gyro_calib_duration = 0###
set gyro_calib_noise_limit = 0
set gyro_overflow_detect = OFF
set acc_calibration = 30,25,-2,1
set mag_hardware = NONE
set min_check = 1039
set rssi_channel = 12
set rc_interp_ch = RP
set rc_smoothing_type = INTERPOLATION
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = OFF
set rc_smoothing_auto_smoothness = 0
set serialrx_provider = SBUS
set airmode_start_throttle_percent = 35
set min_throttle = 1028
set dshot_idle_value = 450
set motor_poles = 0
###ERROR: diff: CORRUPTED CONFIG: motor_poles = 0###
set failsafe_delay = 22
set failsafe_off_delay = 100
set failsafe_throttle = 1300
set failsafe_procedure = GPS-RESCUE
set align_board_yaw = 270
set bat_capacity = 1500
set vbat_min_cell_voltage = 325
set vbat_warning_cell_voltage = 336
set small_angle = 6
set gps_sbas_mode = AUTO
set gps_rescue_min_sats = 7
set pid_process_denom = 4
set osd_vbat_pos = 2466
set osd_rssi_pos = 2401
set osd_throttle_pos = 2241
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 449
set osd_mah_drawn_pos = 459
set osd_gps_sats_pos = 2093
set osd_home_dir_pos = 2099
set osd_home_dist_pos = 2101
set osd_altitude_pos = 2263
set osd_warnings_pos = 14729
set osd_disarmed_pos = 2282
set osd_nvario_pos = 2327
set gyro_1_sensor_align = DEFAULT
set gyro_1_align_yaw = 2700
set gyro_2_sensor_align = DEFAULT

profile 0

# profile 0
set throttle_boost = 0
set p_pitch = 58
set i_pitch = 50
set d_pitch = 35
set p_roll = 40
set i_roll = 40
set d_roll = 30
set p_yaw = 70
set i_yaw = 45
set d_yaw = 20
set f_yaw = 0
set horizon_level_strength = 0
set horizon_transition = 0

rateprofile 0

# rateprofile 0
set pitch_rc_rate = 0
###ERROR: diff: CORRUPTED CONFIG: pitch_rc_rate = 0###
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_percent = 0
###ERROR: diff: CORRUPTED CONFIG: throttle_limit_percent = 0###

# end the command batch
batch end
```

and 

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: FREE
B11: SERIAL_RX 3
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

etracer65, 2020-08-18:
It should be quite obvious by all the warnings that your configuration is corrupted. Reset by doing the "defaults" command in the CLI and set up from scratch - don't paste in a config from a previous version.

makalele666, 2020-08-18:
ok i did it, pitch control works but there is a new problem, GPS stopped working. I set everything up as before, 0 satellites no fix.
changed set gps_sbas_mode = AUTO with no improvement, what could it be this time?
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun 14 2020 / 03:04:22 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# feature
feature -AIRMODE
feature MOTOR_STOP
feature GPS

# map
map TAER1234

# serial
serial 0 2 115200 9600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# aux
aux 0 0 2 1300 2100 0 0
aux 1 1 0 925 1375 0 0
aux 2 2 0 1300 1700 0 0
aux 3 13 1 1700 2100 0 0
aux 4 28 0 1700 2100 0 0

# rxfail
rxfail 4 s 900
rxfail 5 s 2000

# master
set acc_calibration = -11,-15,0,1
set rssi_channel = 12
set serialrx_provider = SBUS
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1500
set vbat_min_cell_voltage = 328
set vbat_warning_cell_voltage = 340
set gps_sbas_mode = AUTO
set gps_rescue_min_sats = 7
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 4
set osd_vbat_pos = 14570
set osd_rssi_pos = 2433
set osd_throttle_pos = 2273
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_gps_sats_pos = 2070
set osd_home_dir_pos = 2062
set osd_home_dist_pos = 2064
set osd_altitude_pos = 2295
set osd_avg_cell_voltage_pos = 2282
set osd_disarmed_pos = 2282
set osd_stat_battery = ON
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```

etracer65, 2020-08-18:
Check your wiring and config. This is not the place to request configuration help. Try the Facebook page:
https://www.facebook.com/groups/betaflightgroup

or Slack:
https://slack.betaflight.com

closed 2020-10-04 by stale[bot]