betaflight/betaflight #10110: Super Drop

ATKCA, 2020-08-18:
Super drop addresses the issue of not falling fast enough at zero throttle while level and upright. When enabled you fall faster with motor reverse assist when at zero throttle. Works much like Air mode but while level and no more than 20° of tilt. When enabled the motors will stabilise in reverse thrust causing an accelerated downward motion and maintained attitude. I think this feature would work best on a momentary switch so it can be enabled when needed and not interfere with normal Air mode flight. The degree of motor reversal can be tuned to preference of assisted level fall speed. 
This feature could also work in angle mode making it useful for Cinewhoop types too. 
This feature could also include a front rear motor bias setting to offset camera tilt. 

Thanks for your consideration.

etracer65, 2020-08-18:
With the types of ESCs we currently use this is not practical. The motor reversing is too slow and generally not very consistent. Also while the motors are spinning down, reversing, and spinning back up you'll have no attitude stability. This is one of the limitations of 3D multi rotor flight performance. Since the motors are rarely all at the same speed the deceleration and reversing takes a different amount of time for each motor causing more instability. While it could technically be possible, the flight behavior would be ugly. Also I'd be very worried about the reversal at the bottom of the "dive" to regain control. Instability and delay is not something you want when trying to recover before hitting the ground.

ATKCA, 2020-08-18:
Hi etracer65,

I figured if the feature only functions when at zero throttle the effect of reversal would be minimal given the pilot would have throttled down prior to pressing the momentary activation switch. Also the degree of reversal will only be minimal and about minus 10% max probably minus 5% nominally. It is meant to increase fall speed rather than punch out in reverse. I agree if the reverse throttle was high it could be ugly. Surely low throttle transitions would be less ugly right? Also you could let go of the momentary switch before you hit the ground and slow the decent prior to throttling up. Does this sound any better?

etracer65, 2020-08-18:
I still think it would be ugly. Also extremely limited application and not sure how/if people would use this. But if someone want's to implement and test to prove viability then that's up to them.

ATKCA, 2020-08-18:
Thanks for your input etracer65. The idea just came after noticing even with low idle or Air mode off the fall speed was slow. Now it's loose perhaps someone will take interest.

closed 2020-10-04 by stale[bot]