betaflight/betaflight #10139: Bug with Tmotor F7 HD fc current sensor in BF 4.2.2 ?

emil737, 2020-08-24:
Setup: 
tmotor f7 stack HD version 
Tmotor F55 pro II 6s 4in1 32bit ESC
Tmotor F722 flight controller
They are connected together with the pin connector (no soldering)

Problem:
Current reading (volt and amps) do not work on BF 4.2.2. It reads 0 in the battery and power tab.
When I downgrade to 4.2.1 the current reading works.

The settings are default after upgrade, so there is no «diff all» to post.

haslinghuis, 2020-08-24:
If you are in for it you could check the development builds to see where it went wrong. I guess starting with 4.3.0. 2247 and go up?

emil737, 2020-08-25:
> If you are in for it you could check the development builds to see where it went wrong. I guess starting with 4.3.0. 2247 and go up?

Unfortunally I do not have the knowledge to do that. I can try, but I do not know what to look for. I just wanted to help report a possible bug.

haslinghuis, 2020-08-25:
In the Update Firmware Tab in the Betaflight Configurator select `Show unstable releases` and `Enable Expert Mode`. In the first box select `Development`, the second box your Board and in the third box the `version`. Please report the version where ADC stops working. (When done with testing please flash a stable release before flying)

[image]

emil737, 2020-08-25:
> In the Update Firmware Tab in the Betaflight Configurator select `Show unstable releases` and `Enable Expert Mode`. In the first box select `Development`, the second box your Board and in the third [...]

Thank you for the guidance. I will try that.

emil737, 2020-08-31:
I have done some more testing and it is most likely not a bug in the software. Maby something with the hardware. Maby mine is defect or something wierd with tmotors products.

Here is the problem. I have got the current sensor to work after reflashing on every betaflight version.

The voltage is showing correct values. The current sensor is displaying the correct values right after software flash. The amperage draw shows values around 0.3 amps when just plugging inn battery.

But after selecting the motor protocol (dshot 600) or any other protocoll the amperage draw jumps to 3.8 amps when just plugging inn battery.

Does anyone know what the problem is? Is the current sensor located in the esc or the fc? Have I connected wrong or is my hardware defect?

haslinghuis, 2020-08-31:
Please try: https://intofpv.com/t-current-sensor-telemetry
But looking at the manuals it should just work.
https://store-en.tmotor.com/goods.php?id=935
https://store-en.tmotor.com/goods.php?id=821

emil737, 2020-08-31:
Thanks for the help. Really appriciate it. I am just going to forget this issue and learn to use battery voltage instead. I do not want to use more time on this issue.

closed 2020-08-31 by emil737