betaflight/betaflight #10140: usb_hid_cdc = on not detecting all controls by windows 10 game controller.

simtcrom, 2020-08-25:
**Describe the bug**
When using usb_hid_cdc = on not all controls are detected by windows 10 game controller only few works.

**To Reproduce**
In Betaflight receiver tab, I got all switches and pots working fine (So I guess all inputs and mixes setup correctly in taranis)

The issue is, after I run set usb_hid_cdc = on in betaflight and check Game controller properties in windows, I don't see movement in windows for all the switches/pots.
Throttle/Yaw, Rolle/Pitch, SD and SC works.
SA, SB, SE and the pot has no movements in windows.
Any idea how to get this fixed?

If I directly connect the taranis using usb cable, I got everything works.

**Expected behavior**
All controls should get decked by windows 10.

haslinghuis, 2020-08-25:
Found this: https://github.com/betaflight/betaflight/pull/6099

> It's just as strange as the fact that out of the 8 channels that are sent in an array, channels 2 and 6 seem to be not available in Windows...

simtcrom, 2020-08-25:
> Found this: #6099
> 
> > It's just as strange as the fact that out of the 8 channels that are sent in an array, channels 2 and 6 seem to be not available in Windows...

Any fix?

dkustec, 2020-08-26:
As declared in the HID Description in `usbd_hid_core.c` there are 8 channels possible to map:
**Throttle, Yaw, Pitch, Roll, Aux1, Aux2, Aux3 and Aux4**
```
__ALIGN_BEGIN static uint8_t HID_MOUSE_ReportDesc[HID_MOUSE_REPORT_DESC_SIZE] __ALIGN_END =
{
0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
0x09, 0x05,                    // USAGE (Game Pad)
0xa1, 0x01,                    // COLLECTION (Application)
0xa1, 0x00,                    //   COLLECTION (Physical)
0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
0x09, 0x30,                    //     USAGE (X)     <-- ROLL
0x09, 0x31,                    //     USAGE (Y)     <-- PITCH
0x09, 0x32,                    //     USAGE (Z)     <-- AUX3
0x09, 0x33,                    //     USAGE (Rx)    <-- YAW
0x09, 0x35,                    //     USAGE (Rz     <-- AUX1
0x09, 0x34,                    //     USAGE (Ry)    <-- THROTTLE
0x09, 0x40,                    //     USAGE (Vx)    <-- AUX4
0x09, 0x38,                    //     USAGE (Wheel) <-- AUX2
0x15, 0x81,                    //     LOGICAL_MINIMUM (-127)
0x25, 0x7f,                    //     LOGICAL_MAXIMUM (127)
0x75, 0x08,                    //     REPORT_SIZE (8)
0x95, 0x08,                    //     REPORT_COUNT (8)
0x81, 0x02,                    //     INPUT (Data,Var,Abs)
0xc0,                          //   END_COLLECTION
0xc0                           /*     END_COLLECTION	             */
};
```

OpenTX, in the other, hand has slightly more usages in the HID descriptor, so mapping more channels there is possible.
In the BF, it may be done by refactoring **HID Report Descriptor** and adding more channel values in `usb_cdc_hid` file.
As the result, this is the Windows Joystic Properties Window for the Betaflight Joystick:
[image]

And for the reference, this is the OpenTX Joystick Properties Window:
[image]

simtcrom, 2020-08-26:
> As declared in the HID Description in `usbd_hid_core.c` there are 8 channels possible to map:
> **Throttle, Yaw, Pitch, Roll, Aux1, Aux2, Aux3 and Aux4**
> 
> ```
> __ALIGN_BEGIN static uint8_t HID_MOUSE_ReportDesc[HID_MOUSE_REPORT_DESC_SIZE] __ALIGN_END =
> {
> 0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
> 0x09, 0x05,                    // USAGE (Game Pad)
> 0xa1, 0x01,                    // COLLECTION (Application)
> 0xa1, 0x00,                    //   COLLECTION (Physical)
> 0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
> 0x09, 0x30,                    //     USAGE (X)     <-- ROLL
> 0x09, 0x31,                    //     USAGE (Y)     <-- PITCH
> 0x09, 0x32,                    //     USAGE (Z)     <-- AUX3
> 0x09, 0x33,                    //     USAGE (Rx)    <-- YAW
> 0x09, 0x35,                    //     USAGE (Rz     <-- AUX1
> 0x09, 0x34,                    //     USAGE (Ry)    <-- THROTTLE
> 0x09, 0x40,                    //     USAGE (Vx)    <-- AUX4
> 0x09, 0x38,                    //     USAGE (Wheel) <-- AUX2
> 0x15, 0x81,                    //     LOGICAL_MINIMUM (-127)
> 0x25, 0x7f,                    //     LOGICAL_MAXIMUM (127)
> 0x75, 0x08,                    //     REPORT_SIZE (8)
> 0x95, 0x08,                    //     REPORT_COUNT (8)
> 0x81, 0x02,                    //     INPUT (Data,Var,Abs)
> 0xc0,                          //   END_COLLECTION
> 0xc0                           /*     END_COLLECTION	             */
> };
> ```
> 
> **OpenTX**, in the other hand has slightly more usages in the HID descriptor, so mapping more channels there is possible:
> 
> ```
> __ALIGN_BEGIN static const uint8_t HID_JOYSTICK_ReportDesc[] __ALIGN_END =
> {
>     0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
>     0x09, 0x05,                    //     USAGE (Game Pad)
>     0xa1, 0x01,                    //     COLLECTION (Application)
>     0xa1, 0x00,                    //       COLLECTION (Physical)
>     0x05, 0x09,                    //         USAGE_PAGE (Button)
>     0x19, 0x01,                    //         USAGE_MINIMUM (Button 1)
>     0x29, 0x18,                    //         USAGE_MAXIMUM (Button 24)
>     0x15, 0x00,                    //         LOGICAL_MINIMUM (0)
>     0x25, 0x01,                    //         LOGICAL_MAXIMUM (1)
>     0x95, 0x18,                    //         REPORT_COUNT (24)
>     0x75, 0x01,                    //         REPORT_SIZE (1)
>     0x81, 0x02,                    //         INPUT (Data,Var,Abs)
>     0x05, 0x01,                    //         USAGE_PAGE (Generic Desktop)
>     0x09, 0x30,                    //         USAGE (X)
>     0x09, 0x31,                    //         USAGE (Y)
>     0x09, 0x32,                    //         USAGE (Z)
>     0x09, 0x33,                    //         USAGE (Rx)
>     0x09, 0x34,                    //         USAGE (Ry)
>     0x09, 0x35,                    //         USAGE (Rz)
>     0x09, 0x36,                    //         USAGE (Slider)
>     0x09, 0x36,                    //         USAGE (Slider)
>     0x16, 0x00, 0x00,              //         LOGICAL_MINIMUM (0)
>     0x26, 0xFF, 0x07,              //         LOGICAL_MAXIMUM (2047)
>     0x75, 0x10,                    //         REPORT_SIZE (16)
>     0x95, 0x08,                    //         REPORT_COUNT (8)
>     0x81, 0x02,                    //         INPUT (Data,Var,Abs)
>     0xc0,                          //       END_COLLECTION
>     0xc0                           //     END_COLLECTION
> };
> ```
> 
> In the BF, it may be done by refactoring of the **HID Report Descriptor** and adding more channel values in `usb_cdc_hid` file.
> As the result, this is the Windows Joystic Properties Window for the Betaflight Joystick:
> [image]
> 
> And for the reference, this is the OpenTX Joystick Properties Window:
> [image]

Sorry I am not able to understand what need to be done to get more channels working in **STM32 Virtual ComPort in FS Mode** mode.
Do I have to make changes in betaflight? or on my radio inputs/mixes ?

dkustec, 2020-08-26:
> Do I have to make changes in betaflight? or on my radio inputs/mixes ?
As an user, you can do nothing to get more channels. Some mechanisms in the BF code need changes.

simtcrom, 2020-08-26:
> > Do I have to make changes in betaflight? or on my radio inputs/mixes ?
> 
> As an user, you can do nothing to get more channels. Some mechanisms in the BF code need changes.

Got it. So a change in the source code is needed and it need to be compiled and built.
Would you be able to make a custom version built for me? 
I don’t intent to use this version for flying outdoor just to make game controller work.
My fc is matek F7 se.

dkustec, 2020-08-26:
> Got it. So a change in the source code is needed and it need to be compiled and built.

Correct.

> Would you be able to make a custom version built for me?

Refactoring of the HID Descriptor is not simple task. If I do that, I'll mention here.

simtcrom, 2020-08-26:
> > Got it. So a change in the source code is needed and it need to be compiled and built.
> 
> Correct.
> 
> > Would you be able to make a custom version built for me?
> 
> Refactoring of the HID Descriptor is not simple task. If I do that, I'll mention here.

Sure. And thank you.

dkustec, 2020-08-27:
@simtcrom , I did it LOL. It's prototyped and it works. Just tested with my Tango 2 and F4 board. There should be 16 channles, but on the Tango 2, I can't map more than 12. I'll try with Taranis later. 
[image]

simtcrom, 2020-08-27:
> , I can't map more than 10 physical. I'll try with Taranis later.

@dkustec 
Wow that,s great.
My radio is FrSky Taranis X9 Lite. Its got 4 + 6 (AUX) physical channels. Hope it will work for me too.
Please let me know if you can build for Matek F722 SE (target - MATEKF722SE).

dkustec, 2020-08-27:
@simtcrom , I have that board too. I'll try when I'm done with the F4. (Drivers are different between mcu's so it needs more codings).

dkustec, 2020-08-27:
@simtcrom , Here it is. 
There are two approaches how to flash this board:

1.  Flash this unified binary: 
[betaflight_4.3.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/5138124/betaflight_4.3.0_STM32F7X2_norevision.zip)
... and follow installation instructions from here: https://youtu.be/I1uN9CN30gw
2. Use this hex file and flash it directly (please do 'Erase chip' command)
[betaflight_4.3.0_MATEKF722SE_norevision.zip](https://github.com/betaflight/betaflight/files/5138131/betaflight_4.3.0_MATEKF722SE_norevision.zip)

Please try with the both methods.

When you do `usb_hid_cdc = on`, you should see this properties window:
[image]

Remember: this is an unoptimized piece of code and **do not fly with**. And please, give me some feedback here.

simtcrom, 2020-08-28:
> @simtcrom , Here it is.
> There are two approaches how to flash this board:
>
> 1. Flash this unified binary: [...]

@dkustec 
Thanks a lot.
I have tested both methods.

After attempting first method, when I try to connect using betaflight, I get error unable to open serial port. I tried disconnecting and connecting the usb and still same error.
Then went for second method and it worked great. Now I could get 10 channels working.

I am flying Microsoft Flight Simulator 2020 with my taranis. Having extra 4 channels will be definitely useful.

Once again thank you.

simtcrom, 2020-08-30:
Just curious if this will be merged to main branch and released on normal betaflight release?

dkustec, 2020-08-30:
@simtcrom , Well, I made an PR #10150 for this improvement. We'll see after some code reviewing will it be merged to the Master repo, but I see no reason why not.

dkustec, 2020-09-17:
> Just curious if this will be merged to main branch and released on normal betaflight release?

@simtcrom , just to inform you - it's merged and it'll be in the very next release of BF.

simtcrom, 2020-09-17:
> > Just curious if this will be merged to main branch and released on normal betaflight release?
> 
> @simtcrom , just to inform you - it's merged and it'll be in the very next release of BF.

Thank you Sir.

minhnkt, 2020-09-20:
> @simtcrom , I did it LOL. It's prototyped and it works. Just tested with my Tango 2 and F4 board. There should be 16 channels, but on the Tango 2, I can't map more than 10 physical. I'll try with Taranis later.
> [image]

Amazing !!! Im using F4 pro v3 too and Futaba t8fg with sbus via R6208SB, could you share me your hex file and guide me ? Thank you very much, sir.

mikeller, 2020-09-20:
@minhnkt: If you are willing to use experimental / unstable firmware you can use the latest build from the 'Development' branch in configurator, which has these changes.

minhnkt, 2020-09-29:
Hi @simtcrom, 
glad to know that it works for you, as @mikeller advise, i flashed 4.3 firmware to my F4 and my T8FG works like a charm without do anything, windows now can detect my transmitter as a game controller, but MS Flight Simulator 2020 doesn't, im reading around few days to find how to make MFS 2020 can detect my transmitter but no luck, can you tell me how to do that? 
Thank you.

simtcrom, 2020-09-29:
> Hi @simtcrom,
> glad to know that it works for you, as @mikeller advise, i flashed 4.3 firmware to my F4 and my T8FG works like a charm without do anything, windows now can detect my transmitter as a [...]

Can you refer https://youtu.be/S16Ja9XrqNY ? I had made this video.

dkustec, 2020-09-29:
@minhnkt , @simtcrom , Just to inform you - you don't need Development branch anymore, because all of this is already in the stable, 4.2.3 official version of BF.

minhnkt, 2020-09-29:
@simtcrom
Hi, i watched your video, it's quite excited! But i still don't know how you make MFS 2020 can detect your transmitter via your F4.
Hope you can tell me something to do it.
Thank you .

@dkustec
Thanks God, my quad can still fly :)) thank you so much !

simtcrom, 2020-09-29:
> @simtcrom
> Hi, i watched your video, it's quite excited, i can see how you config the transmitter, but i still don't know how you make MFS 2020 can detect your transmitter via your F4.
> **"SMT 32 Virtual Comport IN FS..."** do not come up in my Control section :(
> Hope you can tell me something to do it.
> Thank you .

> @simtcrom
> Hi, i watched your video, it's quite excited, i can see how you config the transmitter, but i still don't know how you make MFS 2020 can detect your transmitter via your F4.
> **"SMT 32 Virtual Comport IN FS..."** do not come up in my Control section :(
> Hope you can tell me something to do it.
> Thank you .

All I had to do was, in betaflight ran the command usb_hid_cdc = on and the controller got dented by windows. I made sure that I have movements on windows game controller when I move sticks.
Then launch Ms flight sim, and under controllers I get stm32 virtual....
My game pass got over and lost the simulator.

minhnkt, 2020-09-30:
> All I had to do was, in betaflight ran the command usb_hid_cdc = on and the controller got dented by windows. I made sure that I have movements on windows game controller when I move sticks.
> Then [...]

I did the same, but my MFS still cannot recognize >_< I try to flash BF 4.3 (Dev) and 4.2.3 (STABLE) but no luck :(

When i check Device Manager, even i "set usb_hid_cdc = on", my F4 controller still listed in **PORTS (COM & LPD) section** via COM port, DO NOT SHOW in **Human Interface Devices section** :( 
do you think that's the reason MFS cannot detect F4 as game controller?

Do me a favor, can you check and give me firmware version of BetaFlight, what version F4 FC and Windows version do you use?
I'll try to reinstall Windows and buy another FC and flash firmware as yours.

Thank you very much.

dkustec, 2020-09-30:
@minhnkt , Did you save your config in CLI? After typing `set usb_hid_cdc = on` press enter and then type `save`.

simtcrom, 2020-09-30:
@minhnkt minhnkt

Attached the screenshot of device manager when I connect my FC.

# version
# Betaflight / MATEKF722SE (MF7S) 4.3.0 Aug 27 2020 / 20:22:37 (norevision) MSP API: 1.44
I still have the custom version provided by @dkustec flashed. I dont fly these days. Hence never bothered to flash release version
[image]

Windows Version	10.0.19041 Build 19041

The very beginning I only had to run set usb_hid_cdc = on in bf once, after that every time I connect FC, I get **STM32 Virtual ComPort in FS Mode** in Devices and Printers automatically. At same time, I can connect to FC using betaflight using com3 port. I dont know this is normal or not.

Could it be because microsoft blocked using this controller in flight simulator using an update or something? Since I lost gamepass, I cannot test.

minhnkt, 2020-09-30:
> @minhnkt , Did you save your config in CLI? After typing `set usb_hid_cdc = on` press enter and then type `save`.
@dkustec 
Of course i did it, for sure i recheck FC mode by "get usb" command and get:
# get usb
usb_hid_cdc = ON
Allowed values: OFF, ON
Default value: OFF

usb_msc_pin_pullup = ON
Allowed values: OFF, ON

minhnkt, 2020-10-01:
@simtcrom
I've just buy a new MF7S FC, hope it'll work.
Huh, why you said that lost gamepass (gamepad ?) you couldn't test ? 
Now you still using your taranis TX for playing MSF2020 after 1.8.3.0 patch or not? If you can play, i don't think MS block this feature.

simtcrom, 2020-10-01:
> @simtcrom
> I've just buy a new MF7S FC, hope it'll work.
> Huh, why you said that lost gamepass (gamepad ?) you couldn't test ?
> Now you still using your taranis TX for playing MSF2020 after 1.8.3.0 patch or not? If you can play, i don't think MS block this feature.

I had got this game as part of ms gamepass subscription. Which I had got for a month only.
Once it’s over I lost ability to launch and play the game.
I don’t know what was the version of the game when I was playing with taranis.

closed 2020-12-25 by stale[bot]

simtcrom, 2021-01-26:
@minhnkt 
Today I installed Flight Sim again (got gamespass for 3 monts). Version is 1.12.13.0.
I used Matek F7 SE and installed betaflight 4.2.X stable version.
And everything working as expected.

simtcrom, 2021-01-29:
@dkustec 
If I am not asking too much,
Currently on my FrSky Taranis X9 lite, momentary switch and rotary dial are detected as Buttons in Windows Game Controller properties. Any way these also can be set as others?

And I believe due to this, these two controls are not available on MS Flight Sim to choose. If it does get detected, Dial could have been used for flaps.

May be,

Throttle/Yaw in Y/X axis (this is currently on Y and X rotation)
Pitch/Roll in Y/X (this is set correct)
AUX 1 on Z Rotation (this is set correct)
AUX 2 on Z Axis (this is set correct)
AUX 3 on Slider (this is set correct)
Dial to X Rotation (this is currently n Button 2)
AUX 4 on Dial (this is set correct)
Momentary Switch to Y Rotation (this is currently n Button 1)

dkustec, 2021-02-04:
@simtcrom , I think I already try that. Exactly the same config but it didn't end well. All axes got weird jitters in Windows. 
It's because this driver register itself to Windows as a game pad and not the joystick, but Windows expects correct axes and rotations alignments, eg. the Y Rotation should be on exact bit-order if we want to Windows recognize that bit(s). If it fails, we could get weird behavior of one or all controls. So I can try again following your advice, but I can't promise when this will happen, because I'm flooded with other personal projects and I can't afford much time for BF currently.

simtcrom, 2021-02-05:
> @simtcrom , I think I already try that. Exactly the same config but it didn't end well. All axes got weird jitters in Windows.
> It's because this driver register itself to Windows as a game pad and [...]

Totally understand.
Thank you.