betaflight/betaflight #10143: HGLRCF722 - USER1 Issue

Barux, 2020-08-25:
**Describe the bug**
USER1 mode switch off camera 1 instead of vtx

**To Reproduce**
Wire Cam and vtx as written in the manual. Set  USER1 on and off to check what happens

**Expected behavior**
VTx to switch off instead of camera

**Setup / Versions**
 - Flight controller [HGLRCF722 - BF 4.2.2];
 - Other components [Nano Predator 4, TBS nano 32];

**Additional context**
Not sure if software issue or hardware since the fc can support up to 2 cameras but actually USER1 is useful to power off the vtx

haslinghuis, 2020-08-25:
Please add a link to the manual and provide a `diff`.

Barux, 2020-08-26:
**Link Manual**
https://bit.ly/2YvlEBp

**Diff All**
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.2 Aug 16 2020 / 01:48:06 (e833ac612) MSP API: 1.43
# config: manufacturer_id: HGLR, board_name: HGLRCF722, version: d1c47a55, date: 2020-04-18T09:09:56Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name HGLRCF722
manufacturer_id HGLR
mcu_id 0014002a4638500420383937
signature 

# name: SpeedyPizzaDrones

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature OSD

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 2 2048 115200 57600 0 115200

# led
led 0 0,0::CT:13
led 1 1,0::CT:13
led 2 2,0::CT:13
led 3 3,0::CT:13
[... 262 lines cut ...]
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Barux, 2020-08-26:
@haslinghuis

haslinghuis, 2020-08-26:
> Not sure if software issue or hardware since the fc can support up to 2 cameras but actually USER1 is useful to power off the vtx

USER1/USER2 are defined for controlling Camera Input Selection here and cannot be used to use PIT mode this way.

But because the TBS Nano Pro32 supports SmartAudio 2.1 with Pit mode you could take a look at: https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio

**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

closed 2020-10-12 by stale[bot]