betaflight/betaflight #10144: HOTT Telemetry Textmode not working from 4.2.0

rogere66, 2020-08-25:
HOTT Telemetry Textmode not working from version 4.2.0 - OK in version 4.1.7
GPS and basic EAM works fine.

**Flight controller configuration**
Diatone Mamba F722 MINI MK2 Flight Controller
SUMD Serial on UART1
GPS on UART4
HOTT Telemetry on UART5 Tx pin
Graupner GR-12 Rx, MX-20 Tx
```
diff:
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 14 2020 / 03:05:26 (8f2d21460) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: a5269145, date: 2020-04-09T22:32:24Z

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# feature
feature -RX_PARALLEL_PWM
feature GPS

# map
map TAER1234

# serial
serial 3 2 115200 9600 0 115200
serial 4 8 115200 57600 0 115200

# master
set acc_calibration = 58,-25,107,1
set serialrx_provider = SUMD
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
```

haslinghuis, 2020-08-26:
Did it work on a previous release?
See note 1: https://forum.flitetest.com/index.php?threads/kiss-fc-to-graupner.32565/#lg=post-341280&slide=0

rogere66, 2020-08-26:
Yes, it worked fine on 4.1.7

haslinghuis, 2020-08-26:
Nothing to complain about your configuration (looks fine to me). Not finding any issues. So without this hardware it is hard to diagnose for me.

haslinghuis, 2020-08-26:
Only maybe try to set uart5 (port 4) to 19200 but the code already seems the enforce this: `#define HOTT_BAUDRATE 19200`

rogere66, 2020-08-28:
Telemetry communication is OK since GPS and basic EAM telemetry works fine. It is only text mode that doesn't work after 4.1.7 - it only shows a blank screen.
I have now tested this also on a Mateksys F722-STD with the same result. It is probably the same for all boards, at least F722 based.

mikeller, 2020-09-01:
Hard to tell what is wrong without hardware - but maybe @Ziege-One can help?

Ziege-One, 2020-09-02:
Hello,
I can conform that the Textmode not work with 4.2.0 (blank screen)

4.1.7 works very well.

Hardware:
OMNIBUSF4SD clone
GR12L

mikeller, 2020-10-04:
@Ziege-One: I had a quick look but could not spot any obvious culprit for why this has broken. Do you have time to have a peek into how this can be fixed?

Ziege-One, 2020-10-07:
Hello
the only change on the file displayport_hott.c is "Add support for character attribute rich displayport" from 
jflyper committed on 25 Nov 2019.
the mornal telemtry works fine so there are no hardware problems.

mikeller, 2020-10-11:
@Ziege-One: #9212 does not look like it would contain anything that could break HOTT Textmode - but there have been some other changes that might have done it. Unfortunately I do not have any test gear to allow me to debug this - are you set up to see if you can figure out what is going wrong?

Ziege-One, 2020-10-11:
@mikeller Yes I can compile and test but I don't have a live debugging System.

mikeller, 2020-10-12:
@Ziege-One: Beats me - I don't have any TX / RX that does HoTT. :stuck_out_tongue:

Ziege-One, 2020-10-12:
@mikeller > but there have been some other changes that might have done it. Can you say the changes then I will look?

mikeller, 2020-10-13:
@Ziege-One: One area where I'd look is changes to the CMS (the framework that generates the menus):

```
git log -- src/main/cms/cms.c
```

This will show you quite a few changes that have been made since 4.1.7.

old-frizt, 2021-01-08:
> It is probably the same for all boards, at least F722 based.

I have the same problem with a Mamba F405 Mk2. So F4 Boards seem to have the same problem.

klutvott123, 2022-04-21:
This is probably because the cms initialization was moved so it wasn't initialized at the time when telemetry tried registering a disaplyport. This has been fixed and it should work in 4.3. Could you guys try it and report back, please?

closed 2022-05-12 by github-actions[bot]