betaflight/betaflight #10145: Bug that make motor 1 and 4 fail  under certain circumstances

ss8688, 2020-08-26:
I am using a Iflight SucceX micro F4 tower（Iflight f411 pro with betaflight4.2.2 and a  G_H_30 esc with JazzMaverick beheli s 16.8 48khz ). When set PID to 8khz ，set esc protocol to dshot 600 and enable bidirectional dshot，cpu load is around 80%.most of the time motor 1 and 4 can not work and get 100% error rates，sometime all motor  can work，sometime all motor fail.but if I set PID to 4 khz everything
 is OK

ss8688, 2020-08-26:
Is that a bug of betaflight or Blheli s?

Asizon, 2020-08-26:
You are using experimental blheli version in Sync mode, please try a recomended blheli version 16.73 or try runing last version in Async mode

ss8688, 2020-08-26:
**diff when all motor can work**
# diff

# version
# Betaflight / STM32F411 (S411) 4.2.2 Aug 16 2020 / 01:47:01 (e833ac612) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F411_PRO, version: bc19b7dc, date: 2020-04-01T04:37:12Z

# start the command batch
batch start

board_name IFLIGHT_F411_PRO
manufacturer_id IFRC

# name: CineBee4K

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1900 2075 0 0
aux 1 1 0 900 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 25 100 200
vtxtable powerlabels 25 100 200

# master
set acc_trim_pitch = -10
set acc_trim_roll = -10
set acc_calibration = 66,7,-2,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = FPORT
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set current_meter = NONE
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_compass_bar_pos = 2282
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5865
set name = CineBee4K

profile 0

# profile 0
set p_pitch = 53
set i_pitch = 39
set d_pitch = 27
set p_roll = 49
set i_roll = 38
set d_roll = 25
set p_yaw = 63
set i_yaw = 25

rateprofile 0

# end the command batch
batch end

**diff when motor 1 and 4 fail**

# diff

# version
# Betaflight / STM32F411 (S411) 4.2.2 Aug 16 2020 / 01:47:01 (e833ac612) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F411_PRO, version: bc19b7dc, date: 2020-04-01T04:37:12Z

# start the command batch
batch start

board_name IFLIGHT_F411_PRO
manufacturer_id IFRC

# name: CineBee4K

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1900 2075 0 0
aux 1 1 0 900 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 25 100 200
vtxtable powerlabels 25 100 200

# master
set acc_trim_pitch = -10
set acc_trim_roll = -10
set acc_calibration = 66,7,-2,1
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = FPORT
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set current_meter = NONE
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_compass_bar_pos = 2282
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5865
set name = CineBee4K

profile 0

# profile 0
set p_pitch = 53
set i_pitch = 39
set d_pitch = 27
set p_roll = 49
set i_roll = 38
set d_roll = 25
set p_yaw = 63
set i_yaw = 25

rateprofile 0

# end the command batch
batch end

Asizon, 2020-08-26:
Wrong configurator, for F411 use Dshot 400 8k/4k

ss8688, 2020-08-26:
> Wrong configuration, for F411 use Dshot 300 8k/4k

I don't think it is a configuration problem.
Dshot 600  bidirectional  can work on  f411 because  bf reworked the gyro loop on 4.2 and it make cpu load lower ，I have trid it on another f411(cpu overclocking to 102mhz) with blheli 32 esc .It work  well.
so I will try blheli s 16.3 to see if it works。

Asizon, 2020-08-26:
No, it can run 8k,please read 4.2 release notes

Asizon, 2020-08-26:
https://github.com/betaflight/betaflight/wiki/4.2-Tuning-Notes#more-accurate-loop-times

ss8688, 2020-08-26:
Ok I understand，I tried blheli_s 16.77 5 minute ago and Dshot 600 bidirectional work with 85 % cpu load . is that a bug make it work?I have no idea.What is going on?

Asizon, 2020-08-26:
Again, you are use wrong configuration, you need to use 4k,also you are playing with experimentals blheli version. This issue will be closed by user configuration problem.

ss8688, 2020-08-26:
Ok,but I think some should try it again because I saw it work with Blheli s 16.77 48hz . In principle It should  not work.but it work so there might be another bug.

ss8688, 2020-08-26:
[G_H_30_48_REV16_77 (1)  (8位电调固件）.zip](https://github.com/betaflight/betaflight/files/5127944/G_H_30_48_REV16_77.1.8.zip)
This is the 16.77 blheli s that I use

Asizon, 2020-08-26:
Again, we doesnt support Jazzmaverick experimental versions

haslinghuis, 2020-08-26:
@ss8688 As a warning, having two drones with exact same hardware, flashed latest Jazz release, one seems okay, the other must be recovered because it bricked the firmware. 🎶 No experimental Jazz for me anymore!

closed 2020-10-12 by stale[bot]